nextcloud/nextcloudpi #1334: Trusted proxy and trusted domain settings are lost

vcbranco, 2021-09-16:
Trusted proxy and trusted domain settings are lost if you reboot the Nextcloud.

Debian 10 x86_64, curl installation

After the reboot all the configuration always need to be set again in order to have the HPB service running again

vcbranco, 2021-09-16:
ncc notify_push:setup https://cloud.emcontacto.eu/push

🗴 can't connect to push server: Server error: `GET https://cloud.emcontacto.eu/push/test/cookie` resulted in a `503 Service Unavailable` response:
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>503 Service Unavailable</title>
</head><body>
<h1 (truncated...)

ncp-report

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.39.1
NextCloudPi image    NextCloudPi_09-16-21
distribution         Debian GNU/Linux 10 \n \l
automount            yes
USB devices          sda
datadir              /data/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      299G/450G
rootfs usage         5,9G/16G
swapfile             /var/swap
dbdir                /data/ncdatabase
Nextcloud check      ok
Nextcloud version    21.0.4.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eno1
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               10min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "22": "cloud.emcontacto.eu",
            "3": "An unhandled exception has been thrown:\nRedisException: No such file or directory in \nStack trace:\n#0 \n#1 \n#2 \n#3 \n#4 \n#5 \n#6 \n#7 \n#8 \n#9 \n#10 \n#11 \n#12 \n#13 \n#14 \n#15 \n#16 \n#17 \n#18 \n#19 \n#20 \n#21 \n#22 \n#23 {main}:\nRedisException: read error on connection in \nStack trace:\n#0 \n#1 \n#2 \n#3 \n#4 \n#5 \n#6 \n#7 \n#8 \n#9 \n#10 \n#11 \n#12 {main}",
            "11": "85.245.180.61",
            "1": "10.8.6.2"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "21.0.4.1",
        "overwrite.cli.url": "https:\/\/An unhandled exception has been thrown:\nRedisException: No such file or directory in \nStack trace:\n#0 \n#1 \n#2 \n#3 \n#4 \n#5 \n#6 \n#7 \n#8 \n#9 \n#10 \n#11 \n#12 \n#13 \n#14 \n#15 \n#16 \n#17 \n#18 \n#19 \n#20 \n#21 \n#22 \n#23 {main}:\nRedisException: read error on connection in \nStack trace:\n#0 \n#1 \n#2 \n#3 \n#4 \n#5 \n#6 \n#7 \n#8 \n#9 \n#10 \n#11 \n#12 {main}\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
[... 113 lines cut ...]
```

{"reqId":"YUMcGQUDshSZs7-@PZdGEgAAExE","level":2,"time":"2021-09-16T10:27:38+00:00","remoteAddr":"10.8.6.117","user":"--","app":"core","method":"PROPFIND","url":"/remote.php/dav/addressbooks/users/saracb/contacts/","message":"Login failed: 'saracb' (Remote IP: '10.8.6.117')","userAgent":"DAVx5/3.4.0.1-gplay (2021/09/09; dav4jvm; okhttp/4.9.1) Android/11","version":"21.0.4.1"}
{"reqId":"YUMcfAUDshSZs7-@PZdGOgAACQ0","level":2,"time":"2021-09-16T10:29:16+00:00","remoteAddr":"10.8.6.32","user":"vascocb","app":"suspicious_login","method":"POST","url":"/login","message":"Could not predict suspiciousness: No models found","userAgent":"Hmd global Nokia 7.1 (Android)","version":"21.0.4.1"}
{"reqId":"HklxJ2OzTX1gfSQZTSUu","level":2,"time":"2021-09-16T13:15:01+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /data/ncdata/tmp is not present or writable","userAgent":"--","version":"21.0.4.1"}
{"reqId":"YUNIF-3BclZe4bcsNaIhxAAAAAM","level":3,"time":"2021-09-16T13:35:20+00:00","remoteAddr":"10.8.6.1","user":"--","app":"index","method":"POST","url":"/login","message":{"Exception":"Exception","Message":"Argument 1 passed to OC\\Core\\Controller\\LoginController::tryLogin() must be of the type string, null given, called in /var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php on line 218","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/AppFramework/App.php","line":157,"function":"dispatch","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->","args":[{"__class__":"OC\\Core\\Controller\\LoginController"},"tryLogin"]},{"file":"/var/www/nextcloud/lib/private/Route/Router.php","line":302,"function":"main","class":"OC\\AppFramework\\App","type":"::","args":["OC\\Core\\Controller\\LoginController","tryLogin",{"__class__":"OC\\AppFramework\\DependencyInjection\\DIContainer"},{"_route":"core.login.tryLogin"}]},{"file":"/var/www/nextcloud/lib/base.php","line":993,"function":"match","class":"OC\\Route\\Router","type":"->","args":["/login"]},{"file":"/var/www/nextcloud/index.php","line":37,"function":"handleRequest","class":"OC","type":"::","args":[]}],"File":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","Line":159,"Previous":{"Exception":"TypeError","Message":"Argument 1 passed to OC\\Core\\Controller\\LoginController::tryLogin() must be of the type string, null given, called in /var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php on line 218","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":218,"function":"tryLogin","class":"OC\\Core\\Controller\\LoginController","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":127,"function":"ex [... cut ...] ",{"__class__":"OC\\AppFramework\\DependencyInjection\\DIContainer"},{"_route":"core.login.tryLogin"}]},{"file":"/var/www/nextcloud/lib/base.php","line":993,"function":"match","class":"OC\\Route\\Router","type":"->","args":["/login"]},{"file":"/var/www/nextcloud/index.php","line":37,"function":"handleRequest","class":"OC","type":"::","args":[]}],"File":"/var/www/nextcloud/core/Controller/LoginController.php","Line":290},"CustomMessage":"--"},"userAgent":"--","version":"21.0.4.1"}

```
</details>

nachoparker, 2021-09-16:
so weird. There's more people seeing this

https://help.nextcloud.com/t/configuring-hpb-with-ncp/123729/9

I'll try to reproduce but it didn't happen during my testing

vcbranco, 2021-09-16:
Sometimes only on the second restart.
The instance is working with the service down.
It's also a little slow.

vcbranco, 2021-09-16:
When I run nc-info I always have the notice that there are no certificates configured and they are.
Could be related?

Running nc-info
Gathering information...
NextCloudPi version  v1.39.1
NextCloudPi image    NextCloudPi_09-16-21
distribution         Debian GNU/Linux 10 \n \l
automount            yes
USB devices          sda 
datadir              /data/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      299G/450G
rootfs usage         5,9G/16G
swapfile             /var/swap
dbdir                /data/ncdatabase
Nextcloud check      ok
Nextcloud version    21.0.4.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   xxx.xxx.xxx.xxx
gateway              xxx.xxx.xxx.xxx
interface            eno1
certificates         none
NAT loopback         no
uptime               4:56

You should run Lets Encrypt for trusted encrypted access
Done. Press any key...

theCalcaholic, 2021-09-16:
Might be related to #1333. The current notify_push configuration likely doesn't work for dynamic ips

theCalcaholic, 2021-09-17:
Can you please connect to your instance over ssh, execute the following command and include its output (make sure to replace all IP addresses and urls with placeholders):

```sh
ncc notify_push:self-test
```

vcbranco, 2021-09-17:
🗴 can't connect to push server: Server error: `GET https://my.domain.tld/push/test/cookie` resulted in a `503 Service Unavailable` response:
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>503 Service Unavailable</title>
</head><body>
<h1 (truncated...)

vcbranco, 2021-09-17:
Maube the IP address. How can I reset the service to test that?
The router is always on and normally the IP doesn't change for months

theCalcaholic, 2021-09-17:
That looks weird. Can you post the output of `ncc notidy_push:logs` as well, please?

You can also try to rerun the notify_push setup with the following command:

```sh
ncc notify_push:setup https://<your-nextcloud.domain>/push
```

vcbranco, 2021-09-17:
I was thinking about dynamic IP and I remembered something
In my DNS server I have the following setting

my.domain.tld is my Nextcloudpi LAN IP

I don't think this is important but maybe can be
What do you think?

vcbranco, 2021-09-17:
That DNS setting and my router acting as proxy for everything can be the issue here.
I can disable the router DHCP server and enable the DHCP from the DNS server but I don't know how to reset the HPB service to test that

theCalcaholic, 2021-09-17:
I think you can reset the HPB by rerunning `ncc notify_push:setup`.

Also, please check, if there's anything interesting in the output of `journalctl -eu notify_push`

vcbranco, 2021-09-18:
I cleaned up the config.php and configured it correctly according to the configuration of another one in a test instance. The HPB service started to run correctly after a few seconds without requiring any intervention from me.

Turned off the router's DHCP server and setup a DHCP/DNS server on the network to solve the problem with the router configured as a proxy.

Forced a WAN IP update and rebooted the Nextcloupi.

The new IP was automatically configured in config.php and the HPB service started normally.

I think the problem is not caused by the dynamic IP.

vcbranco, 2021-09-18:
config.php

trusted_domains
0 => 'localhost',
22 => 'nextcloudpi',  -  Manually configured with Nextcloudpi control panel
3 => 'my.domain.tld',
11 => 'WAN IP',
1 => 'nextcloudpi IP',

trusted_proxies
11 => '127.0.0.1',
12 => '::1',
13 => 'my.domain.tld',
14 => 'nextcloudpi IP',

vcbranco, 2021-09-18:
I noticed that the Client push app from Nextcloud app store was updated today right before this tests

theCalcaholic, 2021-09-18:
Did you manually add "my.domain.tld" to the trusted proxies? I don't think that will do anything (would require a reverse DNS lookup, which you most likely have not setup and I would surprised if NC would support that, in other words: Nextcloud doesn't know whether a request comes from the IP associated with that domain or not).

Do you know what you changed in the config.php? Overwriting the trusted domains might cause other issues along the way, because nextcloudpi assumes various "slots" in that array to mean specific things (see [here](https://github.com/nextcloud/nextcloudpi/blob/master/etc/library.sh#L16)).

vcbranco, 2021-09-18:
No, the domain was added to the trusted proxies by the system in the test instance installation. Only added nextcloudpi to the trusted domains due to a error shown in the installation also in the test instance.

I talked about the Client push app because the update has a warning that says it is necessary to reconfigure with "notify_push:setup" after the update. When I installed the test instance (Debian 10 x86_64, curl installation), the client push app installed was the updated version. Maybe the issue was adjusted there.

What I changed in config.php was:
Delete garbage lines
Add the trusted domains and trusted proxies has showed in the post

vcbranco, 2021-09-18:
Tested v1.39.2 RPI4 4GB last rpi image

The HBP service was down
Has you can see below was necessary to set the trusted proxie (set with nc-trusted-proxies) and run https://my.domain.tld/push
After that all services up

I'm sure that issue is due to my router but I can not test in another one.

Created symlink /etc/systemd/system/multi-user.target.wants/notify_push.service → /etc/systemd/system/notify_push.service.
System config value trusted_domains => 3 set to string my.domain.tld
System config value overwrite.cli.url set to string https://my.domain.tld/
System config value trusted_proxies => 11 set to string 127.0.0.1
System config value trusted_proxies => 12 set to string ::1
System config value trusted_proxies => 13 set to string my.domain.tld
System config value trusted_proxies => 14 set to string WAN IP
✓ redis is configured
✓ push server is receiving redis messages
✓ push server can load mount info from database
✓ push server can connect to the Nextcloud server
🗴 push server is not a trusted proxy, please add 'router IP' to the list of trusted proxies or configure any existing reverse proxy to forward the 'x-forwarded-for' send by the push server.
  See https://docs.nextcloud.com/server/latest/admin_manual/configuration_server/reverse_proxy_configuration.html#defining-trusted-proxies for how to set trusted proxies.
  The following trusted proxies are currently configured: "127.0.0.1", "::1", "my.domain.tld", "WAN IP"
    of which the following seem to be invalid: "my.domain.tld"
  The following x-forwarded-for header was received by Nextcloud: 1.2.3.4
    from the following remote: router IP

  If you're having issues getting the trusted proxy setup working, you can try bypassing any existing reverse proxy
  in your setup by setting the `NEXTCLOUD_URL` environment variable to point directly to the internal Nextcloud webserver url
  (You will still need the ip address of the push server added as trusted proxy)
notify_push setup failed. You are probably behind a proxy
Run 'ncc config:system:set trusted_proxies 15 --value=<proxy_IP>' and then 'ncc notify_push:setup https://<domain>/push to enable
Check https://help.nextcloud.com/tags/ncp for support
Backup stored at /var/www/nextcloud/data/ncp-update-backups/nextcloud-bkp_20210918_1631998302-20.0.2.2.tar.gz
Clean up...

vcbranco, 2021-09-18:
The previous test was a clean installation With Debian 10 x86_64 and curl installation.
The config.php is different but I don't know why. Pls, see the previous test.
The two installations are working as expected.

nachoparker, 2021-09-19:
503 sounds like the notify_push service is not working. About the configurations being overwritten with the redis error message. Can you reproduce this any time you want?

I pushed something today that I think might fix it. This would be related to https://github.com/nextcloud/nextcloudpi/issues/1332

nachoparker, 2021-09-19:
Unless we are still seeing the 503, let's close this and follow up on the redis junk error on #1332

closed 2021-09-19 by nachoparker

thor9898, 2024-04-28:
this issue is now giving me error "Your "trusted_proxies" setting is not correctly set, it should be an array of IP addresses - optionally with range in CIDR notation" in the latest Nextcloudpi

theCalcaholic, 2024-04-29:
@thor9898 Are you on Nextcloud 25.0.8? That's not officially supported yet and I'm just now resolving this issue

thor9898, 2024-04-29:
> @thor9898 Are you on Nextcloud 25.0.8? That's not officially supported yet and I'm just now resolving this issue

I am on 28.0.5

theCalcaholic, 2024-05-02:
@thor9898 Please update to NCP v1.54.0 (run `ncp-update`), that should fix your issue

thor9898, 2024-05-02:
> @thor9898 Please update to NCP v1.54.0 (run `ncp-update`), that should fix your issue

Thanks, it worked!!!

lemurenswe, 2024-05-17:
I am having the same issue where on reboot, my public hostname (duckdns) and wan ip is added to the array of proxies in config.php.

NextcloudPi version | v1.54.2
Nextcloud version | 28.0.5.1

This means i get the warning that thor9898 had in the overview. I can manually remove the hostname from the array and restart php fpm using systemd and the config is loaded without the hostname, but as soon as i reboot the machine the hostname is added again.