nextcloud/nextcloudpi #1335: LetsEncrypt renewal error

mjeshurun, 2021-09-16:
Hello Friends,
I’m running NextckoudPi 1.38.6 on a Raspberry Pi 4 4GB with Raspberry Pi OS 64bit.
Everything was running well for the past two months, but in the past two weeks I’ve received a few Nextcloud notifications saying the LetsEncrypt SSL renewal failed.
I tried to log into the NCP panel and run the Letsencrypt renewal manually, but I got this error message:

```
[ letsencrypt ] (Sun Sep 12 14:07:24 IDT 2021)
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Plugins selected: Authenticator webroot, Installer None
Renewing an existing certificate
Performing the following challenges:
http-01 challenge for ********.duckdns.org
Using the webroot path /var/www/nextcloud for all unmatched domains.
Waiting for verification...
Cleaning up challenges
Failed authorization procedure. ********.duckdns.org (http-01): urn:ietf:params:acme:error:connection :: The server could not connect to the client to verify the domain :: Fetching https://********.duckdns.org/.well-known/acme-challenge/jspBZsvex_V1R42Th7E-9RMQ4GyvIy8cI8r0GhdHxHw: Timeout during connect (likely firewall problem)
IMPORTANT NOTES:
- The following errors were reported by the server:

Domain: ********.duckdns.org
Type: connection
Detail: Fetching
https://********.duckdns.org/.well-known/acme-challenge/jspBZsvex_V1R42Th7E-9RMQ4GyvIy8cI8r0GhdHxHw:
Timeout during connect (likely firewall problem)

To fix these errors, please make sure that your domain name was
entered correctly and the DNS A/AAAA record(s) for that domain
contain(s) the right IP address. Additionally, please check that
your computer has a publicly routable IP address and that no
firewalls are preventing the server from communicating with the
client. If you're using the webroot plugin, you should also verify
that you are serving files from the webroot path you provided.
```

ncp-report --> https://pastebin.com/Dw1vGSe6

Based on community feedback on the Nextcloud forum, the LetsEncrypt SSL renewal process fails when nc-httpsonly is enabled. Once nc-httpsonly is disabled, the SSL renewal works successfully.
Link to the Nextcloud forum topic: https://help.nextcloud.com/t/solved-letsencrypt-renewal-error/123525/6

ZendaiOwl, 2021-09-16:
It was me providing the solution on the forums, thought I'd make a comment and show myself here on GitHub as well 🥳

ZendaiOwl, 2021-09-29:
This issue is now #resolved :)
https://help.nextcloud.com/t/solved-letsencrypt-renewal-error/123525/14

I've updated to the development branch and confirmed it myself, no issues whatsoever with renewing the certificate

closed 2021-09-29 by ovpc

nachoparker, 2021-09-29:
Thanks for confirming. Yes, I wrote the redirect rules to allow LE to work, so it should be good now on the latest version