nextcloud/nextcloudpi #1336: HPB service down

Jeanoo, 2021-09-17:
Hello
I saw this today
[image]
Nextcloudpi 1.39.1

theCalcaholic, 2021-09-17:
Can you please connect to your instance over ssh, execute the following command and include its output (make sure to replace all IP addresses and urls with placeholders):

```sh
ncc notify_push:self-test
```

Jeanoo, 2021-09-17:
I have copied the command in Putty, but I get the following error  

`There are no commands defined in the "notify_push" namespace.`

theCalcaholic, 2021-09-17:
Oh, I see. HPB is only relevant for Nextcloud version 21 and higher. You can ignore that error as long as you didn't update your NC version

closed 2021-09-17 by theCalcaholic

nephilim75, 2021-12-03:
One of my customers is facing the same issue. He is running NC 22.2.2 on a RasPi 4 and HPB service is still down.

theCalcaholic, 2021-12-06:
@nephilim75 I have created a new issue, as this is a different bug: #1397

Jeanoo, 2021-12-07:
I am in exactly the same situation 

> One of my customers is facing the same issue. He is running NC 22.2.2 on a RasPi 4 and HPB service is still down.

theCalcaholic, 2021-12-07:
@Jeanoo @nephilim75 Please reboot ncp or run `systemctl restart notify_push`. A fix has been pushed to devel, so this shouldn't occur anymore in the future

Jeanoo, 2021-12-07:
after executing the order, it's always the same HPB service down 

```
per@raspberrypi:~ $ systemctl restart notify_push
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentification requise pour redémarrer « notify_push.service ».
Authenticating as: ,,, (pi)
Password:
==== AUTHENTICATION COMPLETE ===
per@raspberrypi:~ $
```

ovpc, 2021-12-08:
@Jeanoo try again with 
> sudo systemctl restart notify_push

Also please use the [forum](https://help.nextcloud.com/tag/ncp) if you need assistance or have questions.

Jeanoo, 2021-12-08:
thank you for the help 

always the same thing 
HPB service | down

theCalcaholic, 2021-12-08:
@Jeanoo In that case, that's yet a different bug or configuration issue. Please use the forum (as recommended by @ovpc) to check that. If it turns out to be another bug, you can create a new issue and describe the bug exactly (including the output of `ncp-report` and `ncc notify_push:self-test`).

Jeanoo, 2021-12-09:
this is very strange! i updated several apps, then ran sudo systemctl restart notify_push

and the HPB service is up

Jeanoo, 2022-07-14:
I have again problems with HPB service
`
sudo ncp-report`
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.48.1
OS                   Raspbian GNU/Linux 10. 5.10.103-v7l+ (armv7l)
automount            yes
USB devices          sda sdb
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      453G/1.9T
rootfs usage         5.5G/15G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    22.0.0.11
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               2:40

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.1.123",
            "2": "213.55.240.212",
            "3": "site.site-site.com",
            "5": "nextcloudpi.local",
            "6": "nextcloudpi",
            "7": "nextcloudpi.lan",
            "8": "***********",
            "14": "******",
            "11": "************"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "22.0.0.11",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
[... 110 lines cut ...]
```

{"reqId":"Ys-WskYaMZwhgpRZTmYWLAAAABQ","level":0,"time":"2022-07-14T08:41:22+00:00","remoteAddr":"192.168.1.1","user":"--","app":"no app in context","method":"DELETE","url":"/remote.php/webdav/Joplin/locks/1_1_13afc93449414763a72c13ea59d16168.json","message":"Token is not valid: Token does not exist","userAgent":"Joplin/1.0","version":"22.0.0.11","exception":{"Exception":"OC\\Authentication\\Exceptions\\InvalidTokenException","Message":"Token does not exist","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/Authentication/Token/Manager.php","line":146,"function":"getToken","class":"OC\\Authentication\\Token\\DefaultTokenProvider","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":531,"function":"getToken","class":"OC\\Authentication\\Token\\Manager","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":447,"function":"isTokenPassword","class":"OC\\User\\Session","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/apps/dav/lib/Connector/Sabre/Auth.php","line":129,"function":"logClientIn","class":"OC\\User\\Session","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Auth/Backend/AbstractBasic.php","line":103,"function":"validateUserPass","class":"OCA\\DAV\\Connector\\Sabre\\Auth","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/apps/dav/lib/Connector/Sabre/Auth.php","line":252,"function":"check","class":"Sabre\\DAV\\Auth\\Backend\\AbstractBasic","type":"->","args":[{"__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"/var/www/nextcloud/apps/dav/lib/Connector/Sabre/Auth.php","line":154,"function":"auth","class":"OCA\\DAV\\Connector\\Sabre\\Auth","type":"->","args":[{"__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Auth/Plugin.php","line":182,"function":"check","class":"OCA\\DAV\\Connector\\Sabre\\Auth","type":"->","args":[{"__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Auth/Plugin.php","line":137,"function":"check","class":"Sabre\\DAV\\Auth\\Plugin","type":"->","args":[{"__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"/var/www/nextcloud/3rdparty/sabre/event/lib/WildcardEmitterTrait.php","line":89,"function":"beforeMethod","class":"Sabre\\DAV\\Auth\\Plugin","type":"->","args":[{"__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":456,"function":"emit","class":"Sabre\\DAV\\Server","type":"->","args":["beforeMethod:DELETE",[{"__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":253,"function":"invokeMethod","class":"Sabre\\DAV\\Server","type":"->","args":[{"__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":321,"function":"start","class":"Sabre\\DAV\\Server","type":"->","args":[]},{"file":"/var/www/nextcloud/apps/dav/appinfo/v1/webdav.php","line":83,"function":"exec","class":"Sabre\\DAV\\Server","type":"->","args":[]},{"file":"/var/www/nextcloud/remote.php","line":166,"args":["/var/www/nextcloud/apps/dav/ [... cut ...]
{"reqId":"Ys-WyoNAXqUMhem8AVTfTwAAAE0","level":0,"time":"2022-07-14T08:41:46+00:00","remoteAddr":"192.168.1.149","user":"ncp","app":"files_sharing","method":"PROPFIND","url":"/remote.php/dav/files/ncp/Documents/Images","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"Mozilla/5.0 (Windows) mirall/3.4.0stable-Win64 (build 20211129) (Nextcloud, windows-10.0.19043 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"22.0.0.11"}

```
</details>

Jeanoo, 2022-07-14:
thank you theCalcaholic
this time did not correct the problem, it remains unavailable

TheShanMan, 2022-07-14:
Same for me. I did an ncp-dist-upgrade, ran into an unmet dependency error with smbclient, solved it by following the steps mentioned [here](https://help.nextcloud.com/t/nextcloudpi-distro-upgrade-to-bullseye-fails-with-unmet-dependencies-error-solved/142097/2), and reran the ncp-dist-upgrade. I've rebooted a few times and run the systemctl command above prior to the most recent reboot. HPB service is down and the 80 and 443 ports are closed.

theCalcaholic, 2022-07-19:
@Jeanoo Okay, I think I understand the issue. I'll work on a fix tomorrow

Jeanoo, 2022-07-19:
Ok

mstegmaier, 2022-08-30:
I'm also getting `HPB service down`. I'm using nextcloudpi v1.49.0 using docker.
`ncc notify_push:self-test` reports the following
```
redis is configured
can't connect to push server: cURL error 6: Could not resolve host: 335f71e044c0 (see ...) for https://335f71e044c0/push/test/cookie
```
I have no idea where it got that `335f71e044c0` from. Can I fix the push server URL anywhere?

nacho-domatix, 2022-09-26:
It happened to me after one day of fresh install. On LXC.

pfructuoso, 2022-12-17:
I have the same issue. Everytime I checked HPB service was down. It looks like it´s related to `php-json` dependency to `php8.2`. I trid to run `ncc` but the PHP client by command line is 8.2 and I got the message 
`This version of Nextcloud is not compatible with PHP>=8.2.<br/>You are currently running 8.2.0.`
I fixed it setting PHP 8.1 as php client
`sudo update-alternatives --set php /usr/bin/php8.1`
After that everything is good.

I hope this could help others!

ZendaiOwl, 2023-01-16:
Thank you

A fix for this is in the pipeline in the development branch, removing the deprecated `php-json` package, it's now included in the base php package instead and if the `php-json` package is installed `apt update` ends up installing some php 8.2 packages which are incompatible with the nextcloud version

nhat1980, 2023-09-26:
I disable Collabora Online - Built-in CODE Server (ARM64) and Nextcloud Office app then enable again. This problem then resolve