nextcloud/nextcloudpi #1338: Upgrade v1.39.1 on Docker - rollback fails on error

gitbjo, 2021-09-18:
Hi,

the auto-upgrade to nc 21.0.4 failed and left me with stall Nextcloud. 

**Executive Summary**

There were 2 issues:
- letsencrypt script changed my certificate configuration (although it was disabled in configuration) and Apache reload led to an error
- after the error rollback failed due to the location of the backup /var/www whose filesystem is overlayfs in docker. It should have put the backup in /data. -> this is the reason of this issue report.

**Details**

I'm running on Raspberry Pi 4 (4GB) with docker.
Current   Nextcloud version 20.0.12.1
Available Nextcloud version 21.0.4

Here my analysis commenting log (/var/log/ncp.log):
```
...
notify_push 0.2.4 installed  **--> line 200 of ncp-update-nc**
notify_push enabled
notify_push already enabled
/usr/local/etc/ncp-templates/nextcloud.conf.sh: line 21: /usr/local/bin/ncp/SYSTEM/metrics.sh: No such file or directory
/usr/local/etc/ncp-templates/nextcloud.conf.sh: line 22: tmpl_metrics_enabled: command not found
```
--> line 202: for those 2 errors above I have no idea lines do not match with error and it should not have been run as I'm on docker.

--> as we can see below there is an issue with the generated nextcloud.conf file it took the wrong domain.

--> reson for it, I use self generated letsencrypt certificate (DNS method), but it considered active the letsencrypt configuration, but it is not:

--> letsencrypt.cfg has: `"name": "Active", value": "no"`
```
Apache self check:
AH00526: Syntax error on line 11 of /etc/apache2/sites-enabled/nextcloud.conf:
SSLCertificateFile: file '/etc/letsencrypt/live/mycloud.ownyourbits.com/fullchain.pem' does not exist or is empty
Action '-t' failed.
The Apache error log may have more information.
```
--> Apache commands failed as it re-gernerate nextcloud.conf with the wrong domain name in the path.
```
Abort --> rollback is called line 158.
Clean up...
Rolling back to backup /var/www/nextcloud-bkp_20210917_1631854237-20.0.12.1.tar.gz...
{"reqId":"D8J5HwD3WWHHlUizVMyS","level":2,"time":"2021-09-17T07:44:49+02:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","messa
ge":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"21.0.4.1"}
Can only restore from ext/btrfs filesystems
```
--> line 167: ncp-restore fails on check for "ext/btrfs filesystems" as on docker it is: 

--> command "stat -fc%T ncp-restore.loGwXx" returns: overlayfs

`Rollback failed! Data left at /var/www/ncp-data.Ia9BQq`

--> game over !

## Additional findings:

```
stat -fc%T /data
ext2/ext3

root@ad7dda4836ab:/var/www# ls -l /data
total 136940
drwxr-xr-x  3 root     root          4096 Dec 11  2019 backups
drwxr-xr-x  3 root     root          4096 Apr  6  2020 bin
drwxr-xr-x  5 mysql    mysql         4096 Aug 11 18:10 database
drwxr-xr-x  9 root     root          4096 Dec 10  2019 etc
drwxr-xr-x  3 root     www-data      4096 Sep 18 14:18 ncp
drwxr-x--- 14 www-data www-data      4096 Sep 17 07:44 nextcloud
-rw-r--r--  1 root     root     140200496 Aug  3 17:46 nextcloud.tar.bz2
```

--> maybe the upgrade did not run in the right location ? (above we see an old nextcould software archive)

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.39.1
NextCloudPi image    NextCloudPi_docker_03-20-20
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          sda 
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      335G/440G
rootfs usage         335G/440G
swapfile             /mnt/data-ext4/.swapfile
dbdir                /data/database
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               38days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Your data directory is invalid
Ensure there is a file called ".ocdata" in the root of the data directory.

An unhandled exception has been thrown:
Exception: Environment not properly prepared. in /data/nextcloud/lib/private/Console/Application.php:168
Stack trace:
#0 /data/nextcloud/console.php(99): OC\Console\Application->loadCommands(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#1 /data/nextcloud/occ(11): require_once('/data/nextcloud...')
#2 {main}
```
</details>

<details>
<summary>HTTPd logs</summary>

```
[Tue Aug 31 02:29:46.866672 2021] [mpm_event:notice] [pid 94:tid 3069996464] AH00489: Apache/2.4.38 (Debian) OpenSSL/1.1.1d configured -- resuming normal operations
[Tue Aug 31 02:29:46.866779 2021] [core:notice] [pid 94:tid 3069996464] AH00094: Command line: '/usr/sbin/apache2'
[Fri Sep 17 06:50:24.071184 2021] [mpm_event:notice] [pid 94:tid 3069996464] AH00493: SIGUSR1 received.  Doing graceful restart
[Fri Sep 17 06:50:24.118687 2021] [ssl:warn] [pid 94:tid 3069996464] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Fri Sep 17 06:50:24.121107 2021] [mpm_event:notice] [pid 94:tid 3069996464] AH00489: Apache/2.4.38 (Debian) OpenSSL/1.1.1d configured -- resuming normal operations
[Fri Sep 17 06:50:24.121160 2021] [core:notice] [pid 94:tid 3069996464] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

2021-08-11 18:10:29 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2021-08-11 18:10:29 0 [Note] InnoDB: 10.3.29 started; log sequence number 4574519176; transaction id 11703664
2021-08-11 18:10:29 0 [Note] InnoDB: Loading buffer pool(s) from /data/database/ib_buffer_pool
2021-08-11 18:10:29 0 [Note] Plugin 'FEEDBACK' is disabled.
2021-08-11 18:10:29 0 [Note] Server socket created on IP: '127.0.0.1'.
2021-08-11 18:10:29 0 [Note] Reading of all Master_info entries succeeded
2021-08-11 18:10:29 0 [Note] Added new Master_info '' to hash table
2021-08-11 18:10:29 0 [Note] mysqld: ready for connections.
Version: '10.3.29-MariaDB-0+deb10u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Debian 10
2021-08-11 18:10:30 0 [Note] InnoDB: Buffer pool(s) load completed at 210811 18:10:30
2021-08-11 18:10:35 0 [Note] mysqld (initiated by: root[root] @ localhost []): Normal shutdown
2021-08-11 18:10:35 0 [Note] Event Scheduler: Purging the queue. 0 events
2021-08-11 18:10:35 0 [Note] InnoDB: FTS optimize thread exiting.
2021-08-11 18:10:35 0 [Note] InnoDB: Starting shutdown...
2021-08-11 18:10:35 0 [Note] InnoDB: Dumping buffer pool(s) to /data/database/ib_buffer_pool
2021-08-11 18:10:35 0 [Note] InnoDB: Buffer pool(s) dump completed at 210811 18:10:35
2021-08-11 18:10:37 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
2021-08-11 18:10:37 0 [Note] InnoDB: Shutdown completed; log sequence number 4574519185; transaction id 11703667
2021-08-11 18:10:37 0 [Note] mysqld: Shutdown complete

```
</details>

nachoparker, 2021-09-19:
Regarding LE

Indeed, it checks if there's any domain configured (by any method)

https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/NETWORKING/letsencrypt.sh#L16

, so it does the wrong thing for your case. The reason is that if you try to activate it and it fails, then I don't want it to show as "active and running and ok". I can complicate the logic though to account for your case.

nachoparker, 2021-09-19:
pushed the fix for LE

nachoparker, 2021-09-19:
Regarding the metrics error

```
if [[ "$DOCKERBUILD" != 1 ]] && [[ "$1" != "--defaults" ]]; then
  METRICS_IS_ENABLED="$(
  source "${BINDIR}/SYSTEM/metrics.sh"
  tmpl_metrics_enabled && echo yes || echo no
  )"
else                                                                                                                                                                                                                                                                                                                           
  METRICS_IS_ENABLED=no
fi
```

This doesn't make sense, and I surely tested this in docker before release. Do you not have a `$DOCKERBUILD` variable inside your docker container?

nachoparker, 2021-09-19:
Regarding

```
--> line 167: ncp-restore fails on check for "ext/btrfs filesystems" as on docker it is:

--> command "stat -fc%T ncp-restore.loGwXx" returns: overlayfs
```

The command that's failing is 

```
ncp-restore /var/www/nextcloud-bkp_20210917_1631854237-20.0.12.1.tar.gz
```

So you are saying that

```
"stat -fc%T /var/www/ncp-restore.loGwXx
```
Returns `overlayfs`?

nachoparker, 2021-09-19:
I did a quick test

```
docker exec -ti $( docker ps -q | head -1 ) bash
root@9d7a61a4516b:/# stat -fc%T /
btrfs
root@9d7a61a4516b:/# stat -fc%T /var/www
btrfs
root@9d7a61a4516b:/# mktemp -d /var/www/ncp-restore.XXXXXX
/var/www/ncp-restore.DXpjg3
root@9d7a61a4516b:/# stat -fc%T /var/www/ncp-restore.DXpjg3/
btrfs
root@9d7a61a4516b:/#
```

gitbjo, 2021-09-19:
> I did a quick test

root@ad7dda4836ab:/# stat -fc%T / /data /var/www
overlayfs
ext2/ext3
overlayfs

> 
> What kernel are you running?

root@ad7dda4836ab:/# uname -a
Linux ad7dda4836ab 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux

I suppose if `BKPDIR` would be equal to `BASEDIR` there would be less risk to have overlayfs, may be as second choice if the first returns not the expected one. Or accept overlayfs, as if the file is there, why not restore it?

gitbjo, 2021-09-19:
> Regarding the metrics error
>
> ```
> if [[ "$DOCKERBUILD" != 1 ]] && [[ "$1" != "--defaults" ]]; then
>   METRICS_IS_ENABLED="$(
>   source "${BINDIR}/SYSTEM/metrics.sh"
>   tmpl_metrics_enabled && echo yes [...]

I agree, I cannot understand neither and also the line numbers do not match (even in version -1 of the file).

If it happens again, I will open an issue for that (it did not trigger the rollback).

nachoparker, 2021-09-19:
Can you run inside your container the following command?

```
echo $DOCKERBUILD
```
?

Also,

```
root@ad7dda4836ab:/# stat -fc%T / /data /var/www
overlayfs
ext2/ext3
overlayfs
```

This is very interesting. I think it's easy to add support for this output so things work for this case. Never in all these years have I seen this error though, so there must be a behavior change with overlayfs interacting with stat

gitbjo, 2021-09-20:
Hi,
just to be complete:
```
root@ad7dda4836ab:/# echo $DOCKERBUILD
1
```