nextcloud/nextcloudpi #1339: nextcloud update to 21.0.4 error

Jeanoo, 2021-09-19:
Hello, I tried to update nextcloudpi to 21.0.4 and I get this error 

```
[ nc-update-nextcloud ] (Sun Sep 19 15:44:30 CEST 2021)
Current Nextcloud version 20.0.12.1
Available Nextcloud version 21.0.4
Download Nextcloud 21.0.4...
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup files...
backup /var/www//nextcloud-bkp_20210919_1632059080.tar.gz generated
Maintenance mode disabled
Install Nextcloud 21.0.4...
Fix permissions...
Upgrade...
Nextcloud is not installed - only a limited number of commands are available

Command "upgrade" is not defined.

Abort
Clean up...
```

Jeanoo, 2021-09-19:
a solution has been found here
https://github.com/nextcloud-snap/nextcloud-snap/issues/1492 
Maybe I am in the same case?

nachoparker, 2021-09-19:
I suspect something is down for you, database, redis or similar.

Can you share the output of `sudo ncp-report`? ( as requested when you created the bug report )

Jeanoo, 2021-09-20:
thank you for your help 

```
sudo ncp-report
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.39.6
distribution         Raspbian GNU/Linux 10 \n \l
automount            yes
USB devices          sda sdb
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      427G/1.9T
rootfs usage         5.2G/15G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    20.0.12.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               11:47

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "11": "*********",
            "1": "*********",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "20": "********",
            "21": "********",
            "12": "**********",
            "3": "*********"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "20.0.12.1",
        "overwrite.cli.url": "https:\/\/***********\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
[... 76 lines cut ...]
{"reqId":"LBBbhW6p8Y2lE7FLALRS","level":0,"time":"2021-09-20T07:27:17+00:00","remoteAddr":"","user":"--","app":"extract","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"20.0.12.1"}
{"reqId":"LBBbhW6p8Y2lE7FLALRS","level":0,"time":"2021-09-20T07:27:17+00:00","remoteAddr":"","user":"--","app":"files_sharing","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"20.0.12.1"}
{"reqId":"LBBbhW6p8Y2lE7FLALRS","level":0,"time":"2021-09-20T07:27:17+00:00","remoteAddr":"","user":"--","app":"guests","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"20.0.12.1"}
{"reqId":"LBBbhW6p8Y2lE7FLALRS","level":3,"time":"2021-09-20T07:27:17+00:00","remoteAddr":"","user":"--","app":"passman","method":"","url":"--","message":"/appinfo/app.php is not loaded when \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class is used. Migrate everything from app.php to the Application class.","userAgent":"--","version":"20.0.12.1"}
{"reqId":"LBBbhW6p8Y2lE7FLALRS","level":0,"time":"2021-09-20T07:27:17+00:00","remoteAddr":"","user":"--","app":"pdfdraw","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"20.0.12.1"}
{"reqId":"LBBbhW6p8Y2lE7FLALRS","level":0,"time":"2021-09-20T07:27:17+00:00","remoteAddr":"","user":"--","app":"quicknotes","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"20.0.12.1"}
{"reqId":"LBBbhW6p8Y2lE7FLALRS","level":0,"time":"2021-09-20T07:27:17+00:00","remoteAddr":"","user":"--","app":"scanner","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"20.0.12.1"}
{"reqId":"LBBbhW6p8Y2lE7FLALRS","level":0,"time":"2021-09-20T07:27:17+00:00","remoteAddr":"","user":"--","app":"talk_simple_poll","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"20.0.12.1"}
{"reqId":"LBBbhW6p8Y2lE7FLALRS","level":0,"time":"2021-09-20T07:27:17+00:00","remoteAddr":"","user":"--","app":"whiteboard","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"20.0.12.1"}
{"reqId":"BvnPjrbwYUM4zdJbf0Xt","level":0,"time":"2021-09-20T07:27:24+00:00","remoteAddr":"","user":"--","app":"appointments","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"20.0.12.1"}
{"reqId":"BvnPjrbwYUM4zdJbf0Xt","level":0,"time":"2021-09-20T07:27:24+00:00","remoteAddr":"","user":"--","app":"groupfolders","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"20.0.12.1"}
[... cut ...]
{"reqId":"BvnPjrbwYUM4zdJbf0Xt","level":0,"time":"2021-09-20T07:27:24+00:00","remoteAddr":"","user":"--","app":"talk_simple_poll","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"20.0.12.1"}
{"reqId":"BvnPjrbwYUM4zdJbf0Xt","level":0,"time":"2021-09-20T07:27:24+00:00","remoteAddr":"","user":"--","app":"whiteboard","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"20.0.12.1"}

```
</details>

```

nachoparker, 2021-09-21:
So your NC works fine, and when you try to upgrade you see that?

Jeanoo, 2021-09-21:
yes, everything works fine, unless I update it

nachoparker, 2021-09-21:
I don't see anything unique, and I know this works for everybody else, can you please add `set -x` as a second line in `/usr/local/bin/ncp-update-nc`, then try to update with `sudo ncp-update-nc 21.0.4` and share the output?

Jeanoo, 2021-09-21:
Thank you for your help 
I can't do it right away, as I use Nextcloud in my job, I have to wait until I have at least two days off.
Then, I prefer to do the update with the nextcloudpi interface, because if there is a problem the restoration is done by itself.

Jeanoo, 2021-09-22:
Ok, Thanks!

nachoparker, 2021-09-22:
right after `#!/bin/bash` :)

githubtefo, 2021-09-25:
I am having the exact same issue here, updating from the same two versions.
My nc-diagnostics looks fine as well.
I am updating using the NextCloudPi Panel, do you think that make a difference?

nachoparker, 2021-10-04:
It shouldn't make a difference. Waiting for your debug output to dig in

closed 2021-10-04 by nachoparker

githubtefo, 2021-10-09:
No need to; it has been solved with [https://github.com/nextcloud/nextcloudpi/commit/7c361c5].
I can update to nc v21 without errors. Thanks!