nextcloud/nextcloudpi #1340: nc-previews crashing since upgrading to nc 21 via nextcloudpi panel

dsmic, 2021-09-20:
Here is a manual execution ...

```
sudo nc-previews 
2021-09-20T07:04:56+00:00 
2021-09-20T07:04:56+00:00 In ExceptionConverter.php line 65:
2021-09-20T07:04:56+00:00                                                                                
2021-09-20T07:04:56+00:00   [Doctrine\DBAL\Exception\InvalidFieldNameException (1054)]                   
2021-09-20T07:04:56+00:00   An exception occurred while executing a query: SQLSTATE[42S22]: Column not   
2021-09-20T07:04:56+00:00   found: 1054 Unknown column 'locked' in 'where clause'                        
2021-09-20T07:04:56+00:00                                                                                
2021-09-20T07:04:56+00:00 
2021-09-20T07:04:56+00:00 Exception trace:
2021-09-20T07:04:56+00:00   at /var/www/nextcloud/3rdparty/doctrine/dbal/src/Driver/API/MySQL/ExceptionConverter.php:65
2021-09-20T07:04:56+00:00  Doctrine\DBAL\Driver\API\MySQL\ExceptionConverter->convert() at /var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php:1728
2021-09-20T07:04:56+00:00  Doctrine\DBAL\Connection->handleDriverException() at /var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php:1667
2021-09-20T07:04:56+00:00  Doctrine\DBAL\Connection->convertExceptionDuringQuery() at /var/www/nextcloud/3rdparty/doctrine/dbal/src/Connection.php:1040
2021-09-20T07:04:56+00:00  Doctrine\DBAL\Connection->executeQuery() at /var/www/nextcloud/lib/private/DB/Connection.php:226
2021-09-20T07:04:56+00:00  OC\DB\Connection->executeQuery() at /var/www/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php:210
2021-09-20T07:04:56+00:00  Doctrine\DBAL\Query\QueryBuilder->execute() at /var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php:287
2021-09-20T07:04:56+00:00  OC\DB\QueryBuilder\QueryBuilder->execute() at /var/www/ncp-previewgenerator/ncp-previewgenerator-nc21/lib/Command/PreGenerate.php:139
2021-09-20T07:04:56+00:00  OCA\PreviewGenerator\Command\PreGenerate->startProcessing() at /var/www/ncp-previewgenerator/ncp-previewgenerator-nc21/lib/Command/PreGenerate.php:124
2021-09-20T07:04:56+00:00  OCA\PreviewGenerator\Command\PreGenerate->execute() at /var/www/nextcloud/3rdparty/symfony/console/Command/Command.php:255
2021-09-20T07:04:56+00:00  Symfony\Component\Console\Command\Command->run() at /var/www/nextcloud/3rdparty/symfony/console/Application.php:1009
2021-09-20T07:04:56+00:00  Symfony\Component\Console\Application->doRunCommand() at /var/www/nextcloud/3rdparty/symfony/console/Application.php:273
2021-09-20T07:04:56+00:00  Symfony\Component\Console\Application->doRun() at /var/www/nextcloud/3rdparty/symfony/console/Application.php:149
2021-09-20T07:04:56+00:00  Symfony\Component\Console\Application->run() at /var/www/nextcloud/lib/private/Console/Application.php:215
2021-09-20T07:04:56+00:00  OC\Console\Application->run() at /var/www/nextcloud/console.php:100
2021-09-20T07:04:56+00:00  require_once() at /var/www/nextcloud/occ:11
2021-09-20T07:04:56+00:00 
2021-09-20T07:04:56+00:00 In Exception.php line 26:
2021-09-20T07:04:56+00:00                                                                                
2021-09-20T07:04:56+00:00   [Doctrine\DBAL\Driver\PDO\Exception (1054)]                                  
2021-09-20T07:04:56+00:00   SQLSTATE[42S22]: Column not found: 1054 Unknown column 'locked' in 'where c  
2021-09-20T07:04:56+00:00   lause'                              
```

nachoparker, 2021-09-21:
did you at any point install the official previewgenerator app in the past?

dsmic, 2021-09-22:
Yes I had it installed, after removing I get:

```
pi@raspberrypi:~ $ sudo nc-previews 

In Application.php line 676:
                                                                  
  [Symfony\Component\Console\Exception\CommandNotFoundException]  
  Command "preview:pre-generate" is not defined.                  
                                                                  
  Did you mean one of these?                                      
      migrations:generate                                         
      notification:generate                                       
      preview:repair                                              
      preview:reset-rendered-texts                                
                                                                  

Exception trace:
  at /var/www/nextcloud/3rdparty/symfony/console/Application.php:676
 Symfony\Component\Console\Application->find() at /var/www/nextcloud/3rdparty/symfony/console/Application.php:237
 Symfony\Component\Console\Application->doRun() at /var/www/nextcloud/3rdparty/symfony/console/Application.php:149
 Symfony\Component\Console\Application->run() at /var/www/nextcloud/lib/private/Console/Application.php:215
 OC\Console\Application->run() at /var/www/nextcloud/console.php:100
 require_once() at /var/www/nextcloud/occ:11

```

nachoparker, 2021-09-22:
yes, you have a bit of a frankenstein there... you can re-install the official one and it should work, but you lose the performance benefits

themorlan, 2021-09-22:
Hi, 
I also get this error after upgrading to nc 21 via the NCP-Panel yesterday (Docker install):

`In Application.php line 676:
                                                                  
  [Symfony\Component\Console\Exception\CommandNotFoundException]  
  Command "preview:pre-generate" is not defined.                  
                                                                  
  Did you mean one of these?                                      
      migrations:generate                                         
      notification:generate                                       
      preview:repair                                              
      preview:reset-rendered-texts                                
                                                                  

Exception trace:
  at /data/nextcloud/3rdparty/symfony/console/Application.php:676
 Symfony\Component\Console\Application->find() at /data/nextcloud/3rdparty/symfony/console/Application.php:237
 Symfony\Component\Console\Application->doRun() at /data/nextcloud/3rdparty/symfony/console/Application.php:149
 Symfony\Component\Console\Application->run() at /data/nextcloud/lib/private/Console/Application.php:215
 OC\Console\Application->run() at /data/nextcloud/console.php:100
 require_once() at /data/nextcloud/occ:11
`

The apps - panel in nc shows Preview Generator 3.1.1 as installed but inactive.

dsmic, 2021-09-22:
> yes, you have a bit of a frankenstein there... you can re-install the official one and it should work, but you lose the performance benefits

Thanks, I did this, I had to remove the sym link in the aps folder to ncp preciew by hand before installing…

closed 2021-09-22 by nachoparker