nextcloud/nextcloudpi #1341: Erroneous no space left on device error

Iolaum, 2021-09-20:
# Error Information.

I was updating ncp while this happened. Logs are a better illustrator.

```
# manually ssh in to ncp
NextCloudPi v1.39.3 is outdated
update to v1.39.6 through 'ncp-config' or type 'sudo ncp-update'
Last login: Mon Sep 20 23:12:48 2021 from $some_ip
$ sudo ncp-update
Downloading updates
Performing updates
Installing nc-restore
Running nc-autoupdate-nc
/usr/local/etc/library.sh: line 209: echo: write error: No space left on device
$ df -Th
Filesystem     Type      Size  Used Avail Use% Mounted on
/dev/root      ext4       15G  2.6G   12G  19% /
devtmpfs       devtmpfs  430M     0  430M   0% /dev
tmpfs          tmpfs     462M     0  462M   0% /dev/shm
tmpfs          tmpfs     462M   13M  450M   3% /run
tmpfs          tmpfs     5.0M  4.0K  5.0M   1% /run/lock
tmpfs          tmpfs     462M     0  462M   0% /sys/fs/cgroup
/dev/mmcblk0p1 vfat       44M   25M   19M  58% /boot
log2ram        tmpfs      40M   40M     0 100% /var/log
/dev/sda       ext4      916G  300G  571G  35% /media/ncphd
tmpfs          tmpfs      93M     0   93M   0% /run/user/1000

```

Looking at [line 209 of library.sh](https://github.com/nextcloud/nextcloudpi/blob/f3e3b01ab54ff6b8827683f451323fdb8d9ce5cf/etc/library.sh#L209) we see:

```
echo "[ $ncp_app ] ($(date))" >> $log
```

and coupling that with what the line does it appears this error was because of log2ram being full ! :?

This is the first time I've seen that happen, not sure what to do about it.

**System information**

Iolaum, 2021-09-20:
I rebooted and re-run ncp-update and didn't get the error again. Maybe this was a weird one-off ?

Iolaum, 2021-09-20:
Upon further investigation I noticed that on further logins there was no log2ram entry:

```
$ df -Th
Filesystem     Type      Size  Used Avail Use% Mounted on
/dev/root      ext4       15G  2.6G   12G  19% /
devtmpfs       devtmpfs  430M     0  430M   0% /dev
tmpfs          tmpfs     462M     0  462M   0% /dev/shm
tmpfs          tmpfs     462M  6.3M  456M   2% /run
tmpfs          tmpfs     5.0M  4.0K  5.0M   1% /run/lock
tmpfs          tmpfs     462M     0  462M   0% /sys/fs/cgroup
/dev/mmcblk0p1 vfat       44M   25M   19M  58% /boot
/dev/sda       ext4      916G  300G  571G  35% /media/ncphd
tmpfs          tmpfs      93M     0   93M   0% /run/user/1000
```

I tried to re-enable `nc-ramlogs` and the process failed:

```
Running nc-ramlogs
Job for log2ram.service failed because the control process exited with error code.
See "systemctl status log2ram.service" and "journalctl -xe" for details.
Logs in RAM. Reboot to take effect
# Let's try to look for the error:
$ systemctl status log2ram.service
● log2ram.service - Log2Ram
   Loaded: loaded (/etc/systemd/system/log2ram.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2021-09-20 23:50:46 EEST; 22s ago
  Process: 2177 ExecStart=/usr/local/bin/log2ram start (code=exited, status=1/FAILURE)
 Main PID: 2177 (code=exited, status=1/FAILURE)

Sep 20 23:50:46 nextcloud systemd[1]: Starting Log2Ram...
Sep 20 23:50:46 nextcloud log2ram[2177]: ERROR: RAM disk too small. Can't sync.
Sep 20 23:50:46 nextcloud log2ram[2177]: /usr/local/bin/log2ram: 35: /usr/local/bin/log2ram: mail: not found
Sep 20 23:50:46 nextcloud log2ram[2177]: /usr/local/bin/log2ram: 35: echo: echo: I/O error
Sep 20 23:50:46 nextcloud systemd[1]: log2ram.service: Main process exited, code=exited, status=1/FAILURE
Sep 20 23:50:46 nextcloud systemd[1]: log2ram.service: Failed with result 'exit-code'.
Sep 20 23:50:46 nextcloud systemd[1]: Failed to start Log2Ram.
```

It appears that there's an error about RAM disk being to small - not sure what may have happened here.

Could it be that `/var/log/` is too crowded?

```
$ pwd
/var/log
$ sudo du -h .
4.0K	./prometheus
12K	./wicd
256K	./apt
4.0K	./samba
768K	./redis
72K	./letsencrypt
204K	./unattended-upgrades
1.3M	./apache2
52K	./mysql
4.0K	./private
42M	.
```

theCalcaholic, 2021-09-21:
Thank you for all the details. 40M is the default size of log2ram's RAM disk (configured in /etc/log2ram.conf).
I'm not sure if we should act on this or if this is a scenario we normally shouldn't run into.

@nachoparker is probably more familiar with the common log2ram usage

nachoparker, 2021-09-21:
thanks for the info. We have logrotate settings in place to prevent this kind of thing, but I guess that it is possible to still flood the logs if there's some error that spams too frequently.

nachoparker, 2021-10-04:
these are logrotate settings

```
 /var/log/ncp.log                      
  { 
          rotate 4                                                              
          size 500K                                                                                                                                            
          missingok                    
          notifempty                                                           
          compress                      
  }
```

size is 500K, not sure if a very chatty log would defeat this setting in logrotate

nachoparker, 2021-10-04:
It seems like logrotate only check sizes once  day, which would be insufficient if there is a very spammy log (see `/etc/cron.daily/`)

I guess that if this truly becomes an issue we could move that file to `/etc/cron.hourly`

closed 2021-10-04 by nachoparker

Iolaum, 2021-11-03:
I got the same error again.

```
...
NextCloudPi v1.42.5 is outdated
update to v1.43.3 through 'ncp-config' or type 'sudo ncp-update'
...
$ sudo ncp-update
Downloading updates
Performing updates
Installing nc-encrypt
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:3 http://archive.raspberrypi.org/debian buster/main armhf Packages [393 kB]
Fetched 441 kB in 2s (257 kB/s)   
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  gocryptfs
0 upgraded, 1 newly installed, 0 to remove and 10 not upgraded.
Need to get 1,167 kB of archives.
After this operation, 5,167 kB of additional disk space will be used.
Get:1 http://raspbian.mirror.constant.com/raspbian buster/main armhf gocryptfs armhf 1.6.1-1 [1,167 kB]
Fetched 1,167 kB in 2s (752 kB/s)    
Selecting previously unselected package gocryptfs.
(Reading database ... 53954 files and directories currently installed.)
Preparing to unpack .../gocryptfs_1.6.1-1_armhf.deb ...
Unpacking gocryptfs (1.6.1-1) ...
Setting up gocryptfs (1.6.1-1) ...
Processing triggers for man-db (2.8.5-2) ...
E: Write error - ~LZMAFILE (28: No space left on device)

$ sudo du -h /var/log/
296K	/var/log/unattended-upgrades
8.0K	/var/log/wicd
1.2M	/var/log/redis
16K	/var/log/mysql
33M	/var/log/apache2
0	/var/log/private
0	/var/log/prometheus
92K	/var/log/letsencrypt
0	/var/log/samba
312K	/var/log/apt
40M	/var/log/
```

Tried to manually trigger the logrotate function but it didn't do anything

```
$ bash /etc/cron.daily/logrotate 
$ sudo bash /etc/cron.daily/logrotate 
$ sudo du -h /var/log/
296K	/var/log/unattended-upgrades
8.0K	/var/log/wicd
1.2M	/var/log/redis
16K	/var/log/mysql
33M	/var/log/apache2
0	/var/log/private
0	/var/log/prometheus
92K	/var/log/letsencrypt
0	/var/log/samba
312K	/var/log/apt
40M	/var/log/
```

Doing some deeper digging it looks like the log dirs getting full contributes to the error even more:

```
$ head /var/log/apache2/nc-error.log
[Sat Oct 30 00:00:07.019825 2021] [ssl:warn] [pid 1175:tid 1995739712] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Sat Oct 30 00:00:07.020490 2021] [ssl:error] [pid 1175:tid 1995739712] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=archlinux / issuer: CN=archlinux / serial: FBCE4F86FB5AAA58 / notbefore: Mar  9 16:17:33 2019 GMT / notafter: Mar  6 16:17:33 2029 GMT]
[Sat Oct 30 00:00:07.020538 2021] [ssl:error] [pid 1175:tid 1995739712] AH02604: Unable to configure certificate localhost:443:0 for stapling
[Sat Oct 30 06:28:01.574330 2021] [log_config:warn] [pid 21084:tid 1598026752] (28)No space left on device: [client 192.168.1.103:41426] AH00646: Error writing to /var/log/apache2/nc-access.log
[Sat Oct 30 06:28:01.741160 2021] [log_config:warn] [pid 21084:tid 1396601856] (28)No space left on device: [client 192.168.1.103:41428] AH00646: Error writing to /var/log/apache2/nc-access.log
[Sat Oct 30 06:28:01.839005 2021] [log_config:warn] [pid 21083:tid 1530885120] (28)No space left on device: [client 192.168.1.103:41432] AH00646: Error writing to /var/log/apache2/nc-access.log
[Sat Oct 30 06:28:02.857331 2021] [log_config:warn] [pid 21083:tid 1428141056] (28)No space left on device: [client 192.168.1.103:41438] AH00646: Error writing to /var/log/apache2/nc-access.log
[Sat Oct 30 06:28:03.061164 2021] [log_config:warn] [pid 21084:tid 1404994560] (28)No space left on device: [client 192.168.1.103:41434] AH00646: Error writing to /var/log/apache2/nc-access.log
[Sat Oct 30 06:28:03.106922 2021] [log_config:warn] [pid 21084:tid 1522492416] (28)No space left on device: [client 192.168.1.103:41440] AH00646: Error writing to /var/log/apache2/nc-access.log
[Sat Oct 30 06:28:03.474979 2021] [log_config:warn] [pid 21083:tid 1419748352] (28)No space left on device: [client 192.168.1.103:41436] AH00646: Error writing to /var/log/apache2/nc-access.log

```

After looking a little bit deeper I think I have found cause the problem:

```
$ ls -lah /var/log/apache2/
...
-rw-r-----  1 root adm   15M Nov  3 19:57 nc-access.log
-rw-r-----  1 root adm   17M Oct 30 00:02 nc-access.log.1

```

Those access logs are huge, and I am guessing this happened because I synced my backed up data to a new laptop.
I m guessing this can be a way to create huge logs and test this.

I' m going to manually clean the log directory.

After doing so the update proceeded normally.

nachoparker, 2021-11-04:
and what's spamming your `nc-access.log` files? maybe we need to logrotate them as well

Iolaum, 2021-11-05:
I didn't keep the logs. But those huge logs coincided with me syncing my data to a new laptop from NCP (I noticed them after 2 days). I m guessing the sever logs are a line per transaction (ie file sync/download).

I 'd be in favor of logrotating those files as well.