nextcloud/nextcloudpi #1344: Portcheck 80 443 DSlite

UN-pi, 2021-09-23:
**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.39.12
distribution         Debian GNU/Linux 10 \n \l

automount            no
USB devices          sda sdb
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      919G/1,8T
rootfs usage         4,9G/29G
swapfile             /dev/zram3
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    21.0.4.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               18:14
<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.222.100",
            "2": " "***REMOVED SENSITIVE VALUE***",
            "5": "nextcloudpi.fritz.box",
            "7": "nextcloudpi",
            "9":  "***REMOVED SENSITIVE VALUE***",
            "12": " "***REMOVED SENSITIVE VALUE***",
            "13": " "***REMOVED SENSITIVE VALUE***",
            "11": " "***REMOVED SENSITIVE VALUE***",
            "3": " "***REMOVED SENSITIVE VALUE***"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "21.0.4.1",
        "overwrite.cli.url": "https:\/\***REMOVED SENSITIVE VALUE***\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "mail_smtpmode": "smtp",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "preview_max_x": "1024",
        "preview_max_y": "1024",
        "jpeg_quality": "60",
        "overwriteprotocol": "https",
        "maintenance": false,
        "tempdirectory": "\/media\/USBstick\/nctmp",
        "logfile": "\/media\/USBdrive\/ncdata\/nextcloud.log",
        "loglevel": "2",
        "log_type": "file",
        "mail_sendmailmode": "smtp",
        "theme": "",
        "htaccess.RewriteBase": "\/",
        "auth.bruteforce.protection.enabled": true,
        "versions_retention_obligation": "auto, 7",
        "trashbin_retention_obligation": "auto, 7",
        "app_install_overwrite": [
            "previewgenerator"
        ],
        "enable_previews": true,
        "enabledPreviewProviders": [
            "OC\\Preview\\Movie",
            "OC\\Preview\\PNG",
            "OC\\Preview\\JPEG",
            "OC\\Preview\\GIF",
            "OC\\Preview\\BMP",
            "OC\\Preview\\XBitmap",
            "OC\\Preview\\MP3",
            "OC\\Preview\\MP4",
            "OC\\Preview\\TXT",
            "OC\\Preview\\MarkDown"
        ],
        "mail_smtpauth": 1,
        "mail_smtphost": "***REMOVED SENSITIVE VALUE***",
        "mail_smtpport": "465",
        "mail_smtpname": "***REMOVED SENSITIVE VALUE***",
        "mail_smtppassword": "***REMOVED SENSITIVE VALUE***",
        "mail_smtpsecure": "ssl",
        "data-fingerprint": "a10484b97892ec37f4acba47db21be51",
        "default_phone_region": "DE"
    }
}

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Thu Sep 23 00:00:09.606996 2021] [ssl:warn] [pid 3582:tid 547779081600] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Thu Sep 23 00:00:09.608403 2021] [mpm_event:notice] [pid 3582:tid 547779081600] AH00489: Apache/2.4.38 (Debian) OpenSSL/1.1.1d configured -- resuming normal operations
[Thu Sep 23 00:00:09.608435 2021] [core:notice] [pid 3582:tid 547779081600] AH00094: Command line: '/usr/sbin/apache2'
[Thu Sep 23 07:34:03.767044 2021] [proxy_fcgi:error] [pid 32717:tid 547682156928] [client 192.168.222.20:49997] AH01071: Got error 'PHP message: PHP Notice:  Undefined offset: 1 in /var/www/ncp-web/ncp-launcher.php on line 109PHP message: PHP Notice:  Undefined offset: 1 in /var/www/ncp-web/ncp-launcher.php on line 110PHP message: PHP Notice:  Undefined offset: 1 in /var/www/ncp-web/ncp-launcher.php on line 111PHP message: PHP Notice:  Undefined offset: 1 in /var/www/ncp-web/ncp-launcher.php on line 113PHP message: PHP Notice:  Undefined offset: 1 in /var/www/ncp-web/ncp-launcher.php on line 114PHP message: PHP Notice:  Undefined offset: 1 in /var/www/ncp-web/ncp-launcher.php on line 115PHP message: PHP Notice:  Undefined offset: 1 in /var/www/ncp-web/ncp-launcher.php on line 118PHP message: PHP Notice:  Undefined offset: 1 in /var/www/ncp-web/ncp-launcher.php on line 109PHP message: PHP Notice:  Undefined offset: 1 in /var/www/ncp-web/ncp-launcher.php on line 110PHP message: PHP Notice:  Undefined offset: 1 in /var/www/ncp-web/ncp-launcher.php on line 111PHP message: PHP Notice:  Undefined offset: 1 in /var/www/ncp-web/ncp-launcher.php on line 113PHP message: PHP Notice:  Undefined offset: 1 in /var/www/ncp-web/ncp-launcher.php on line 114PHP message: PHP Notice:  Undefined offset: 1 in /var/www/ncp-web/ncp-launcher.php on line 115PHP message: PHP Notice:  Undefined offset: 1 in /var/www/ncp-web/ncp-launcher.php on line 118'

```
</details>

<details>
<summary>Database logs</summary>

```

```
</details>

<details>
<summary>Nextcloud logs</summary>

```
</details>

nachoparker, 2021-09-23:
Thanks for reporting. We can improve this if you help out with testing, since you have this setup :)

UN-pi, 2021-09-23:
Naturally. I like to help when I can. I would then need a little help with testing

UN-pi, 2021-10-15:
current : ncp 1.40.5 de - I put all commands in a bash-file : 

+ is_port_open 80
+ local port=80
+ local public_ip
++ curl -m4 -4 https://icanhazip.com
+ public_ip=94.31.85.185
++ mktemp
+ local tmp_file=/tmp/user/0/tmp.CEaXH8a3jO
++ wget -T2 -t1 -qO- --keep-session-cookies --save-cookies /tmp/user/0/tmp.CEaXH8a3jO https://portchecker.co
++ grep -oP '_csrf" value="\K.*"'
+ local 'token=EFBelgoZiZnTbWVP-d8UG5SwY7QSatctZth6hVcT1jU"'
+ [[ EFBelgoZiZnTbWVP-d8UG5SwY7QSatctZth6hVcT1jU" != '' ]]
+ wget -T2 -t1 -qO- --load-cookies /tmp/user/0/tmp.CEaXH8a3jO https://portchecker.co --post-data 'target_ip=94.31.85.185&port=80&_csrf=EFBelgoZiZnTbWVP-d8UG5SwY7QSatctZth6hVcT1jU'
+ grep -q '<span class="green">open</span>'
+ rm /tmp/user/0/tmp.CEaXH8a3jO
+ echo closed
closed
+ is_port_open 8484
+ local port=8484
+ local public_ip
++ curl -m4 -4 https://icanhazip.com
+ public_ip=94.31.85.185
++ mktemp
+ local tmp_file=/tmp/user/0/tmp.FOK2h7Jq2U
++ wget -T2 -t1 -qO- --keep-session-cookies --save-cookies /tmp/user/0/tmp.FOK2h7Jq2U https://portchecker.co
++ grep -oP '_csrf" value="\K.*"'
+ local 'token=Z0e-nNeNNnCqehAjQQQrKwNekMIwhApqwmEK2_slcdI"'
+ [[ Z0e-nNeNNnCqehAjQQQrKwNekMIwhApqwmEK2_slcdI" != '' ]]
+ wget -T2 -t1 -qO- --load-cookies /tmp/user/0/tmp.FOK2h7Jq2U https://portchecker.co --post-data 'target_ip=94.31.85.185&port=8484&_csrf=Z0e-nNeNNnCqehAjQQQrKwNekMIwhApqwmEK2_slcdI'
+ grep -q '<span class="green">open</span>'
+ rm /tmp/user/0/tmp.FOK2h7Jq2U
+ echo closed
closed

The webpage:4443 :

[image]

nachoparker, 2021-10-15:
thank you, let's do something else so I can see the complete output

```
sudo su
ncp-update devel
source /usr/local/etc/library.sh 
set -x
is_port_open 80
is_port_open 8484
```

nachoparker, 2021-10-15:
thanks, interesting. We added the `-4` for dual stack. How about just removing the parameter altogether? does that work?

UN-pi, 2021-10-16:
yes.
curl -m4 -4 https://icanhazip.com 
-> returns the unreachable shared DSlite-IPv4-Adress of my pi
curl -m4 -6 https://icanhazip.com 
-> returns the reachable IPv6-Adress of my pi
curl -m4 https://icanhazip.com 
-> returns the reachable IPv6-Adress of my pi
[image]

nachoparker, 2021-10-18:
well, if the user is behind CGNAT there is no chance to be reached from the outside, so I think this is good enough.

Many thanks for the help!