nextcloud/nextcloudpi #1345: Upgrading to NC21 on docker fails to add docker IP to trusted proxies - HPB cannot be enabled

svenb1234, 2021-09-23:
Updating from within docker throws the following error. Full log below.

✓ redis is configured
✓ push server is receiving redis messages
✓ push server can load mount info from database
🗴 push server can't connect to the Nextcloud server
  Response from nextcloud is not a number: invalid digit found in string
notify_push setup failed. You are probably behind a proxy
Run 'ncc config:system:set trusted_proxies 15 --value=<proxy_IP>' and then 'ncc notify_push:setup https://<domain>/push to enable
Check https://help.nextcloud.com/tags/ncp for support

Issueing the commands results in:
root@nextcloudpi:/# ncc config:system:set trusted_proxies 15 --value=172.17.0.3
System config value trusted_proxies => 15 set to string 172.17.0.3
root@nextcloudpi:/# ncc notify_push:setup https://mydomain.eu/push
✓ redis is configured
✓ push server is receiving redis messages
✓ push server can load mount info from database
✓ push server can connect to the Nextcloud server
✓ push server is a trusted proxy
✓ push server is running the same version as the app
  configuration saved

root@nextcloudpi:/# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
7: eth0@if8: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default 
    link/ether 02:42:ac:11:00:03 brd ff:ff:ff:ff:ff:ff link-netnsid 0
    inet 172.17.0.3/16 brd 172.17.255.255 scope global eth0
       valid_lft forever preferred_lft forever

That IP could have been retrieved from the container.

Full update log. I wondered a bit about all the disabled apps, but the rest seems ok.

root@nextcloudpi:/# ncp-update-nc 21.0.4
Current   Nextcloud version 20.0.12.1
Available Nextcloud version 21.0.4
Download Nextcloud 21.0.4...
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup files...
backup /var/www//nextcloud-bkp_20210923_1632421521.tar.gz generated
Maintenance mode disabled
Install Nextcloud 21.0.4...
Restore data...
Fix permissions...
Upgrade...
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Setting log level to debug
Turned on maintenance mode
Updating database schema
Updated database
Disabled incompatible app: apporder
Disabled incompatible app: bruteforcesettings
Disabled incompatible app: calendar
Disabled incompatible app: contacts
Disabled incompatible app: deck
Disabled incompatible app: end_to_end_encryption
Disabled incompatible app: extract
Disabled incompatible app: files_automatedtagging
Disabled incompatible app: groupfolders
Disabled incompatible app: notes
Disabled incompatible app: tasks
Disabled incompatible app: twofactor_nextcloud_notification
[... 227 lines cut ...]
Module proxy already enabled
Considering dependency proxy for proxy_http:
Module proxy already enabled
Module proxy_http already enabled
Considering dependency proxy for proxy_wstunnel:
Module proxy already enabled
Enabling module proxy_wstunnel.
To activate the new configuration, you need to run:
  service apache2 restart
System config value trusted_domains => 3 set to string mydomain.eu
System config value overwrite.cli.url set to string https://mydomain.eu/
System config value trusted_proxies => 11 set to string 127.0.0.1
System config value trusted_proxies => 12 set to string ::1
System config value trusted_proxies => 13 set to string mydomain.eu
System config value trusted_proxies => 14 set to string 77.6.220.222
✓ redis is configured
✓ push server is receiving redis messages
✓ push server can load mount info from database
🗴 push server can't connect to the Nextcloud server
  Response from nextcloud is not a number: invalid digit found in string
notify_push setup failed. You are probably behind a proxy
Run 'ncc config:system:set trusted_proxies 15 --value=<proxy_IP>' and then 'ncc notify_push:setup https://<domain>/push to enable
Check https://help.nextcloud.com/tags/ncp for support
Backup stored at /var/www/nextcloud/data/ncp-update-backups/nextcloud-bkp_20210923_1632421521-20.0.12.1.tar.gz
Clean up...

svenb1234, 2021-09-23:
Systeminfo before updating:

NextCloudPi version | v1.39.13
-- | --
NextCloudPi image | NextCloudPi_docker_01-19-21
distribution | Debian GNU/Linux 10 \n \l
automount | no
USB devices | sda sdb
datadir | /var/www/nextcloud/data
data in SD | yes
data filesystem | ext2/ext3
data disk usage | 944G/2.0T
rootfs usage | 944G/2.0T
swapfile | /var/swap
dbdir | /data/database
Nextcloud check | ok
Nextcloud version | 20.0.12.1
HTTPD service | up
PHP service | up
MariaDB service | up
Redis service | up
HPB service | down
Postfix service | up
internet check | ok
port check 80 | open
port check 443 | open
IP | 172.17.0.3
gateway | 172.17.0.1
interface | eth0

svenb1234, 2021-09-23:
Actually this did not work completely since using the dynamic DNs domain results in the usage of a self-signed certificate instead of the available letsencrypt one.

nachoparker, 2021-09-23:
try activating letsencrypt, even if it's already active

svenb1234, 2021-09-23:
Tried that already. But trying to access via the domain still uses the localhost certificate.

root@nextcloudpi:/# ncc config:system:set trusted_proxies 15 --value=172.17.0.3
System config value trusted_proxies => 15 set to string 172.17.0.3
Running letsencrypt
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Plugins selected: Authenticator webroot, Installer None
Obtaining a new certificate
IMPORTANT NOTES:
 - Congratulations! Your certificate and chain have been saved at:
   /etc/letsencrypt/live/mydomain.eu-0001/fullchain.pem
   Your key file has been saved at:
   /etc/letsencrypt/live/mydomain.eu-0001/privkey.pem
   Your cert will expire on 2021-12-22. To obtain a new or tweaked
   version of this certificate in the future, simply run certbot
   again. To non-interactively renew *all* of your certificates, run
   "certbot renew"
 - If you like Certbot, please consider supporting our work by:

   Donating to ISRG / Let's Encrypt:   https://letsencrypt.org/donate
   Donating to EFF:                    https://eff.org/donate-le

System config value trusted_domains => 22 set to string mydomain.eu
System config value trusted_domains => 3 set to string mydomain.eu
System config value overwrite.cli.url set to string https://mydomain.eu/
System config value trusted_proxies => 11 set to string 127.0.0.1
System config value trusted_proxies => 12 set to string ::1
System config value trusted_proxies => 13 set to string mydomain.eu
System config value trusted_proxies => 14 set to string 77.6.220.222
✓ redis is configured
✓ push server is receiving redis messages
✓ push server can load mount info from database
✓ push server can connect to the Nextcloud server
✓ push server is a trusted proxy
✓ push server is running the same version as the app
  configuration saved
enabled letsencrypt
Done. Press any key...

svenb1234, 2021-09-24:
root@nextcloudpi:/# service apache2 reload
[FAIL] Reloading Apache httpd web server: apache2 failed!
[warn] The apache2 configtest failed. Not doing anything. ... (warning).
Output of config test was:
AH00526: Syntax error on line 5 of /etc/apache2/sites-enabled/ncp.conf:
SSLCertificateFile: file '/etc/letsencrypt/live/yxc.selfhost.eu/fullchain.pem' does not exist or is empty
Action 'configtest' failed.
The Apache error log may have more information.

svenb1234, 2021-09-24:
root@nextcloudpi:/# ls /etc/letsencrypt/live/mydomain.eu
ls: cannot access '/etc/letsencrypt/live/mydomain.eu': No such file or directory
root@nextcloudpi:/# ls /etc/letsencrypt/live/               
README                mydomain.eu-0001/ 
root@nextcloudpi:/# ls /etc/letsencrypt/live/mydomain.eu-0001/
README         cert.pem       chain.pem      fullchain.pem  privkey.pem

svenb1234, 2021-09-24:
After adding the -0001 to the docker data folder  etc/apache2/sites-available/ncp.conf it works. The nextcloud.conf did have it already.

Is that sa temporary fix and will be overwritten? If so, how do I fix it permanentely?

nachoparker, 2021-10-04:
I added some improvements to detect the `-0001`  suffix that some people are seeing lately. Can you update to the latest NCP and try again? just run the following and share output please

```
sudo su
set -x
source /usr/local/etc/ncp-templates/nextcloud.conf.sh
```

githubtefo, 2021-10-09:
I'm not on docker, but also have to reactivate Let's Encrypt manually and run it after NC21 update to avoid the use of self-signed certificate.
I'm also having the issue
`🗴 push server can't connect to the Nextcloud server
ncpi is not configured as a trusted domain`
I tried to manually add my domain to the trusted list but it seems not working.
Everything seems to be working anyway.

closed 2021-10-12 by nachoparker