nextcloud/nextcloudpi #1346: HPB does not started due to error in trusted_proxies WITHOUT reverse proxy

gadric, 2021-09-23:
My setup: ncp x64 in docker on INTEL NUC. I am not behind any reserve proxy. But my HPB is down and when I ran self-test on notify_push, it tells me, that I have to add the public IP to the trusted proxies. When I do this, it works. But when I restart, this problems shows up again. 

I should add that my public IP changes every day. 

`
root@2bcf9dc196f8:/# ncc notify_push:self-test
✓ redis is configured
✓ push server is receiving redis messages
✓ push server can load mount info from database
✓ push server can connect to the Nextcloud server
🗴 push server is not a trusted proxy, please add '<<<PUBLIC_IP>>>' to the list of trusted proxies or configure any existing reverse proxy to forward the 'x-forwarded-for' send by the push server.
  See https://docs.nextcloud.com/server/latest/admin_manual/configuration_server/reverse_proxy_configuration.html#defining-trusted-proxies for how to set trusted proxies.
  The following trusted proxies are currently configured: "127.0.0.1", "::1", "FQDN", "XXXXX.myfritz.net.XXXX"
    of which the following seem to be invalid: "FQDN", "XXXXX.myfritz.net.XXXX"
  The following x-forwarded-for header was received by Nextcloud: 1.2.3.4
    from the following remote: <<<PUBLIC_IP>>>

  If you're having issues getting the trusted proxy setup working, you can try bypassing any existing reverse proxy
  in your setup by setting the `NEXTCLOUD_URL` environment variable to point directly to the internal Nextcloud webserver url
  (You will still need the ip address of the push server added as trusted proxy) 
`

**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.39.13
NextCloudPi image    NextCloudPi_docker_09-20-21
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          none
datadir              /data/nextcloud/data
data in SD           no
data filesystem      ext2/ext3
data disk usage      563G/884G
rootfs usage         17G/32G
swapfile             none
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    21.0.4.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               3days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
[... 122 lines cut ...]
```

{"reqId":"F0BicYmcc3ozIpNhdIwi","level":0,"time":"2021-09-23T19:47:29+00:00","remoteAddr":"","user":"--","app":"apporder","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"F0BicYmcc3ozIpNhdIwi","level":0,"time":"2021-09-23T19:47:29+00:00","remoteAddr":"","user":"--","app":"files_external","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"F0BicYmcc3ozIpNhdIwi","level":0,"time":"2021-09-23T19:47:29+00:00","remoteAddr":"","user":"--","app":"files_sharing","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"F0BicYmcc3ozIpNhdIwi","level":0,"time":"2021-09-23T19:47:29+00:00","remoteAddr":"","user":"--","app":"metadata","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"F0BicYmcc3ozIpNhdIwi","level":0,"time":"2021-09-23T19:47:29+00:00","remoteAddr":"","user":"--","app":"user_usage_report","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"YUzZ0UVT2A@O3uoQJbBW4QAAEAQ","level":0,"time":"2021-09-23T19:47:29+00:00","remoteAddr":"79.251.3.157","user":"--","app":"files_external","method":"GET","url":"/index.php/apps/notify_push/test/cookie","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"YUzZ0UVT2A@O3uoQJbBW4QAAEAQ","level":0,"time":"2021-09-23T19:47:29+00:00","remoteAddr":"79.251.3.157","user":"--","app":"files_sharing","method":"GET","url":"/index.php/apps/notify_push/test/cookie","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"YUzZ0UVT2A@O3uoQJbBW4QAAEAQ","level":0,"time":"2021-09-23T19:47:29+00:00","remoteAddr":"79.251.3.157","user":"--","app":"user_usage_report","method":"GET","url":"/index.php/apps/notify_push/test/cookie","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"YUzZ0UVT2A@O3uoQJbBW4QAAEAQ","level":0,"time":"2021-09-23T19:47:29+00:00","remoteAddr":"79.251.3.157","user":"--","app":"apporder","method":"GET","url":"/index.php/apps/notify_push/test/cookie","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"YUzZ0UVT2A@O3uoQJbBW4QAAEAQ","level":0,"time":"2021-09-23T19:47:29+00:00","remoteAddr":"79.251.3.157","user":"--","app":"metadata","method":"GET","url":"/index.php/apps/notify_push/test/cookie","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"YUzZ0UVT2A@O3uoQJbBW4gAAEAQ","level":0,"time":"2021-09-23T19:47:29+00:00","remoteAddr":"79.251.3.157","user":"--","app":"files_external","method":"GET","url":"/index.php/apps/notify_push/test/remote","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"YUzZ0UVT2A@O3uoQJbBW4gAAEAQ","level":0,"time":"2021-09-23T19:47:29+00:00","remoteAddr":"79.251.3.157","user":"--","app":"files_sharing","method":"GET","url":"/index.php/apps/notify_push/test/remote","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"YUzZ0UVT2A@O3uoQJbBW4gAAEAQ","level":0,"time":"2021-09-23T19:47:29+00:00","remoteAddr":"79.251.3.157","user":"--","app":"user_usage_report","method":"GET","url":"/index.php/apps/notify_push/test/remote","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"YUzZ0UVT2A@O3uoQJbBW4gAAEAQ","level":0,"time":"2021-09-23T19:47:29+00:00","remoteAddr":"79.251.3.157","user":"--","app":"apporder","method":"GET","url":"/index.php/apps/notify_push/test/remote","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"YUzZ0UVT2A@O3uoQJbBW4gAAEAQ","level":0,"time":"2021-09-23T19:47:29+00:00","remoteAddr":"79.251.3.157","user":"--","app":"metadata","method":"GET","url":"/index.php/apps/notify_push/test/remote","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"BoZ2vOhkhB18li5WhA9D","level":0,"time":"2021-09-23T19:47:34+00:00","remoteAddr":"","user":"--","app":"apporder","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"BoZ2vOhkhB18li5WhA9D","level":0,"time":"2021-09-23T19:47:34+00:00","remoteAddr":"","user":"--","app":"files_external","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Boo [... cut ...]
{"reqId":"BoZ2vOhkhB18li5WhA9D","level":0,"time":"2021-09-23T19:47:34+00:00","remoteAddr":"","user":"--","app":"metadata","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"BoZ2vOhkhB18li5WhA9D","level":0,"time":"2021-09-23T19:47:34+00:00","remoteAddr":"","user":"--","app":"user_usage_report","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}

```
</details>

nachoparker, 2021-10-04:
Do you access by public IP? do you not have a domain name?

gadric, 2021-10-05:
I do not access by public IP. I have a domain name. I have already added the domain name to the trusted proxies list. But no improvement.

nachoparker, 2021-10-05:
@gadric do you have your domain name in `overwrite.cli.url`? if not, please make than change and reboot and let's see if that helps

gadric, 2021-10-06:
@nachoparker: yes my domain is in `overwrite.cli.url` In  `config.php`

Hajo0611, 2021-10-07:
Hello all,

I have exactly the same problem. After the daily change of the dynamic IP, the HPB service is down. In the config.php the following entries are stored:

'overwrite.cli.url' => 'https://xxxxxx.goip.de/',

'trusted_proxies' => 
  array (
     0 => '192.168.178.38',
    11 => '127.0.0.1',
    12 => '::1',
    13 => 'xxxxxxx.goip.com',
    14 => 'xxx.196.192.224',
  ),

The following entries are stored in notify_push.service:

[Unit].
Description = Push daemon for Nextcloud clients
After = mysql.service

[Service]
Environment=PORT=7867
Environment=NEXTCLOUD_URL=https://xxxxxxx.goip.de
ExecStart = /var/www/nextcloud/apps/notify_push/bin/"armv7"/notify_push --allow-self-signed /var/www/nextcloud/config/config.php
User=www-data

[Install]
WantedBy = multi-user.targe

If I put the new IP in config.php after changing the dynamic IP, the HPB service is up again until after 24 hours the dynamic IP changes again.

Hajo0611, 2021-10-11:
Hello all, 

I have solved my problem with the HPB Services for my installation.

I have adjusted the following entries. Now the service stays up even if the dynamic IP changes.

I added the following entry to the etc/hosts:

127.0.0.1 localhost odroidxu4
::1 localhost odroidxu4 ip6-localhost ip6-loopback
fe00::0 ip6-localnet
ff00::0 ip6-mcastprefix
ff02::1 ip6-allnodes
127.0.1.1 nextcloudpi
192.168.178.38 xxxxxxx.goip.de (added)

In etc/system/system/notify_push.service I deleted the entry ENVIRONMENT=NEXTCLOUD_URL=https://xxxxxxx.goip.de:

[Unit]
Description = Push daemon for Nextcloud clients
After = mysql.service

[Service.]
Environment=PORT=7867
ExecStart = /var/www/nextcloud/apps/notify_push/bin/"armv7"/notify_push --allow-self-signed /var/www/nextcloud/config/config.php
User=www-data

[Install]
WantedBy = multi-user.target

In var/www/nextcloud/config/config.php I changed the entry trust_proxies from the external dynamic IP to the internal IP.

'trusted_proxies' => 
  array (
    0 => '192.168.178.38',
    11 => '127.0.0.1',
    12 => '::1',
    13 => 'nextcloudpi',
    14 => '192.168.178.38', (changed to internal IP)
  ),