nextcloud/nextcloudpi #1347: Can not access activation page after installation of Nextcloudpi (ARM64) on M1 mac

MarkWieczorek, 2021-09-24:
I have been able to successfully install nextcloudpi on a raspberry pi and on an intel mac using docker. However, I have not been able to successfully install nextcloudpi on a new M1 mac.

For the installation, I am using the command

```
docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v ncdata:/data --name nextcloudpi ownyourbits/nextcloudpi my-local-ip
```

The logs show “init done” and that everything is running, and I can even access the process by the terminal. However, when I go to https://localhost:443 to activate Nexcloudpi, after accepting to use a self-signed certificate, either nothing happens (firefox) or the browser gives an uniformative error (safari). 

I note that if I install the x86 version of nextcloudpi on my M1 that *everything works* (though, obviously with a huge performance decrease as it is run in emulation mode). This seems to indicate that there is some sort of problem with the ARM64 docker container (at least when used on an M1 mac).

### Steps to reproduce
1. `docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v ncdata:/data --name nextcloudpi ownyourbits/nextcloudpi my-local-ip` on M1 mac
2. go to https://localhost:443 to activate Nexcloudpi

### Expected behaviour
You should be able to access the activation page

### Actual behaviour
After accepting the browser warning that the site uses a self-signed certificate, nothing happens (firefox), or an error appears (safari)

### Server configuration

**Operating system:** 
macos 11.6
**Web server:**

**Database:**

**PHP version:**

**Nextcloud version:** (see Nextcloud admin page)

**Updated from an older Nextcloud/ownCloud or fresh install:**

**Where did you install Nextcloud from:**

**Signing status:**
<details>
<summary>Signing status</summary>

```
Login as admin user into your Nextcloud and access 
http://example.com/index.php/settings/integrity/failed 
paste the results here.
```
</details>

**List of activated apps:**
<details>
<summary>App list</summary>

```
If you have access to your command line run e.g.:
sudo -u www-data php occ app:list
from within your Nextcloud installation folder
```
</details>

**Nextcloud configuration:**
<details>
<summary>Config report</summary>

```
If you have access to your command line run e.g.:
sudo -u www-data php occ config:list system
from within your Nextcloud installation folder

or 

Insert your config.php content here. 
Make sure to remove all sensitive content such as passwords. (e.g. database password, passwordsalt, secret, smtp password, …)
```
</details>

**Are you using external storage, if yes which one:** local/smb/sftp/...

**Are you using encryption:** yes/no

**Are you using an external user-backend, if yes which one:** LDAP/ActiveDirectory/Webdav/...

#### LDAP configuration (delete this part if not used)
<details>
<summary>LDAP config</summary>

```
With access to your command line run e.g.:
sudo -u www-data php occ ldap:show-config
from within your Nextcloud installation folder

Without access to your command line download the data/owncloud.db to your local
computer or access your SQL server remotely and run the select query:
SELECT * FROM `oc_appconfig` WHERE `appid` = 'user_ldap';

Eventually replace sensitive data as the name/IP-address of your LDAP server or groups.
```
</details>

### Client configuration
**Browser:**

**Operating system:**

### Logs

#### Web server error log
<details>
<summary>Web server error log</summary>

```
Insert your webserver log here
```
</details>

#### Nextcloud log (data/nextcloud.log)
<details>
<summary>Nextcloud log</summary>

```
Insert your Nextcloud log here
```
</details>

#### Browser log
<details>
<summary>Browser log</summary>

```
Insert your browser log here, this could for example include:

a) The javascript console log
b) The network log
c) ...
```
</details>

svenb1234, 2021-09-25:
Since it has been a while since I activated nextcloudpi, I am not sure but can you access the webpanel it via my-local-ip without any port or my-local-ip:**444**3?

ovpc, 2021-09-25:
Local IP:4443

MarkWieczorek, 2021-09-25:
I have tried every possible combination of https, http, port 443, port 4443, localhost, nextcloudpi.local, and my-local-ip, all to the same result.

Regardless, I did discover using `ncp-config` that `nc-webui `was set to "no" by default. However, setting this to "yes" had no effect. I have also tried updating nextcloud to version 21 and then 22, and have also updated nextcloudpim all using `ncp-config` in the terminal. I am starting to think that this has something to do with the webserver/apache not working, but I don't really know how to debug this.

br6, 2021-09-25:
any messages like this if you run docker logs -f nextcloudpi

Cannot load Zend OPcache - it was already loaded

MarkWieczorek, 2021-09-25:
I see that about 15 times in the log. I'll do a search on what that means, but is there a quick fix? I think that after I updated from nextcloud v20 to 21 and 22 (this morning) that this error might have gone away.

ovpc, 2021-09-25:
It is harmless but reoccurring 'upstream' bug when apt updates php.
Check for extra 20-opcache.ini files next to:
/etc/php/7.3/cli/conf.d/10-opcache.ini
/etc/php/7.3/fpm/conf.d/10-opcache.ini

Obviously when opcache is loaded for a 2nd time it throws an error in the logs, it is safe to rm them.

MarkWieczorek, 2021-09-30:
I have just confirmed on a separate intel mac the following:

Intel mac
* docker install of nextcoudpi-x86 works
* docker install of nextcloudpi-arm64 doesn't work

M1 mac
* docker install of nextcoudpi-x86 works
* docker install of nextcloudpi-arm64 doesn't work

Please label this issue as a BUG. The nextcloudpi-arm64 docker image simply does not work (as described above). Thanks!

closed 2021-10-15 by nachoparker