nextcloud/nextcloudpi #1351: Passwords app believes HTTPS is not configured properly

mohamnag, 2021-09-29:
I have been using the Passwords app for a while now with no problem but after upgrading to  v1.39.16 now I see this message when I open the passwords app on web:

[image]

my setup is that I have a router that redirects ports 80 and 443 to NCP and NCP is doing the rest. I have no additional proxy or whatsoever. So I assume that I don't have to setup anything manually.

I also see following information on same page (`1.2.3.4` is my public IP and `cloud.mydomain.com` my domain. `192.168.1.1` is the router):

HTTPS detection | Actual | Expected
-- | -- | --
Nextcloud reported protocol | http | https
Client reported protocol | https | https
Proxy detected | no | Yes if Proxy
Detected remote address | 1.2.3.4 | 1.2.3.4
overwriteprotocol | http | not set
overwritecondaddr | not set | not set
trusted_proxies | {"11":"127.0.0.1","12":"::1","13":"cloud.mydomain.com","14":"1.2.3.4","0":"192.168.1.1","1":"1.2.3.4","2":"127.0.0.1"} | []
$_SERVER['HTTPS'] | on | on
$_SERVER['REQUEST_SCHEME'] | https | https
$_SERVER['REMOTE_ADDR'] | 1.2.3.4 |  
$_SERVER['HTTP_X_FORWARDED_PROTO'] | not set | not set
$_SERVER['HTTP_X_FORWARDED_FOR'] | not set

**System information**

mohamnag, 2021-09-30:
So I just found out that funny enough, when I deactivate "Force HTTPS" on NCP control panel, it all works fine! I actually expected other way around that this config shall enforce https and not show secure connection as a non-secure one!

nachoparker, 2021-09-30:
thanks for the report, and thanks @DJCrapsody for the PR

I want to try something else first, can you please run `sudo ncp-update devel` and let me know if that fixes things?

One of you is behind a proxy and the other one isn'.

DJCrapsody, 2021-09-30:
> thanks for the report, and thanks @DJCrapsody for the PR
> 
> I want to try something else first, can you please run `sudo ncp-update devel` and let me know if that fixes things?
> 
> One of you is behind a proxy and the other one isn'.

I am behind a proxy and I had to force https to "off" to connect my desktop client to my nextcloud server.

nachoparker, 2021-09-30:
> I am behind a proxy and I had to force https to "off" to connect my desktop client to my nextcloud server.

@DJCrapsody is this after running `sudo ncp-update devel`?

DJCrapsody, 2021-09-30:
> > I am behind a proxy and I had to force https to "off" to connect my desktop client to my nextcloud server.
> 
> @DJCrapsody is this after running `sudo ncp-update devel`?
> 
> NOTE: It is expected that if you are behind a SSL terminating proxy you need to disable `nc-httpsonly`, that's the whole point of that feature :)

Using the devel branch, it's working whether nc-httpsonly is enabled or not.
I did see the commit where you force overwriteprotocol to always https.

nachoparker, 2021-09-30:
@DJCrapsody beautiful, thanks for confirming. Yes, that's what I wanted to try.

You are behind a proxy, correct?

DJCrapsody, 2021-09-30:
> @DJCrapsody beautiful, thanks for confirming. Yes, that's what I wanted to try.
> 
> You are behind a proxy, correct?

Correct, SSL terminating proxy.

nachoparker, 2021-09-30:
nice! thank you. If you are using the "passwords" app, does it complaint at all about anything as shown above?

DJCrapsody, 2021-09-30:
> nice! thank you. If you are using the "passwords" app, does it complaint at all about anything as shown above?

I've tried using the Passwords app with both on and off. Works fine with devel build.

nachoparker, 2021-09-30:
thanks! closing then

closed 2021-09-30 by nachoparker