nextcloud/nextcloudpi #1358: letsencrypt makes nextcloud not accessible after nc-restore & nc-import-config

WeiChihChern, 2021-10-09:
nc-store from a drive to another, and once I import previous config to the current one, I cannot access nextcloud anymore.

I've tried ncp v1.39.1, v1.39.6, v1.40.4 (docker images), and they all have the same issue. ncp-report's log section contains some login info, therefore not included.

> pi@raspberrypi:/Desktop/script_tools $ docker logs nextcloudpi
> Starting PHP-fpm
> Starting Apache
> AH00526: Syntax error on line 5 of /etc/apache2/sites-enabled/ncp.conf:
> SSLCertificateFile: file '/etc/letsencrypt/live/my-domain/fullchain.pem' does not exist or is empty
> Action 'start' failed.
> The Apache error log may have more information.
> Starting Redis
> Starting Cron
> Starting Postfix
> ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/var/run/mysqld/mysqld.sock' (2)
> Init done

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.39.1
NextCloudPi image    NextCloudPi_docker_09-14-21
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          sda sdb sdc
datadir              /data/nextcloud/data
data in SD           no
data filesystem      ext2/ext3
data disk usage      991G/13T
rootfs usage         8.8G/110G
swapfile             /var/swap
dbdir                /data-ro/database
Nextcloud check      error
HTTPD service        down
PHP service          up
MariaDB service      down
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               13:52

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

An unhandled exception has been thrown:
Doctrine\DBAL\Exception: Failed to connect to the database: An exception occurred in the driver: SQLSTATE[HY000] [2002] No such file or directory in /data/nextcloud/lib/private/DB/Connection.php:85
Stack trace:
#0 /data/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1486): OC\DB\Connection->connect()
#1 /data/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1014): Doctrine\DBAL\Connection->getWrappedConnection()
#2 /data/nextcloud/lib/private/DB/Connection.php(226): Doctrine\DBAL\Connection->executeQuery('SELECT * FROM `...', Array, Array, NULL)
#3 /data/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php(210): OC\DB\Connection->executeQuery('SELECT * FROM `...', Array, Array)
#4 /data/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php(287): Doctrine\DBAL\Query\QueryBuilder->execute()
#5 /data/nextcloud/lib/private/AppConfig.php(345): OC\DB\QueryBuilder\QueryBuilder->execute()
#6 /data/nextcloud/lib/private/AppConfig.php(110): OC\AppConfig->loadConfigValues()
#7 /data/nextcloud/lib/private/AppConfig.php(301): OC\AppConfig->getApps()
#8 /data/nextcloud/lib/private/legacy/OC_App.php(965): OC\AppConfig->getValues(false, 'installed_versi...')
#9 /data/nextcloud/lib/private/Server.php(682): OC_App::getAppVersions()
#10 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(155): OC\Server->OC\{closure}(Object(OC\Server))
#11 /data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(Pimple\Container))
#12 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(122): Pimple\Container->offsetGet('OC\\Memcache\\Fac...')
#13 /data/nextcloud/lib/private/ServerContainer.php(137): OC\AppFramework\Utility\SimpleContainer->query('OC\\Memcache\\Fac...', false)
#14 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query('OC\\Memcache\\Fac...')
#15 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(177): OC\AppFramework\Utility\SimpleContainer->get('OC\\Memcache\\Fac...')
#16 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(155): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#17 /data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(114): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(Pimple\Container))
#18 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(122): Pimple\Container->offsetGet('OCP\\ICacheFacto...')
#19 /data/nextcloud/lib/private/ServerContainer.php(137): OC\AppFramework\Utility\SimpleContainer->query('OCP\\ICacheFacto...', false)
#20 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query('OCP\\ICacheFacto...')
#21 /data/nextcloud/lib/private/Server.php(1041): OC\AppFramework\Utility\SimpleContainer->get('OCP\\ICacheFacto...')
#22 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(155): OC\Server->OC\{closure}(Object(OC\Server))
#23 /data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(Pimple\Container))
#24 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(122): Pimple\Container->offsetGet('OCP\\Lock\\ILocki...')
#25 /data/nextcloud/lib/private/ServerContainer.php(137): OC\AppFramework\Utility\SimpleContainer->query('OCP\\Lock\\ILocki...', false)
#26 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query('OCP\\Lock\\ILocki...')
#27 /data/nextcloud/lib/private/Server.php(2000): OC\AppFramework\Utility\SimpleContainer->get('OCP\\Lock\\ILocki...')
#28 /data/nextcloud/lib/private/Files/View.php(119): OC\Server->getLockingProvider()
#29 /data/nextcloud/lib/private/Server.php(430): OC\Files\View->__construct()
#30 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(155): OC\Server->OC\{closure}(Object(OC\Server))
#31 /data/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(Pimple\Container))
#32 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(122): Pimple\Container->offsetGet('OC\\Files\\Node\\H...')
#33 /data/nextcloud/lib/private/ServerContainer.php(137): OC\AppFramework\Utility\SimpleContainer->query('OC\\Files\\Node\\H...', false)
#34 /data/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query('OC\\Files\\Node\\H...')
#35 /data/nextcloud/lib/private/Server.php(1363): OC\AppFramework\Utility\SimpleContainer->get('OC\\Files\\Node\\H...')
#36 /data/nextcloud/lib/base.php(588): OC\Server->boot()
#37 /data/nextcloud/lib/base.php(1076): OC::init()
#38 /data/nextcloud/console.php(49): require_once('/data/nextcloud...')
#39 /data/nextcloud/occ(11): require_once('/data/nextcloud...')
#40 {main}
```
</details>

<details>
<summary>HTTPd logs</summary>

```

tail: cannot open '/var/log/apache2/error.log' for reading: No such file or directory

```
</details>

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

WeiChihChern, 2021-10-09:
#88 might be related.

WeiChihChern, 2021-10-09:
If nc-restore also restore the config, then the problem might not comes from nc-import-config

ovpc, 2021-10-09:
No connection to database, this does not look like a bug imho.
You are the only one reporting this, so it seems, so  please use the [forum](https://help.nextcloud.com/tag/ncp) for assistance.

ovpc, 2021-10-09:
Then please provide detailed info on your set up and procedures followed, so someone can (try to) reproduce the issue.

svenb1234, 2021-10-10:
The letsencrypt part looks like https://github.com/nextcloud/nextcloudpi/issues/1345#issuecomment-926874981 

But for the mysql part you will have to get some support from the forum.

closed 2021-10-12 by nachoparker