nextcloud/nextcloudpi #1373: ncp-restore support for zfs

Malakai13, 2021-11-02:
**Description**  
Backup tar files on zfs are not able to be restored. I'm not sure if the check was just put in before zfs support on linux, or if more work needs to be done here. Either way for those running nextcloudpi on zfs this would be a handy feature.

**Example**  
I did this in a docker container on truenas scale, but 
```
ncp-restore /zfs/path/to/backup.tar
```
https://github.com/nextcloud/nextcloudpi/blob/05f0d352cf228518d6333a375e89ca30d645ebfb/bin/ncp/BACKUPS/nc-restore.sh#L34

nachoparker, 2021-11-03:
well, you can try removing that line (or adding `-e zfs`) and if it works we can add it

Malakai13, 2021-11-03:
I tested out just adding ZFS to the check, and was able to complete the restore.
I can put in a PR if that sounds good.

I had this ZFS pool from my old server running freenas so it was the default. I think ZFS has snapshot functionality as well, but I'll have to check out BTRFS and compare to see the difference.

nachoparker, 2021-11-04:
sweet, thanks for https://github.com/nextcloud/nextcloudpi/pull/1375

ZFS and BTRFS pretty much have the same capabilities, there was just an issue with licensing historically so it would not be included in the Linux kernel, which made Red Hat focus on BTRFS instead, but yeah if we were to do snapshotting and all that with ZFS we would have to adapt all NCP tools and that's not under the radar for now, so I still recommend BTRFS

closed 2021-11-04 by nachoparker