nextcloud/nextcloudpi #1374: Error when running nc-encrypt

vcbranco, 2021-11-03:
Clean curl installation on Debian 10 x86_64

When I run nc-encrypt always stops with the following error:

[ nc-encrypt ] (Wed Nov 3 10:10:45 WET 2021)
/usr/local/bin/ncp/SECURITY/nc-encrypt.sh: line 71: PASSWORD: unbound variable
Got empty Password from stdin

ncp-report

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.43.3
NextCloudPi image    NextCloudPi_11-03-21
OS                   Debian GNU/Linux 10. 4.19.0-18-amd64 (x86_64)
automount            yes
USB devices          none
datadir              /data/ncdata/data
data in SD           no
data filesystem      btrfs
data disk usage      297M/208G
rootfs usage         4,5G/16G
swapfile             /var/swap
dbdir                /data/ncdatabase
Nextcloud check      ok
Nextcloud version    21.0.4.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enp1s0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         yes
uptime               1min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "22": "nextcloudpi",
            "11": "***REMOVED SENSITIVE VALUE***",
            "1": "10.8.6.3",
            "14": "nextcloudpi",
            "": "***REMOVED SENSITIVE VALUE***",
            "12": "***REMOVED SENSITIVE VALUE***"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "21.0.4.1",
        "overwrite.cli.url": "***REMOVED SENSITIVE VALUE***",
[... 96 lines cut ...]
Version: '10.3.31-MariaDB-0+deb10u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Debian 10
2021-11-03 11:23:08 0 [Note] InnoDB: Buffer pool(s) load completed at 211103 11:23:08

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"kZrO7xOeEP2j7jL05Ajq","level":3,"time":"2021-11-03T09:55:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/base.php","line":643,"function":"checkInstalled","class":"OC","type":"::","args":[{"__class__":"OC\\SystemConfig"}]},{"file":"/var/www/nextcloud/lib/base.php","line":1076,"function":"init","class":"OC","type":"::","args":[]},{"file":"/var/www/nextcloud/cron.php","line":43,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/base.php","Line":276,"CustomMessage":"--"},"userAgent":"--","version":""}
{"reqId":"d5kAOaCdnFRtrPSzobE0","level":3,"time":"2021-11-03T09:57:49+00:00","remoteAddr":"","user":"ncp","app":"no app in context","method":"","url":"--","message":{"Exception":"OCP\\AppFramework\\QueryException","Message":"Could not resolve trashManager! Class trashManager does not exist","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":126,"function":"resolve","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["trashManager"]},{"file":"/var/www/nextcloud/lib/private/ServerContainer.php","line":162,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["trashManager",true]},{"file":"/var/www/nextcloud/lib/private/AppFramework/DependencyInjection/DIContainer.php","line":434,"function":"query","class":"OC\\ServerContainer","type":"->","args":["trashManager",true]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":56,"function":"query","class":"OC\\AppFramework\\DependencyInjection\\DIContainer","type":"->","args":["trashManager"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/FunctionInjector.php","line":58,"function":"get","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["trashManager"]},{"function":"OC\\AppFramework\\Bootstrap\\{closure}","class":"OC\\AppFramework\\Bootstrap\\FunctionInjector","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/FunctionInjector.php","line":68,"function":"array_map","args":[{"__class__":"Closure"},[{"name":"serverContainer","__class__":"ReflectionParameter"},{"name":"logger","__class__":"ReflectionParameter"},{"name":"appManager","__class__":"ReflectionParameter"},"*** sensitive parameter replaced ***"]]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/BootContext.php","line":52,"function":"injectFn","class":"OC\\AppFramework\\Bootstrap\\FunctionInjector","type":"->","args":[[{"__class__":"OCA\\Files_Trashbin\\AppInfo\\Application"},"registerTrashBackends"]]},{"file":"/var/www/nextcloud/apps/files_trashbin/lib/AppInfo/Application.php","line":57,"function":"injectFn","class":"OC\\AppFramework\\Bootstrap\\BootContext","type":"->","args":[[{"__class__":"OCA\\Files_Trashbin\\AppInfo\\Application"},"registerTrashBackends"]]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php","line":178,"function":"boot","class":"OCA\\Files_Trashbin\\AppInfo\\Application","type":"->","args":[{"__class__":"OC\\AppFramework\\Bootstrap\\BootContext"}]},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":205,"function":"bootApp","class":"OC\\AppFramework\\Bootstrap\\Coordinator","type":"->","args":["files_trashbin"]},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":138,"function":"loadApp","class":"OC_App","type":"::","args":["files_trashbin"]},{"file":"/var/www/nextcloud/apps/dav/lib/AppInfo/Application.php","line":124,"function":"loadApps","class":"OC_App","type":"::","args":[["dav"]]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php","line":178,"function":"boot","class":"OCA\\DAV\\AppInfo\\Application","type":"->","args":[{"__class__":"OC\\AppFramework\\Bootstrap\\BootContext"}]},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":205,"function":"bootApp","class":"OC\\AppFramework\\Bootstrap\\Coordinator","type":"->","args":["dav"]},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":138,"function":"loadApp","class":"OC_App","type":"::","args":["dav"]},{"file":"/var/www/nextcloud/lib/private/legacy/OC_Util.php","line":203,"function":"loadApps","class":"OC_App","type":"::","args":[["filesystem"]]},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":554,"function":"setupFS","class":"OC_Util","type":"::","args":["*** sensitive parameter replaced ***"]},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":415,"function":"prepareUserLogin","class":"OC\\User\\Session","type":"->","args":[true,"*** sensitive parameter replaced ***"]},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":624,"function":"completeLogin","class":"OC\\User\\Session","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":367,"function":"loginWithPassword","class":"OC\\User\\Session","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/Setup.php","line":440,"function":"login","class":"OC\\User\\Session","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/core/Command/Maintenance/Install.php","line":108,"function":"install","class":"OC\\Setup","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/3rdparty/symfony/console/Command/Command.php","line":255,"function":"execute","class":"OC\\Core\\Command\\Maintenance\\Install","type":"->","args":[{"__class__":"Symfony\\Component\\Console\\Input\\ArgvInput"},{"__class__":"Symfony\\Component\\Console\\Output\\ConsoleOutput"}]},{"file":"/var/www/nextcloud/3rdparty/symfony/console/Application.php","line":1009,"function":"run","class":"Symfony\\Component\\Console\\Command\\Command","type":"->","args":[{"__class__":"Symfony\\Component\\Console\\Input\\ArgvInput"},{"__class__":"Symfony\\Component\\Console\\Output\\ConsoleOutput"}]},{"file":"/var/www/nextcloud/3rdparty/symfony/console/Application.php","line":273,"function":"doRunCommand","cla [... cut ...]
{"reqId":"GCBqELTyBocWbooKQhOD","level":2,"time":"2021-11-03T10:03:14+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"21.0.4.1"}
{"reqId":"GCBqELTyBocWbooKQhOD","level":2,"time":"2021-11-03T10:03:14+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"21.0.4.1"}

```
</details>

ncp log has the same error

nachoparker, 2021-11-03:
https://help.nextcloud.com/t/error-when-running-nc-encrypt/126599/2

nachoparker, 2021-11-03:
since you started a thread in the forums, we can initially use that

closed 2021-11-03 by nachoparker