nextcloud/nextcloudpi #1380: Something wrong with the NCP LXD Container? tar.bz2 vs tar.gz

michuvon, 2021-11-07:
First of all, thank you @nachoparker and all who have been involved to put together a LXD container.
In the near future, I would like to write documentation on how to use the LXC/LXD template, in general and on [Proxmox](https://help.nextcloud.com/t/getting-started-with-nextcloudpi-on-proxmox/113487) in particular. But I run into some issues.

1. There is some inconsistency in the naming:
- On [ownyourbits/nextcloudpi ](https://ownyourbits.com/nextcloudpi/) it says it's a `tar.bz` file.
- In [ownyourbits/downloads](https://ownyourbits.com/downloads/NextCloudPi_LXD_10-08-21/) it is a `tar.bz2` file. 
I guess that is just a typo.

2.  [ownyourbits/nextcloudpi ](https://ownyourbits.com/nextcloudpi/) suggests _"Run NextCloudPi in any system that supports LXC containers, such as **Proxmox**."_ however, Proxmox is using `tar.gz` for the LXC Templates. It is therefore not even possible to upload the `NextCloudPi_LXD_10-08-21.tar.bz2` file.

[I asked about this in the proxmox forum,](https://forum.proxmox.com/threads/running-a-lxd-image-on-proxmox.98878/#post-427502) Where I was suggested to simply try and rename. I tried with my Mac, but it did not work. It still won't let me upload it.

The question is, is the NCP LXD Container compressed correctly?
- Is `tar.bz2`the usual format for LXC/LXD and Proxmox with 'tar.gz' is simply using something strange?

Where can the LXD Container be run in its current stage, did someone try and test it successfully? Did I simply miss something?

___
Ad: Please consider contributing to the [NCP LXC Documentation for Proxmox.](https://help.nextcloud.com/t/getting-started-with-nextcloudpi-on-proxmox/113487)

nachoparker, 2021-11-08:
The filename extension doesn't come from a typo, but from doing `lxc export`. I do not what format Proxmox expects, so we would have to check what the process is to import an LXD image into proxmox.

At any rate, this is better suited for the forums

> Where can the LXD Container be run in its current stage, did someone try and test it successfully? Did I simply miss something?

It's right there in the README.md -> https://github.com/nextcloud/nextcloudpi#run-in-lxd

```
lxc import NextCloudPi_LXD_09-29-21.tar.bz
lxc start ncp
```

closed 2021-11-08 by nachoparker

michuvon, 2021-11-08:
ok, thanks.
In that case it is a case for the forum. Thanks, I will also try to get some more infos on the proxmox forum so we can write a doc on how to get the LXD container run in Proxmox.

You can close this issue.

nachoparker, 2021-11-08:
awesome, thanks. You can link the forum post here for reference