nextcloud/nextcloudpi #1386: www-data: command not allowed when openeing NCP-webui

feutl, 2021-11-18:
For some time now, I do see error messages on my system when opening the NCP-WebUi

```
www-data : command not allowed ; TTY=unknown ; PWD=/var/www/ncp-web ; USER=www-data ; COMMAND=/usr/bin/php /var/www/nextcloud/occ maintenance:mode
www-data : command not allowed ; TTY=unknown ; PWD=/var/www/ncp-web ; USER=www-data ; COMMAND=/usr/bin/php /var/www/nextcloud/occ status
```

What is it in the sudoers file, which I have not touch for this
```
www-data ALL = NOPASSWD: /home/www/ncp-launcher.sh , /home/www/ncp-backup-launcher.sh, /sbin/halt, /sbin/reboot
````

* NextCloudPi version | v1.43.7
* Nextcloud version | 21.0.4.1

I do not know if there should have been something added to the sudoers file or if I missed something during the many years of running NextcloudPi on my x86 server using Debian 9.

ovpc, 2021-11-18:
We have not the manpower to support backward compatibility with Debian 9, we are on 10, preparing to move to 11.
Can try to find [help on forum](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm/33)

closed 2021-11-18 by ovpc

feutl, 2021-11-18:
Damnit, sorry this is my stupid mistake, wanted to say started with Debian 9 and upgraded to Debian 10.
Got lost, sorry for this.
I am running Debian 10 for some time too, but thought the upgrade information is important but did not finish the sentence. Sorry

ovpc, 2021-11-18:
No problem, please use the forum, until we can establish it is a reproducible bug and/or affecting multiple users, in which case we can re-open this thread.

feutl, 2021-11-19:
I think it would be good just for reference to mention that the problem should be fixed by
https://github.com/nextcloud/nextcloudpi/commit/4717eb334fcc7963c1215b83f26f7b251e8969f1

As stated by @nachoparker in this comment
https://help.nextcloud.com/t/www-data-command-not-allowed-when-openeing-ncp-webui/127519/25

Thanks to everyone helping out.