nextcloud/nextcloudpi #1387: Error on renewing existing letsencrypt certificate

bitti1, 2021-11-19:
I'd like to manually renew an existing certificate via: ncp-config >> Network >> letsencrypt >> Start.
The process went through successfully, I've got a new certificate, but the step "System config value trusted_domains =>  set to string foo.mooo.com" failed:

Running letsencrypt
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Plugins selected: Authenticator webroot, Installer None
Renewing an existing certificate
Running deploy-hook command: /etc/letsencrypt/renewal-hooks/deploy/ncp
IMPORTANT NOTES:
 - Congratulations! Your certificate and chain have been saved at:
   /etc/letsencrypt/live/foo.mooo.com/fullchain.pem
   Your key file has been saved at:
   /etc/letsencrypt/live/foo.mooo.com/privkey.pem
   Your cert will expire on 2022-02-17. To obtain a new or tweaked
   version of this certificate in the future, simply run certbot
   again. To non-interactively renew *all* of your certificates, run
   "certbot renew"
 - If you like Certbot, please consider supporting our work by:

   Donating to ISRG / Let's Encrypt:   https://letsencrypt.org/donate
   Donating to EFF:                    https://eff.org/donate-le

INFO: Letsencrypt domain is foo.mooo.com
INFO: Metrics enabled: no
System config value trusted_domains =>  set to string foo.mooo.com
/usr/local/etc/library.sh: line 130: foo.mooo.com: **syntax error: invalid arithmetic operator (error token is ".mooo.com")**
Done. Press any key...

-----------

I additionally attached the system info (sudo ncp-report):

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.44.3
OS                   Raspbian GNU/Linux 10. 5.10.63-v8+ (aarch64)
automount            yes
USB devices          sda 
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      196G/470G
rootfs usage         196G/470G
swapfile             /dev/zram3
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    22.2.2.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               12days

```
</details>
[... 106 lines cut ...]
{"reqId":"YZeQjoWNk1g9HinfO-lEnQAAAMo","level":2,"time":"2021-11-19T11:54:54+00:00","remoteAddr":"192.168.10.27","user":"ncp","app":"ransomware_protection","method":"PROPFIND","url":"/remote.php/dav/files/ncp/","message":"Prevented upload of .lock because it matches extension pattern \".lock\"","userAgent":"Mozilla/5.0 (Windows) mirall/3.3.6stable-Win64 (build 20211028) (Nextcloud, windows-10.0.22000 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"22.2.2.0"}
{"reqId":"YZeRmoWNk1g9HinfO-lEwQAAANQ","level":2,"time":"2021-11-19T11:59:22+00:00","remoteAddr":"192.168.10.27","user":"ncp","app":"ransomware_protection","method":"PROPFIND","url":"/remote.php/dav/files/ncp/","message":"Prevented upload of .lock because it matches extension pattern \".lock\"","userAgent":"Mozilla/5.0 (Windows) mirall/3.3.6stable-Win64 (build 20211028) (Nextcloud, windows-10.0.22000 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"22.2.2.0"}
{"reqId":"YZeSxrGbdftwwqbuWkkwewAAAFU","level":2,"time":"2021-11-19T12:04:22+00:00","remoteAddr":"192.168.10.27","user":"ncp","app":"ransomware_protection","method":"PROPFIND","url":"/remote.php/dav/files/ncp/","message":"Prevented upload of .lock because it matches extension pattern \".lock\"","userAgent":"Mozilla/5.0 (Windows) mirall/3.3.6stable-Win64 (build 20211028) (Nextcloud, windows-10.0.22000 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"22.2.2.0"}
{"reqId":"YZeT8rGbdftwwqbuWkkwiAAAAEw","level":2,"time":"2021-11-19T12:09:22+00:00","remoteAddr":"192.168.10.27","user":"ncp","app":"ransomware_protection","method":"PROPFIND","url":"/remote.php/dav/files/ncp/","message":"Prevented upload of .lock because it matches extension pattern \".lock\"","userAgent":"Mozilla/5.0 (Windows) mirall/3.3.6stable-Win64 (build 20211028) (Nextcloud, windows-10.0.22000 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"22.2.2.0"}
{"reqId":"YZeUMLGbdftwwqbuWkkwjAAAAFA","level":2,"time":"2021-11-19T12:10:24+00:00","remoteAddr":"192.168.10.27","user":"ncp","app":"ransomware_protection","method":"PROPFIND","url":"/remote.php/dav/files/ncp/","message":"Prevented upload of .lock because it matches extension pattern \".lock\"","userAgent":"Mozilla/5.0 (Windows) mirall/3.3.6stable-Win64 (build 20211028) (Nextcloud, windows-10.0.22000 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"22.2.2.0"}
{"reqId":"YZeVPIWNk1g9HinfO-lE9wAAAMs","level":2,"time":"2021-11-19T12:14:52+00:00","remoteAddr":"192.168.10.27","user":"ncp","app":"ransomware_protection","method":"PROPFIND","url":"/remote.php/dav/files/ncp/","message":"Prevented upload of .lock because it matches extension pattern \".lock\"","userAgent":"Mozilla/5.0 (Windows) mirall/3.3.6stable-Win64 (build 20211028) (Nextcloud, windows-10.0.22000 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"22.2.2.0"}
{"reqId":"YZeWaIWNk1g9HinfO-lFBAAAANI","level":2,"time":"2021-11-19T12:19:52+00:00","remoteAddr":"192.168.10.27","user":"ncp","app":"ransomware_protection","method":"PROPFIND","url":"/remote.php/dav/files/ncp/","message":"Prevented upload of .lock because it matches extension pattern \".lock\"","userAgent":"Mozilla/5.0 (Windows) mirall/3.3.6stable-Win64 (build 20211028) (Nextcloud, windows-10.0.22000 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"22.2.2.0"}
{"reqId":"YZeXlIWNk1g9HinfO-lFGQAAANE","level":2,"time":"2021-11-19T12:24:52+00:00","remoteAddr":"192.168.10.27","user":"ncp","app":"ransomware_protection","method":"PROPFIND","url":"/remote.php/dav/files/ncp/","message":"Prevented upload of .lock because it matches extension pattern \".lock\"","userAgent":"Mozilla/5.0 (Windows) mirall/3.3.6stable-Win64 (build 20211028) (Nextcloud, windows-10.0.22000 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"22.2.2.0"}
{"reqId":"YZeX0IWNk1g9HinfO-lFIAAAAMA","level":2,"time":"2021-11-19T12:25:52+00:00","remoteAddr":"192.168.10.27","user":"ncp","app":"ransomware_protection","method":"PROPFIND","url":"/remote.php/dav/files/ncp/","message":"Prevented upload of .lock because it matches extension pattern \".lock\"","userAgent":"Mozilla/5.0 (Windows) mirall/3.3.6stable-Win64 (build 20211028) (Nextcloud, windows-10.0.22000 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"22.2.2.0"}
{"reqId":"YZeY4IWNk1g9HinfO-lFJQAAAMI","level":2,"time":"2021-11-19T12:30:24+00:00","remoteAddr":"192.168.10.27","user":"ncp","app":"ransomware_protection","method":"PROPFIND","url":"/remote.php/dav/files/ncp/","message":"Prevented upload of .lock because it matches extension pattern \".lock\"","userAgent":"Mozilla/5.0 (Windows) mirall/3.3.6stable-Win64 (build 20211028) (Nextcloud, windows-10.0.22000 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"22.2.2.0"}
{"reqId":"YZeaDIWNk1g9HinfO-lFMwAAANM","level":2,"time":"2021-11-19T12:35:24+00:00","remoteAddr":"192.168.10.27","user":"ncp","app":"ransomware_protection","method":"PROPFIND","url":"/remote.php/dav/files/ncp/","message":"Prevented upload of .lock because it matches extension pattern \".lock\"","userAgent":"Mozilla/5.0 (Windows) mirall/3.3.6stable-Win64 (build 20211028) (Nextcloud, windows-10.0.22000 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"22.2.2.0"}
{"reqId":"YZebOLGbdftwwqbuWkkw2QAAAE4","level":2,"time":"2021-11-19T12:40:24+00:00","remoteAddr":"192.168.10.27","user":"ncp","app":"ransomware_protection","method":"PROPFIND","url":"/remote.php/dav/files/ncp/","message":"Prevented upload of .lock because it matches extension pattern \".lock\"","userAgent":"Mozilla/5.0 (Windows) mirall/3.3.6stable-Win64 (build 20211028) (Nextcloud, windows-10.0.22000 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"22.2.2.0"}
{"reqId":"YZecgLGbdftwwqbuWkkw6wAAAEw","level":2,"time":"2021-11-19T12:45:53+00:00","remoteAddr":"192.168.10.27","user":"ncp","app":"ransomware_protection","method":"PROPFIND","url":"/remote.php/dav/files/ncp/","message":"Prevented upload of [... cut ...]
{"reqId":"YZeg9O0U2oOds4l6tUPTggAAAFM","level":2,"time":"2021-11-19T13:04:53+00:00","remoteAddr":"192.168.10.27","user":"ncp","app":"ransomware_protection","method":"PROPFIND","url":"/remote.php/dav/files/ncp/","message":"Prevented upload of .lock because it matches extension pattern \".lock\"","userAgent":"Mozilla/5.0 (Windows) mirall/3.3.6stable-Win64 (build 20211028) (Nextcloud, windows-10.0.22000 ClientArchitecture: x86_64 OsArchitecture: x86_64)","version":"22.2.2.0"}

```
</details>

Would you please look over this issue?

Yours sincerely, Christoph Bittmann

scholz-sebastian, 2021-11-24:
I've got exactly the same problem too

vwmaus, 2021-11-28:
The same here!

nachoparker, 2021-11-29:
can't reproduce

```
Running letsencrypt
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Plugins selected: Authenticator webroot, Installer None
Renewing an existing certificate
Running deploy-hook command: /etc/letsencrypt/renewal-hooks/deploy/ncp
IMPORTANT NOTES:

    Congratulations! Your certificate and chain have been saved at:
    /etc/letsencrypt/live/foo.mooo.com/fullchain.pem
    Your key file has been saved at:
    /etc/letsencrypt/live/foo.mooo.com/privkey.pem
    Your cert will expire on 2022-02-17. To obtain a new or tweaked
    version of this certificate in the future, simply run certbot
    again. To non-interactively renew all of your certificates, run
    "certbot renew"

    If you like Certbot, please consider supporting our work by:

    Donating to ISRG / Let's Encrypt: https://letsencrypt.org/donate
    Donating to EFF: https://eff.org/donate-le

INFO: Docker installation detected
INFO: Metrics enabled: no
System config value trusted_domains => 12 set to string foo.mooo.com
warning: No domain found. Defaulting to 'cd6121c1c0f5'
System config value trusted_domains =>  set to string cd6121c1c0f5
System config value overwrite.cli.url set to string https://cd6121c1c0f5/
System config value trusted_proxies => 11 set to string 127.0.0.1
System config value trusted_proxies => 12 set to string ::1
System config value trusted_proxies => 13 set to string cd6121c1c0f5
System config value trusted_proxies => 14 set to string 172.21.0.2
✓ redis is configured
✓ push server is receiving redis messages
✓ push server can load mount info from database
✓ push server can connect to the Nextcloud server
✓ push server is a trusted proxy
✓ push server is running the same version as the app
  configuration saved
enabled letsencrypt
Done. Press any key...
```

nachoparker, 2021-11-29:
let's try something, can you run `sudo ncp-update devel` and then try again?

jordancharlier, 2021-12-01:
Same here even after sudo ncp-update devel

> let's try something, can you run `sudo ncp-update devel` and then try again?

FMstar, 2021-12-05:
Hello, 
> had the same issue at two installations

* tried "sudo ncp-update devel" with result that "NextCloudPi updated to version v1.44.5" is shown, but ncp was already 1.44.5
* i rebooted the Device
* after reboot cerificate is fine
* at NCP-Web-IF i found that at each nc-update i previsiously run (two steps from 20.x.x to 22.2.2.0) a notification at WEb-IF is displayed "Your SSl certificate(s) mydomain.xx.xxx has been renewed for another 90 days.- it seems that the certificates have been correctly renewed on each update although the Website (and clients) only presented a self signed certificate validated from "CN=archlinux" before reboot of the device

jmoery, 2021-12-05:
I'm having the same issue.  I updated everything and tried rebooting the device.  Still having the same error in line 130

Jeanoo, 2021-12-13:
One solution would be to uninstall the certificate and do it again, is there an official procedure?  
On my side, it seems that I have problems with the certificate since I can't synchronize my calendar by following the following tuto https://docs.nextcloud.com/server/19/user_manual/pim/sync_ios.html

nachoparker, 2021-12-13:
ok, since I cannot reproduce I took a drastic approach and reverted the problematic code to what it was. It should be fixed now