nextcloud/nextcloudpi #1388: Update on nc-update to 22.2.2 from 22.0.0.11

lockeftw, 2021-11-19:
Thats the Ooutput from WUI by clicking on nc-update-nextcloud and then on apply:
_[ nc-update-nextcloud ] (Fri Nov 19 13:00:40 GMT 2021)
Current Nextcloud version 22.0.0.11
Available Nextcloud version 22.2.2
Download Nextcloud 22.2.2...
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup files...
backup /var/www//nextcloud-bkp_20211119_1637326861.tar.gz generated
Maintenance mode disabled
Install Nextcloud 22.2.2...
Fix permissions...
Upgrade...
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Setting log level to debug
Turned on maintenance mode
Updating database schema
Updated database
Disabled incompatible app: calendar
Disabled incompatible app: cms_pico
Disabled incompatible app: contacts
Disabled incompatible app: cospend
Disabled incompatible app: files_accesscontrol
Disabled incompatible app: files_automatedtagging
Disabled incompatible app: files_markdown
Disabled incompatible app: forms
Disabled incompatible app: health
Disabled incompatible app: integration_whiteboard
Disabled incompatible app: mail
Disabled incompatible app: maps
Disabled incompatible app: notes
Disabled incompatible app: notify_push
Disabled incompatible app: passwords
Disabled incompatible app: phonetrack
Disabled incompatible app: polls
Disabled incompatible app: spreed
Disabled incompatible app: talk_matterbridge
Disabled incompatible app: tasks
Disabled incompatible app: theming_customcss
Updating <federation> ...
Updated <federation> to 1.12.0
Updating <lookup_server_connector> ...
Updated <lookup_server_connector> to 1.10.0
Updating <oauth2> ...
Updated <oauth2> to 1.10.0
Updating <files> ...
Updated <files> to 1.17.0
Updating <circles> ...

Upgrading Circles App

Starting ...
0/1 [>---------------------------] 0%
(Circles) Caching memberships for Members of 'admin'
1/1 [============================] 100%

Upgrading Circles App

Starting ...
0/0 [--->------------------------] 0%
Updated <circles> to 22.1.1
Updating <cloud_federation_api> ...
Updated <cloud_federation_api> to 1.5.0
Updating <dav> ...
Fix broken values of calendar objects

Starting ...
[... 684 lines cut ...]
Checking for update of app theming_customcss in appstore
Checked for update of app "theming_customcss" in App Store
Starting code integrity check...
Finished code integrity check
Update successful
Turned off maintenance mode
Resetting log level
Check indices of the share table.
Check indices of the filecache table.
Check indices of the twofactor_providers table.
Check indices of the login_flow_v2 table.
Check indices of the whats_new table.
Check indices of the cards table.
Check indices of the cards_properties table.
Check indices of the calendarobjects_props table.
Check indices of the schedulingobjects table.
Check indices of the oc_properties table.
Done.
Check columns of the comments table.
Done.
Check primary keys.
Done.
All tables already up to date!
Backup stored at /media/datapiccante/ncdata/ncp-update-backups/nextcloud-bkp_20211119_1637326861-22.0.0.11.tar.gz
Clean up...

CarlosCastaneda, 2021-11-20:
There very same to me:

 sudo /usr/local/bin/ncc upgrade
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Setting log level to debug
Turned on maintenance mode
Updating database schema
Updated database
Updating <previewgenerator> ...
InvalidArgumentException: Column "oc_preview_generation"."locked" is type Bool and also NotNull, so it can not store "false".
Update failed
Maintenance mode is kept active
Resetting log level

dyraig, 2021-11-20:
I just ran into the same issue while running the curl-installer on a freshly installed Raspberry Pi 2 with Raspbian Buster:
[...]
Thanks for using MariaDB!
Installing nc-nextcloud
Running nc-nextcloud
Downloading Nextcloud 22.2.2...
Installing  Nextcloud 22.2.2...
Creating possible missing Directories
chmod Files and Directories
chown Directories
chmod/chown .htaccess
Setting up database...
Setting up Apache...
[...]
asks 0.14.2 installed
tasks enabled
tasks already enabled
updatenotification 1.12.0 disabled
notify_push 0.3.0 installed
notify_push enabled
notify_push already enabled
Column "oc_preview_generation"."locked" is type Bool and also NotNull, so it can not store "false".

nachoparker, 2021-11-20:
should be fixed in the latest version. Otherwise please use the forums for support, you can add a link here to the forum post even after I close it.

Thanks

closed 2021-11-20 by nachoparker

daniel1v, 2021-11-29:
@nachoparker, it seems like the issues persists for several users. the only way to get around this, is to (permanently) disable the preview genearator.