nextcloud/nextcloudpi #1390: ncp-update to 22.2.2 still fails

CarlosCastaneda, 2021-11-20:
I'm referring to https://github.com/nextcloud/nextcloudpi/issues/1388 which should be fixed according to nachparker, but I still run into the very same problem and got stuck!

NextCloudPi v1.44.3 is outdated
update to v1.44.4 through 'ncp-config' or type 'sudo ncp-update'
Last login: Sat Nov 20 22:14:38 2021 from xxxxxxxx
pi@nextcloudpi:~ $ sudo ncp-update
Downloading updates
Performing updates
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Setting log level to debug
Turned on maintenance mode
Updating database schema
Updated database
Updating <previewgenerator> ...
InvalidArgumentException: Column "oc_preview_generation"."locked" is type Bool and also NotNull, so it can not store "false".
Update failed
Maintenance mode is kept active
Resetting log level
Error while applying update 1.45.0. Exiting...
pi@nextcloudpi:~ $ sudo ncp-report
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.44.3
NextCloudPi image    NextCloudPi_07-21-19
OS                   Raspbian GNU/Linux 10. 5.10.63-v7l+ (armv7l)
automount            yes
USB devices          sda sdb sdc
datadir              /media/myCloudDrive1/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      25G/1.9T
rootfs usage         5.7G/15G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    22.2.2.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        closed
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               14days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
{
    "system": {
[... 115 lines cut ...]
```

{"reqId":"HnTAZnVzHeRHSz0Pj6sG","level":1,"time":"2021-11-20T22:12:40+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"22.2.2.0"}
{"reqId":"HnTAZnVzHeRHSz0Pj6sG","level":1,"time":"2021-11-20T22:12:40+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Repair MySQL collation","userAgent":"--","version":"22.2.2.0"}
{"reqId":"HnTAZnVzHeRHSz0Pj6sG","level":1,"time":"2021-11-20T22:12:40+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::info: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"22.2.2.0"}
{"reqId":"HnTAZnVzHeRHSz0Pj6sG","level":1,"time":"2021-11-20T22:12:40+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: All tables already have the correct collation -> nothing to do","userAgent":"--","version":"22.2.2.0"}
{"reqId":"HnTAZnVzHeRHSz0Pj6sG","level":1,"time":"2021-11-20T22:12:40+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"22.2.2.0"}
{"reqId":"HnTAZnVzHeRHSz0Pj6sG","level":1,"time":"2021-11-20T22:12:40+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Repair SQLite autoincrement","userAgent":"--","version":"22.2.2.0"}
{"reqId":"HnTAZnVzHeRHSz0Pj6sG","level":1,"time":"2021-11-20T22:12:40+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"22.2.2.0"}
{"reqId":"HnTAZnVzHeRHSz0Pj6sG","level":1,"time":"2021-11-20T22:12:40+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Copy data from accounts table when migrating from ownCloud","userAgent":"--","version":"22.2.2.0"}
{"reqId":"HnTAZnVzHeRHSz0Pj6sG","level":1,"time":"2021-11-20T22:12:40+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"22.2.2.0"}
{"reqId":"HnTAZnVzHeRHSz0Pj6sG","level":1,"time":"2021-11-20T22:12:40+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Drop account terms table when migrating from ownCloud","userAgent":"--","version":"22.2.2.0"}
{"reqId":"HnTAZnVzHeRHSz0Pj6sG","level":1,"time":"2021-11-20T22:12:40+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::dbUpgradeBefore: Updating database schema","userAgent":"--","version":"22.2.2.0"}
{"reqId":"HnTAZnVzHeRHSz0Pj6sG","level":1,"time":"2021-11-20T22:12:40+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::dbUpgrade: Updated database","userAgent":"--","version":"22.2.2.0"}
{"reqId":"HnTAZnVzHeRHSz0Pj6sG","level":0,"time":"2021-11-20T22:12:40+00:00","remoteAddr":"","user":"--","app":"files_sharing","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"22.2.2.0"}
{"reqId":"HnTAZnVzHeRHSz0Pj6sG","level":0,"time":"2021-11-20T22:12:40+00:00","remoteAddr":"","user":"--","app":"music","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"22.2.2.0"}
{"reqId":"HnTAZnVzHeRHSz0Pj6sG","level":1,"time":"2021-11-20T22:12:40+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::appUpgradeStarted: Updating <previewgenerator> ...","userAgent":"--","version":"22.2.2.0"}
{"reqId":"HnTAZnVzHeRHSz0Pj6sG","level":3,"time":"2021-11-20T22:12:41+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Column \"oc_preview_generation\".\"locked\" is type Bool and also NotNull, so it can not store \"false\".","userAgent":"--","version":"22.2.2.0","exception":{"Exception":"InvalidArgumentException","Message":"Column \"oc_preview_generation\".\"locked\" is type Bool and also NotNull, so it can not store \"false\".","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/DB/MigrationService.php","line":538,"function":"ensureOracleConstraints","class":"OC\\DB\\MigrationService","type":"->","args":[{"__class__":"Doctrine\\DBAL\\Schema\\Schema"},{"__class__":"Doctrine\\DBAL\\Schema\\Schema"},3]},{"file":"/var/www/nextcloud/lib/private/DB/MigrationService.php","line":425,"function":"executeStep","class":"OC\\DB\\MigrationService","type":"->","args":["020200Date20190608205303",false]},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":1010,"function":"migrate","class":"OC\\DB\\MigrationService","type":"->","args":[]},{"file":"/var/www/nextcloud/lib/private/Updater.php","line":352,"function":"updateApp","class":"OC_App","type":"::","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/Updater.php","line":264,"function":"doAppUpgrade","class":"OC\\Updater","type":"->","args":[]},{"file":"/var/www/nextcloud/lib/private/Updater.php","line":132,"function":"doUpgrade","class":"OC\\Updater","type":"->","args":["22.2.2.0","22.2.2.0"]},{"file":"/var/www/nextcloud/core/Command/Upgrade.php","line":241,"function":"upgrade","class":"OC\\Updater","type":"->","args":[]},{"file":"/var/www/nextcloud/3rdparty/symfony/console/Command/Command.php","line":255,"function":"execute","class":"OC\\Core\\Command\\Upgrade","type":"->","args":[{"__class__":"Symfony\\Component\\Console\\Input\\ArgvInput"},{"__class__":"Symfony [... cut ...]
{"reqId":"HnTAZnVzHeRHSz0Pj6sG","level":1,"time":"2021-11-20T22:12:41+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::maintenanceActive: Maintenance mode is kept active","userAgent":"--","version":"22.2.2.0"}
{"reqId":"HnTAZnVzHeRHSz0Pj6sG","level":1,"time":"2021-11-20T22:12:41+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"22.2.2.0"}

```
</details>

wblankenfeldt, 2021-11-21:
Hi there,

I had the same issue with my nextcloudpi (raspberry 4). I am not an expert, but I found a way that seems to working (for now):

- on the computer that runs your nextcloudpi, disable the previewgenerator (this seems to be causing the trouble; you may have to cd /var/www/nextcloud first): **sudo -u www-data php occ app:disable previewgenerator**
- next, disable maintenance mode if it is switched on: **sudo -u www-data php occ maintenance:mode --off**
- then do the upgrade again: **sudo ncp-upgrade**

This seemingly initiated a downgrade to nextcloudpi 1.44.4 (for some reason, it was trying to install 1.45, which does not seem to exist) and brought my nextcloud back to life again.

My first post on github!

ovpc, 2021-11-21:
Has alreay been resolved [here](https://github.com/nextcloud/nextcloudpi/issues/1388)
Please use the forum if need further assistance.

closed 2021-11-21 by ovpc