nextcloud/nextcloudpi #1391: Instalation script fails with: Column "oc_preview_generation"."locked" is type Bool and also NotNull, so it can not store "false".

CaixaNegraPT, 2021-11-20:
Have been trying NCP using a Debian 10 VM in Proxmox ok for the past weeks. I decided to start from scratch and started a new VM... now the script ends with an error:

Column "oc_preview_generation"."locked" is type Bool and also NotNull, so it can not store "false".

ncp-report:
<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.44.4
NextCloudPi image    NextCloudPi_11-20-21
OS                   Debian GNU/Linux 10. 4.19.0-18-amd64 (x86_64)
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      4.7G/943G
rootfs usage         4.7G/943G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    22.2.2.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            ens18
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               11min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "22": "nextcloudpi.lan"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "22.2.2.0",
        "overwrite.cli.url": "http:\/\/localhost",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
[... 49 lines cut ...]
2021-11-20 22:14:56 0 [Note] InnoDB: Loading buffer pool(s) from /var/lib/mysql/ib_buffer_pool
2021-11-20 22:14:56 0 [Note] Plugin 'FEEDBACK' is disabled.
2021-11-20 22:14:56 0 [Note] InnoDB: Buffer pool(s) load completed at 211120 22:14:56
2021-11-20 22:14:56 0 [Note] Server socket created on IP: '127.0.0.1'.
2021-11-20 22:14:56 0 [Note] Reading of all Master_info entries succeeded
2021-11-20 22:14:56 0 [Note] Added new Master_info '' to hash table
2021-11-20 22:14:56 0 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.3.31-MariaDB-0+deb10u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Debian 10

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"L0JmDQ7sUyGXmzoOc3dD","level":3,"time":"2021-11-20T22:15:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Not installed","userAgent":"--","version":"","exception":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/base.php","line":649,"function":"checkInstalled","class":"OC","type":"::","args":[{"__class__":"OC\\SystemConfig"}]},{"file":"/var/www/nextcloud/lib/base.php","line":1089,"function":"init","class":"OC","type":"::","args":[]},{"file":"/var/www/nextcloud/cron.php","line":42,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/base.php","Line":277,"CustomMessage":"--"}}
{"reqId":"7UbNnEOElKliK1aVPUY2","level":3,"time":"2021-11-20T22:17:09+00:00","remoteAddr":"","user":"ncp","app":"no app in context","method":"","url":"--","message":"Could not boot files_trashbin: Could not resolve trashManager! Class trashManager does not exist","userAgent":"--","version":"22.2.2.0","exception":{"Exception":"OCP\\AppFramework\\QueryException","Message":"Could not resolve trashManager! Class trashManager does not exist","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":131,"function":"resolve","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["trashManager"]},{"file":"/var/www/nextcloud/lib/private/ServerContainer.php","line":161,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["trashManager",true]},{"file":"/var/www/nextcloud/lib/private/AppFramework/DependencyInjection/DIContainer.php","line":435,"function":"query","class":"OC\\ServerContainer","type":"->","args":["trashManager",true]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":56,"function":"query","class":"OC\\AppFramework\\DependencyInjection\\DIContainer","type":"->","args":["trashManager"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/FunctionInjector.php","line":57,"function":"get","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["trashManager"]},{"function":"OC\\AppFramework\\Bootstrap\\{closure}","class":"OC\\AppFramework\\Bootstrap\\FunctionInjector","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/FunctionInjector.php","line":67,"function":"array_map","args":[{"__class__":"Closure"},[{"name":"serverContainer","__class__":"ReflectionParameter"},{"name":"logger","__class__":"ReflectionParameter"},{"name":"appManager","__class__":"ReflectionParameter"},"*** sensitive parameter replaced ***"]]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/BootContext.php","line":51,"function":"injectFn","class":"OC\\AppFramework\\Bootstrap\\FunctionInjector","type":"->","args":[[{"__class__":"OCA\\Files_Trashbin\\AppInfo\\Application"},"registerTrashBackends"]]},{"file":"/var/www/nextcloud/apps/files_trashbin/lib/AppInfo/Application.php","line":56,"function":"injectFn","class":"OC\\AppFramework\\Bootstrap\\BootContext","type":"->","args":[[{"__class__":"OCA\\Files_Trashbin\\AppInfo\\Application"},"registerTrashBackends"]]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php","line":178,"function":"boot","class":"OCA\\Files_Trashbin\\AppInfo\\Application","type":"->","args":[{"__class__":"OC\\AppFramework\\Bootstrap\\BootContext"}]},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":205,"function":"bootApp","class":"OC\\AppFramework\\Bootstrap\\Coordinator","type":"->","args":["files_trashbin"]},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":139,"function":"loadApp","class":"OC_App","type":"::","args":["files_trashbin"]},{"file":"/var/www/nextcloud/apps/dav/lib/AppInfo/Application.php","line":168,"function":"loadApps","class":"OC_App","type":"::","args":[["dav"]]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php","line":178,"function":"boot","class":"OCA\\DAV\\AppInfo\\Application","type":"->","args":[{"__class__":"OC\\AppFramework\\Bootstrap\\BootContext"}]},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":205,"function":"bootApp","class":"OC\\AppFramework\\Bootstrap\\Coordinator","type":"->","args":["dav"]},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":139,"function":"loadApp","class":"OC_App","type":"::","args":["dav"]},{"file":"/var/www/nextcloud/lib/private/legacy/OC_Util.php","line":204,"function":"loadApps","class":"OC_App","type":"::","args":[["filesystem"]]},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":553,"function":"setupFS","class":"OC_Util","type":"::","args":["*** sensitive parameter replaced ***"]},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":414,"function":"prepareUserLogin","class":"OC\\User\\Session","type":"->","args":[true,"*** sensitive parameter replaced ***"]},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":625,"function":"completeLogin","class":"OC\\User\\Session","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/User/Session.php","line":366,"function":"loginWithPassword","class":"OC\\User\\Session","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/Setup.php","line":431,"function":"login","class":"OC\\User\\Session","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/core/Command/Maintenance/Install.php", [... cut ...] lass":"OC\\Console\\Application","type":"->","args":[]},{"file":"/var/www/nextcloud/occ","line":11,"args":["/var/www/nextcloud/console.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","Line":116,"CustomMessage":"Could not boot files_versions: Could not resolve OCA\\Files_Versions\\Versions\\IVersionManager! Class can not be instantiated"}}

```
</details>

I have repeated the process several times but the end result is the same :(

feutl, 2021-11-22:
I had the same issue.
Disabling the previewgenerator app via CLI resolved the problem for me.

I have not tested re-enabling the app again, to be honest.

feutl, 2021-11-22:
I tested to re-enable the previewgenerator shipped with NCP (verion 99.99.99) and do get the error again
` Column "oc_preview_generation"."locked" is type Bool and also NotNull, so it can not store "false".`
[image]

I also have another issue with a disabled previewgenerator - which is the reason I started to search the active issues and ended up here, that without the previewgenerator enabled my server throws the following error every day

```
In Application.php line 676:
                                                                  
  [Symfony\Component\Console\Exception\CommandNotFoundException]  
  Command "preview:pre-generate" is not defined.                  
                                                                  
  Did you mean one of these?                                      
      notification:generate                                       
      preview:repair                                              
      preview:reset-rendered-texts                                
                                                                  
Exception trace:
  at /var/www/nextcloud/3rdparty/symfony/console/Application.php:676
 Symfony\Component\Console\Application-
>find() at /var/www/nextcloud/3rdparty/symfony/console/Application.php:237
 Symfony\Component\Console\Application-
>doRun() at /var/www/nextcloud/3rdparty/symfony/console/Application.php:149
 Symfony\Component\Console\Application-
>run() at /var/www/nextcloud/lib/private/Console/Application.php:209
 OC\Console\Application-
>run() at /var/www/nextcloud/console.php:99
 require_once() at /var/www/nextcloud/occ:11
```

Even though I have deactivated preview generation in the webui.
[image]

Related issues:
- https://github.com/nextcloud/nextcloudpi/issues/1388
- https://github.com/nextcloud/nextcloudpi/issues/1390
- https://github.com/nextcloud/nextcloudpi/issues/1340
- https://github.com/nextcloud/nextcloudpi/issues/1349

CaixaNegraPT, 2021-11-25:
Here is what I get on the shell:

```
Preparing to unpack .../avahi-daemon_0.7-4+deb10u1_amd64.deb ...
Unpacking avahi-daemon (0.7-4+deb10u1) ...
Setting up libdaemon0:amd64 (0.14-7) ...
Setting up libavahi-core7:amd64 (0.7-4+deb10u1) ...
Setting up avahi-daemon (0.7-4+deb10u1) ...
Created symlink /etc/systemd/system/dbus-org.freedesktop.Avahi.service → /lib/systemd/system/avahi-daemon.service.
Created symlink /etc/systemd/system/multi-user.target.wants/avahi-daemon.service → /lib/systemd/system/avahi-daemon.service.
Created symlink /etc/systemd/system/sockets.target.wants/avahi-daemon.socket → /lib/systemd/system/avahi-daemon.socket.
Processing triggers for systemd (241-7~deb10u8) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for dbus (1.12.20-0+deb10u1) ...
Processing triggers for libc-bin (2.28-10) ...
Running nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
Nextcloud was successfully installed
Set mode for background jobs to 'cron'
System config value tempdirectory set to string /var/www/nextcloud/data/tmp
System config value mysql.utf8mb4 set to boolean true
System config value trusted_domains => 22 set to string nextcloudpi
System config value trusted_domains => 22 set to string nextcloudpi.local
System config value trusted_domains => 22 set to string nextcloudpi.lan
System config value trusted_domains => 22 set to string nextcloudpi.lan
System config value mail_smtpmode set to string sendmail
System config value mail_smtpauthtype set to string LOGIN
System config value mail_from_address set to string admin
System config value mail_domain set to string ownyourbits.com
nextcloudpi 0.0.1 enabled
calendar 2.3.4 installed
calendar enabled
calendar already enabled
contacts 4.0.6 installed
contacts enabled
contacts already enabled
notes 4.2.0 installed
notes enabled
notes already enabled
tasks 0.14.2 installed
tasks enabled
tasks already enabled
updatenotification 1.12.0 disabled
news 16.2.1 installed
news enabled
news already enabled
notify_push 0.3.0 installed
notify_push enabled
notify_push already enabled
Column "oc_preview_generation"."locked" is type Bool and also NotNull, so it can not store "false".
```

nachoparker, 2021-11-29:
I was able to reproduce this on NC22

nachoparker, 2021-11-29:
should be fixed now. Thanks for reporting

closed 2021-11-29 by nachoparker