nextcloud/nextcloudpi #1399: nc-encrypt: Decryption Passwort Error

kojid0, 2021-12-11:
I want to enable nc-encrypt but wanted to try it on a non-productive system before.
On a test instance (Rapsberry Pi, 4GB RAM, 32GB SD-Card with test data stored on SD-Card), I enabled nc-encrypt with this passwort: pLuMWguqFi963e8maMVe (I tell you because I saw an issue where some characters were not allowed, or maybe it's too long?)

After rebooting the system, I get a 500 Internal Server Error when accessing the NC nextcloud instance. Accessing the config via :4443, it will ask to decrypt. But when I enter the (correct) password, it will say Password Error. NC still shows 500 error. Trying to enter nc-config (:4443) again, it will send me to the very first activating page showing new ncp passwords for NC and nc-config. The entered password is correct. I tried it again with a new fresh install.
After having new passwords, the old data (which I uploaded to user ncp before enabling nc-encrypt) is visible.

output of `sudo ncp-report`:

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.44.5
NextCloudPi image    NextCloudPi_10-08-21
OS                   Debian GNU/Linux 10. 5.4.51-v8+ (aarch64)
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      fuseblk
data disk usage      3.3G/29G
rootfs usage         3.3G/29G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    21.0.4.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               24min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "22": "nextcloudpi",
            "1": "192.168.178.37",
            "11": "***REMOVED SENSITIVE VALUE***",
            "3": "nextcloudpi",
            "": "nextcloudpi",
            "14": "nextcloudpi"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "21.0.4.1",
        "overwrite.cli.url": "https:\/\/nextcloudpi\/",
[... 97 lines cut ...]
```

{"reqId":"n7q5WU39MK6BeBHHEOs9","level":1,"time":"2021-12-11T11:18:40+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"21.0.4.1"}
{"reqId":"n7q5WU39MK6BeBHHEOs9","level":1,"time":"2021-12-11T11:18:40+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Check encryption key format","userAgent":"--","version":"21.0.4.1"}
{"reqId":"n7q5WU39MK6BeBHHEOs9","level":1,"time":"2021-12-11T11:18:40+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"21.0.4.1"}
{"reqId":"n7q5WU39MK6BeBHHEOs9","level":1,"time":"2021-12-11T11:18:40+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Remove old dashboard app config data","userAgent":"--","version":"21.0.4.1"}
{"reqId":"n7q5WU39MK6BeBHHEOs9","level":1,"time":"2021-12-11T11:18:40+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"21.0.4.1"}
{"reqId":"n7q5WU39MK6BeBHHEOs9","level":1,"time":"2021-12-11T11:18:40+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Add job to cleanup the bruteforce entries","userAgent":"--","version":"21.0.4.1"}
{"reqId":"n7q5WU39MK6BeBHHEOs9","level":1,"time":"2021-12-11T11:18:40+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"21.0.4.1"}
{"reqId":"n7q5WU39MK6BeBHHEOs9","level":1,"time":"2021-12-11T11:18:40+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Queue a one-time job to check for user uploaded certificates","userAgent":"--","version":"21.0.4.1"}
{"reqId":"n7q5WU39MK6BeBHHEOs9","level":1,"time":"2021-12-11T11:18:40+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"21.0.4.1"}
{"reqId":"n7q5WU39MK6BeBHHEOs9","level":1,"time":"2021-12-11T11:18:40+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Repair DAV shares","userAgent":"--","version":"21.0.4.1"}
{"reqId":"n7q5WU39MK6BeBHHEOs9","level":1,"time":"2021-12-11T11:18:40+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"21.0.4.1"}
{"reqId":"n7q5WU39MK6BeBHHEOs9","level":1,"time":"2021-12-11T11:18:40+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Add background job to set the lookup server share state for users","userAgent":"--","version":"21.0.4.1"}
{"reqId":"n7q5WU39MK6BeBHHEOs9","level":1,"time":"2021-12-11T11:18:40+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::startCheckCodeIntegrity: Starting code integrity check...","userAgent":"--","version":"21.0.4.1"}
{"reqId":"n7q5WU39MK6BeBHHEOs9","level":1,"time":"2021-12-11T11:19:09+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::finishedCheckCodeIntegrity: Finished code integrity check","userAgent":"--","version":"21.0.4.1"}
{"reqId":"n7q5WU39MK6BeBHHEOs9","level":1,"time":"2021-12-11T11:19:09+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::updateEnd: Update successful","userAgent":"--","version":"21.0.4.1"}
{"reqId":"n7q5WU39MK6BeBHHEOs9","level":1,"time":"2021-12-11T11:19:09+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::maintenanceDisabled: Turned off maintenance mode","userAgent":"--","version":"21.0.4.1"}
{"reqId":"n7q5WU39MK6BeBHHEOs9","level":1,"time":"2021-12-11T11:19:09+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"21.0.4.1"}
{"reqId":"YbSLDUDxmaiG2hxVE42dhgAATQQ","level":3,"time":"2021-12-11T11:27:10+00:00","remoteAddr":"192.168.178.20","user":"ncp","app":"index","method":"GET","url":"/index.php/apps/files/","message":{"Exception":"OCP\\Files\\NotPermittedException","Message":"Could not create folder","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/Files/AppData/AppData.php","line":157,"function":"newFolder","class":"OC\\Files\\Node\\Folder","type":"->","args":["files"]},{"file":"/var/www/nextcloud/lib/private/Template/SCSSCacher.php","line":163,"function":"newFolder","class":"OC\\Files\\AppData\\AppData","type":"->","args":["files"]},{"file":"/var/www/nextcloud/lib/private/Template/CSSResourceLocator.php","line":111,"function":"process","class":"OC\\Template\\SCSSCacher","type":"->","args":["/var/www/nextcloud/apps/files","css/merged.scss","files"]},{"file":"/var/www/nextcloud/lib/private/Template/CSSResourceLocator.php","line":86,"function":"cacheAndAppendScssIfExist","class":"OC\\Template\\CSSResourceLocator","type":"->","args":["/var/www/nextcloud/apps/files","css/merged.scss","files"]},{"fil [... cut ...] rAgent":"Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:94.0) Gecko/20100101 Firefox/94.0","version":"21.0.4.1"}
{"reqId":"YbSSs3bteS3ETA0hwm6qhwAAFQA","level":2,"time":"2021-12-11T11:59:48+00:00","remoteAddr":"192.168.178.20","user":"ncp","app":"core","method":"PUT","url":"/index.php/apps/user_status/heartbeat","message":"Login failed: 'ncp' (Remote IP: '192.168.178.20')","userAgent":"Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:94.0) Gecko/20100101 Firefox/94.0","version":"21.0.4.1"}

```
</details>

nachoparker, 2021-12-13:
Hello, thanks for reporting.

Let's take things one at a time

- when you first activate nc-encrypt, does it succeed? can you still use nextcloud before you reboot?
- after you reboot, don't try to fix anything, let's go through some testing to diagnose things.

What is the output (after reboot) of `source /usr/local/etc/library.sh; needs_decrypt; echo $?`

kojid0, 2021-12-14:
Hello,
FYI: ncp v1.46.7 updated to nc 22.2.2 from 21 (as this is the version of the [download build](https://ownyourbits.com/downloads/NextCloudPi_RPi_10-08-21/)). 

----

> when you first activate nc-encrypt, does it succeed? can you still use nextcloud before you reboot?

Yes, it succeeds and I can use nc before reboot.
[image]

----

> What is the output (after reboot) of source /usr/local/etc/library.sh; needs_decrypt; echo $?

Don't know if this is the correct output, I sent: ```pi@nextcloudpi:~ $ source /usr/local/etc/library.sh; needs_decrypt; echo $```
```
cat: /usr/local/etc/ncp-config.d/nc-encrypt.cfg: Permission denied
-bash: /usr/local/etc/ncp-config.d/nc-encrypt.cfg: Permission denied
cat: /usr/local/etc/ncp-config.d/nc-encrypt.cfg: Permission denied
$
```

nachoparker, 2021-12-14:
sorry, I should have mentioned that you have to be root

```
sudo su
source /usr/local/etc/library.sh; needs_decrypt; echo $?
```

kojid0, 2021-12-14:
On a working nextcloud, I enabled nc-encrypt and rebooted. After that I did not access nextcloud nor the ncp-webpanel, hence did not enter the decrypt password.
```
pi@nextcloudpi:~ $ sudo su
root@nextcloudpi:/home/pi# source /usr/local/etc/library.sh; needs_decrypt; echo $?
0
```

The output is 0, after 2-3 secs.

Now, I accessed the ncp-webpanel (:4443) and entered the decryption password. I got an "Password error". Entering the same command as above, I get 1 as output.

kojid0, 2022-03-08:
Closing this issue since I cannot reproduce it anymore with the latest download build (03-02-22). 
Thank you anyways! :)

closed 2022-03-08 by kojid0

glenn-sorrentino, 2022-07-22:
I'm also having this problem. Came out of nowhere and I get  "Password Error" when trying to decrypt my instance. The output when running `/home/pi# source /usr/local/etc/library.sh; needs_decrypt; echo $?` is:
```
bash: /home/pi#: No such file or directory
bash: needs_decrypt: command not found
127
```

theCalcaholic, 2022-07-24:
@glenn-sorrentino That sounds like a serious issue. Can you verify that the file /usr/local/etc/library.sh does not exist (e.g. with the command `ls -l /usr/local/etc/library.sh`)?

If that's the case, you can try and download the file again with `sudo wget https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/etc/library.sh -O /usr/local/etc/library.sh`. But I have no idea whatsoever how that could have happened and it's likely a sign for an issue that's less easy to fix.

theCalcaholic, 2022-07-24:
@glenn-sorrentino If it doesn't fix your problem, please create a new issue

glenn-sorrentino, 2022-07-25:
> @glenn-sorrentino That sounds like a serious issue. Can you verify that the file /usr/local/etc/library.sh does not exist (e.g. with the command `ls -l /usr/local/etc/library.sh`)?
>
> If that's the [...]

I went ahead and spun up a new instance. Question for you, though. Is NextcloudPi going to be maintained moving into the future? Or should another Nextcloud solution be used? If so, any recommendations?

theCalcaholic, 2022-07-26:
@glenn-sorrentino I took over development and maintenance of the project from Nacho. So, to answer your question: Yes, NCP is still being maintained :)

glenn-sorrentino, 2022-07-27:
> @glenn-sorrentino I took over development and maintenance of the project from Nacho. So, to answer your question: Yes, NCP is still being maintained :)

Great - I'll start over with NCP in that case. Will open an issue if it happens again.