nextcloud/nextcloudpi #1405: NCP Docker container doesnt update from v1.45.0

ovpc, 2021-12-17:
ref : https://help.nextcloud.com/t/ncp-upgrade-from-1-45-0-to-1-46-7-does-not-work/129331/4

> docker exec -it ncp ncp-update
Downloading updates
Performing updates
Nextcloud is already latest version
NextCloudPi updated to version v1.45.0
Error while applying update 1.46.0. Exiting...

Suspect it has to do with fact that there will be fresh images sometime soon (or later), as NCP is moving to Bullseye. It doesn't make sense to dist-upgrade inside a container.

monsieurhannes, 2021-12-20:
My error message is more like the [initial post](https://help.nextcloud.com/t/ncp-upgrade-from-1-45-0-to-1-46-7-does-not-work/129331?u=hannes) from the forum:
```
root@01bc8618bcad:/# ncp-update
Downloading updates
Performing updates
Error while applying update 1.46.0. Exiting...
root@01bc8618bcad:/#
```

It's basically missing the:
```
Nextcloud is already latest version
NextCloudPi updated to version v1.45.0
```

svenb1234, 2021-12-22:
So what shall we do? Just wait and see or provide some debug info?

SoongJr, 2021-12-25:
So far I've only found out that what fails is the execution of this script: https://github.com/nextcloud/nextcloudpi/blob/master/updates/1.46.0.sh

I'm not sure why it fails, running the calls manually does not return an error code, but I did notice a stray quotation mark in there, so maybe that's a clue. Shouldn't affect the execution of _this_ script though, just the execution of the script generated by this code...:
```bash
  cat <<EOF > /etc/cron.daily/refresh_notify_push
#!/usr/bin/env bash
. /usr/local/etc/library.sh
ncc notify_push:self-test || {
  killall notify_push
  sleep 1
  start_notify_push
}"      #<<<<<<< HERE
EOF
```

SoongJr, 2021-12-25:
Ah shit, it's the check afterwards whether this is NOT in a docker image, or rather, it's the omission of the closing `exit 0`. Previous updates-scripts all ended with an `exit 0` (and they started with a `set -e`), so that's the fault!

SoongJr, 2021-12-26:
I tested with my fix (by replacing the URL in bin/ncp-update with my fork and telling it to use my feature branch; after the update the script points at the official repo again).
Except for my fork not having had all the tags, and thus my NCP thinking it was not 1.15, the update-script itself worked fine 😅

svenb1234, 2021-12-26:
Maybe obvious, but people currently installing the docker image cannot update beyond 1.43.0 until this is fixed.

SoongJr, 2021-12-26:
@svenb1234 to answer your question what to do now: Obviously you can wait for nachoparker to merge my PR, but if you (or any future readers) want to update ASAP, below is a command that will tell ncp-update to use my fork of NCP and update with my fixes. Obviously you should verify that my fork doesn't contain malicious changes first. After the update has run, ncp-update will restore itself to the official version, so future updates will automatically use the official repo instead of mine.
``` bash
sudo /usr/bin/docker exec -it nextcloudpi bash -c "sed -i 's/github.com\/nextcloud\/nextcloudpi/github.com\/SoongJr\/nextcloudpi/' data/bin/ncp-update && data/bin/ncp-update 'feature/fix_1.46.0'"
```

Hope this helps someone, if you run into issues please report them in the PR #1406

SiliconAlley, 2022-01-05:
@SoongJr thanks for your deep dive. Worked for me. Lets wait for the next official update.

nachoparker, 2022-01-10:
Thanks to SoongJr this is now fixed. Best

closed 2022-01-10 by nachoparker