nextcloud/nextcloudpi #1408: Trusted domains configuration bugs

szekelyisz, 2021-12-26:
Using NextcloudPi as a container. When you don't fill all 3 options for trusted domains, they are not applied correctly. Filling only the first two entries and clicking Apply results int he following messages:

```
[ nc-trusted-domains ] (Sun Dec 26 19:03:12 UTC 2021)
Cannot load Zend OPcache - it was already loaded
System config value trusted_domains => 22 set to string deley-nextcloud.local
Cannot load Zend OPcache - it was already loaded
System config value trusted_domains => 22 set to string nextcloud.deley.hu
```

And a red dot besides the Apply button. Filling the 3rd entry with 127.0.0.1 seems to change settings correctly. The dot turns green.

After you set the 3rd name and apply, it still doesn't work, I get the "Untrusted domain" error on the nextcloud interface. Restarting the container doesn't help. `config.php` contains:

```
  'trusted_domains' => 
  array (
    0 => 'localhost',
    22 => '127.0.0.1',
    11 => <my external ip address>,
    1 => '172.17.0.2',
    3 => 'f53a549f79ac',
  ),
```

Which is not the 3 domains that I configured.

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v1.40.4
NextCloudPi image    NextCloudPi_10-07-21
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          none
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      overlayfs
data disk usage      3.8G/15G
rootfs usage         3.8G/15G
swapfile             /var/swap
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    21.0.4.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               2:20

```
</details>

<details>
[... 98 lines cut ...]
2021-12-26 18:59:43 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2021-12-26 18:59:43 0 [Note] InnoDB: 10.3.27 started; log sequence number 3183198; transaction id 2696
2021-12-26 18:59:43 0 [Note] InnoDB: Loading buffer pool(s) from /data/database/ib_buffer_pool
2021-12-26 18:59:43 0 [Note] Plugin 'FEEDBACK' is disabled.
2021-12-26 18:59:44 0 [Note] Server socket created on IP: '127.0.0.1'.
2021-12-26 18:59:44 0 [Note] Reading of all Master_info entries succeeded
2021-12-26 18:59:44 0 [Note] Added new Master_info '' to hash table
2021-12-26 18:59:44 0 [Note] mysqld: ready for connections.
Version: '10.3.27-MariaDB-0+deb10u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Debian 10
2021-12-26 18:59:44 0 [Note] InnoDB: Buffer pool(s) load completed at 211226 18:59:44

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"pS9HjPq8VWaJyQsTPwfO","level":3,"time":"2021-10-07T20:25:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"/data-ro/nextcloud/lib/base.php","line":643,"function":"checkInstalled","class":"OC","type":"::","args":[{"__class__":"OC\\SystemConfig"}]},{"file":"/data-ro/nextcloud/lib/base.php","line":1076,"function":"init","class":"OC","type":"::","args":[]},{"file":"/data-ro/nextcloud/cron.php","line":43,"args":["/data-ro/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/data-ro/nextcloud/lib/base.php","Line":276,"CustomMessage":"--"},"userAgent":"--","version":""}
{"reqId":"hFGAKNvfQarjaxXVfGn3","level":3,"time":"2021-10-07T20:26:59+00:00","remoteAddr":"","user":"ncp","app":"no app in context","method":"","url":"--","message":{"Exception":"OCP\\AppFramework\\QueryException","Message":"Could not resolve trashManager! Class trashManager does not exist","Code":0,"Trace":[{"file":"/data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":126,"function":"resolve","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["trashManager"]},{"file":"/data-ro/nextcloud/lib/private/ServerContainer.php","line":162,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["trashManager",true]},{"file":"/data-ro/nextcloud/lib/private/AppFramework/DependencyInjection/DIContainer.php","line":434,"function":"query","class":"OC\\ServerContainer","type":"->","args":["trashManager",true]},{"file":"/data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":56,"function":"query","class":"OC\\AppFramework\\DependencyInjection\\DIContainer","type":"->","args":["trashManager"]},{"file":"/data-ro/nextcloud/lib/private/AppFramework/Bootstrap/FunctionInjector.php","line":58,"function":"get","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["trashManager"]},{"function":"OC\\AppFramework\\Bootstrap\\{closure}","class":"OC\\AppFramework\\Bootstrap\\FunctionInjector","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data-ro/nextcloud/lib/private/AppFramework/Bootstrap/FunctionInjector.php","line":68,"function":"array_map","args":[{"__class__":"Closure"},[{"name":"serverContainer","__class__":"ReflectionParameter"},{"name":"logger","__class__":"ReflectionParameter"},{"name":"appManager","__class__":"ReflectionParameter"},"*** sensitive parameter replaced ***"]]},{"file":"/data-ro/nextcloud/lib/private/AppFramework/Bootstrap/BootContext.php","line":52,"function":"injectFn","class":"OC\\AppFramework\\Bootstrap\\FunctionInjector","type":"->","args":[[{"__class__":"OCA\\Files_Trashbin\\AppInfo\\Application"},"registerTrashBackends"]]},{"file":"/data-ro/nextcloud/apps/files_trashbin/lib/AppInfo/Application.php","line":57,"function":"injectFn","class":"OC\\AppFramework\\Bootstrap\\BootContext","type":"->","args":[[{"__class__":"OCA\\Files_Trashbin\\AppInfo\\Application"},"registerTrashBackends"]]},{"file":"/data-ro/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php","line":178,"function":"boot","class":"OCA\\Files_Trashbin\\AppInfo\\Application","type":"->","args":[{"__class__":"OC\\AppFramework\\Bootstrap\\BootContext"}]},{"file":"/data-ro/nextcloud/lib/private/legacy/OC_App.php","line":205,"function":"bootApp","class":"OC\\AppFramework\\Bootstrap\\Coordinator","type":"->","args":["files_trashbin"]},{"file":"/data-ro/nextcloud/lib/private/legacy/OC_App.php","line":138,"function":"loadApp","class":"OC_App","type":"::","args":["files_trashbin"]},{"file":"/data-ro/nextcloud/apps/dav/lib/AppInfo/Application.php","line":124,"function":"loadApps","class":"OC_App","type":"::","args":[["dav"]]},{"file":"/data-ro/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php","line":178,"function":"boot","class":"OCA\\DAV\\AppInfo\\Application","type":"->","args":[{"__class__":"OC\\AppFramework\\Bootstrap\\BootContext"}]},{"file":"/data-ro/nextcloud/lib/private/legacy/OC_App.php","line":205,"function":"bootApp","class":"OC\\AppFramework\\Bootstrap\\Coordinator","type":"->","args":["dav"]},{"file":"/data-ro/nextcloud/lib/private/legacy/OC_App.php","line":138,"function":"loadApp","class":"OC_App","type":"::","args":["dav"]},{"file":"/data-ro/nextcloud/lib/private/legacy/OC_Util.php","line":203,"function":"loadApps","class":"OC_App","type":"::","args":[["filesystem"]]},{"file":"/data-ro/nextcloud/lib/private/User/Session.php","line":554,"function":"setupFS","class":"OC_Util","type":"::","args":["*** sensitive parameter replaced ***"]},{"file":"/data-ro/nextcloud/lib/private/User/Session.php","line":415,"function":"prepareUserLogin","class":"OC\\User\\Session","type":"->","args":[true,"*** sensitive parameter replaced ***"]},{"file":"/data-ro/nextcloud/lib/private/User/Session.php","line":624,"function":"completeLogin","class":"OC\\User\\Session","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data-ro/nextcloud/lib/private/User/Session.php","line":367,"function":"loginWithPassword","class":"OC\\User\\Session","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data-ro/nextcloud/lib/private/Setup.php","line":440,"function":"login","class":"OC\\User\\Session","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/data-ro/nextcloud/core/Command/Maintenance/Install.ph [... cut ...] nsole.php","line":100,"function":"run","class":"OC\\Console\\Application","type":"->","args":[]},{"file":"/data-ro/nextcloud/occ","line":11,"args":["/data-ro/nextcloud/console.php"],"function":"require_once"}],"File":"/data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","Line":111,"CustomMessage":"Could not boot files_versionsCould not resolve OCA\\Files_Versions\\Versions\\IVersionManager! Class can not be instantiated"},"userAgent":"--","version":"21.0.4.1"}

```
</details>

Fooughhy, 2022-01-21:
This is an issue I am getting as well. I am using docker, and it is much more cumbersome to change the "config.php" file because of that, so I'd like to use the UI. But it seems like it's encoding the string which is my domain name to some random string such as the one in OP (my local IP addresses seem to work fine).

vpn, 2022-02-22:
Having this problem as well.

nachoparker, 2022-03-02:
should be fixed in `v1.46.3`

thanks

closed 2022-03-02 by nachoparker