nextcloud/nextcloudpi #1417: Data folder deleted after restoring from data-less backup

mad73923, 2022-01-28:
I just had massive problems restoring my ncp docker instance after some files got corrupted (due to worn-out SD-card!?)

I make data-less ncp backups every 2nd day and use my own rsync script to backup the data folder. The corrupted files were not part of the data folder!
My expectation was to reinstanciate my docker container, mount the same data volume as before, run the ncp-restore and have a happy day.

BUT:
After running the ncp-restore with my data-less archive, I realized all the data was gone.
The problem seems to be these 2 lines in the nc-restore script:
https://github.com/nextcloud/nextcloudpi/blob/dd8fc3dd7d663faf5d916039ee042ef049aaf728/bin/ncp/BACKUPS/nc-restore.sh#L66-67
Here, the nextcloud folder gets deleted and replaced by the one from the backup.
What seems to be good for a backup WITH data can lead to a nightmare in case of a data-less backup. Luckily, I had my own backup of the data folder!!

To fix this, I ran the script step by step and replaced the lines with an rsync command.

rondadon, 2022-03-02:
That is what I have discovered a month ago. Just wanted to restore the "dataless" Backup and all my Data was gone after that.

But luckily I had an external backup.

nachoparker, 2022-03-02:
That is to be expected. Bear in mind that the php code for Nextcloud is also in the volume together with the data. In the non-docker case, if we have moved the data, then it will honor that but that is a quirk. If you restore a dataless backup in the general case you end up with a clean instance, with the exception I mentioned before

rondadon, 2022-03-02:
@nachoparker 

Thank you for clarification! Indeed I am using the Docker-Version of Nextcloudpi. Again thank you for clarification and have a nice day! :)

mad73923, 2022-05-02:
Now it happened! I lost my data by restoring a data-less backup!!
@nachoparker can you please please find a better solution. In case a data-less restore is issued, the data folder should not be touched.
Please reopen!!

mad73923, 2022-06-07:
Reopen please!

theCalcaholic, 2022-06-07:
I'll have another look at this

mad73923, 2022-06-08:
Thank you!
This severe issue made me switch to nextcloud/docker :cry:

ZendaiOwl, 2022-09-28:
+1 Bumping this 

* -[ ] A way to restore a backup recursively, like with rsync or something similar and not just replace the nextcloud directory as is when restoring the backup tar.

theCalcaholic, 2022-09-29:
I have drafted a fix for this: https://github.com/nextcloud/nextcloudpi/pull/1601

It's still untested and can be improved

closed 2023-01-08 by theCalcaholic

theCalcaholic, 2023-01-09:
The change is now included in version v1.50.5

commit a02ee254 (2023-01-05): nc-restore: Fix potential dataloss when restoring backup (fixes #1417)

- nc-restore: Keep old data directory if present
- nc-restore: Temporarily move and restore user files during nc-restore
commit 73b5000c (2023-01-05): nc-restore: Fix potential dataloss when restoring backup (fixes #1417)

- nc-restore: Keep old data directory if present
- nc-restore: Temporarily move and restore user files during nc-restore
PR #1601 (2023-01-05): Fix ncp-restore deleting data directory if still inside /var/www/nextcloud
Fixes #1417

### Todo:

- [x] Change default datadir on docker

### Changes:

- Keep old data directory if present (partial fix for #1417)
- Temporarily move and restore user files during nc-restore (partial fix for #1417)
- Move default location of data directory out of nextcloud directory (partial fix for #1417)
- Improve robustness of the setup process for notify_push
- build-docker.yml: Fix update-test on architectures other than x86
- ssh: Add option to add ssh user to sudo group
- Disable root user for login
- Man [... cut]
commit 97085dbc (2023-01-08): nc-restore: Fix potential dataloss when restoring backup (fixes #1417)

- nc-restore: Keep old data directory if present
- nc-restore: Temporarily move and restore user files during nc-restore