nextcloud/nextcloudpi #1423: noip don't works. duckdns.org does

DanieleMar, 2022-02-06:
**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.46.7
NextCloudPi image    NextCloudPi_10-07-21
OS                   Debian GNU/Linux 10. 5.10.92-v7l+ (armv7l)
automount            no
USB devices          sda sdb 
datadir              /data/nextcloud/data
data in SD           no
data filesystem      ext2/ext3
data disk usage      328G/1.8T
rootfs usage         4.3G/29G
swapfile             /var/swap
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    21.0.4.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               3:38

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "22": "192.168.1.15",
            "11": "mydomain.duckdns.org",
            "1": "172.17.0.3",
            "3": "mydomain.duckdns.org",
            "14": "a9aa9b86d374",
            "6": "192.168.1.15",
            "20": "mydomain.duckdns.org",
            "21": "mydomain.duckdns.org"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "21.0.4.1",
        "overwrite.cli.url": "https:\/\/mydomain.duckdns.org\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
[... 110 lines cut ...]
{"reqId":"Kc1g7QmliQ0pT6ploOuU","level":0,"time":"2022-02-06T08:16:45+00:00","remoteAddr":"","user":"--","app":"files_sharing","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"kx6VbuW11GCoXejEN6hC","level":0,"time":"2022-02-06T08:16:46+00:00","remoteAddr":"","user":"--","app":"files_sharing","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"iAfOmO9LO3drHAbDBBkE","level":0,"time":"2022-02-06T08:16:46+00:00","remoteAddr":"","user":"--","app":"files_sharing","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"pJ8PWeRWus58qsc4p1Cu","level":0,"time":"2022-02-06T08:16:47+00:00","remoteAddr":"","user":"--","app":"files_sharing","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}
{"reqId":"uqCImSn0xJO8PWZPmnTh","level":0,"time":"2022-02-06T08:16:50+00:00","remoteAddr":"","user":"--","app":"files_sharing","method":"","url":"--","message":"/appinfo/app.php is deprecated, use \\OCP\\AppFramework\\Bootstrap\\IBootstrap on the application class instead.","userAgent":"--","version":"21.0.4.1"}

```
</details>

when trying to connect to noip give this errors
```
Can't get status for eth0@if20. (19)
Incorrect password or username
```

But mine credentials are correct
So i tried with duckdns and goes well

I don't know if it correlate with this message in ncp-report
```
port check 80        closed
port check 443       closed
```

My port on router are open though

DanieleMar, 2022-02-06:
maybe because my password has some invalid character

nachoparker, 2022-03-01:
it works here, so you might be right, maybe ask around in the forums

closed 2022-03-01 by nachoparker