nextcloud/nextcloudpi #1435: nc-hdd-monitor does not work

vcbranco, 2022-03-04:
Tested with Debian 11 x86_64 curl installation because I don’t have any disk available that supports smart to connecto to the rpi at the moment

Error:

Running nc-hdd-monitor

=== START OF ENABLE/DISABLE COMMANDS SECTION ===
SMART Enabled.

Failed to enable unit: Unit file smartd.service does not exist.
HDD monitor enabled

nachoparker, 2022-03-04:
Thanks for reporting. The service name has changed in Bullseye. Fixed in `v1.46.7`

In the bullseye version, they are disallowing VMs though, so a VM is not good for testing

https://github.com/smartmontools/smartmontools/issues/62

closed 2022-03-04 by nachoparker

vcbranco, 2022-03-05:
Is not a VM

nachoparker, 2022-03-05:
ah cool, then smartd will work. In any case the original bug was detected an fixed, thanks!

vcbranco, 2022-03-05:
Yes, is working.
Thanks

madscientist23333, 2023-08-17:
Hi,

it seems I am running into a similar issue on my armbian system using the install.sh as described in ncp documentation.

In total the install.sh has an output of ~1770 lines ending like
```
[...]
Preparing to unpack .../miniupnpc_2.2.1-1_arm64.deb ...
Unpacking miniupnpc (2.2.1-1) ...
Setting up libminiupnpc17:arm64 (2.2.1-1) ...
Setting up miniupnpc (2.2.1-1) ...
Processing triggers for man-db (2.9.4-2) ...
Processing triggers for libc-bin (2.31-13+deb11u6) ...
Installing nc-hdd-monitor
Installing nc-hdd-test
Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://security.debian.org bullseye-security InRelease
Hit:3 http://deb.debian.org/debian bullseye-updates InRelease
Hit:4 http://deb.debian.org/debian bullseye-backports InRelease
Hit:5 https://packages.sury.org/php bullseye InRelease
Hit:6 http://armbian.hosthatch.com/apt bullseye InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
smartmontools is already the newest version (7.2-1).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Failed to enable unit: Unit file smartd.service does not exist.
#
```
The script comes to a hard stop, in some intermediate stage after **Failed to enable unit: Unit file smartd.service does not exist.**
Unfortunately I am totally lost in the install sequence when trying to follow what triggers this.
This is what I experienced already some months ago and got lost.

If someone ( @nachoparker  ) had a hint for me, this would be very much appreciated.

BR
Uwe

madscientist23333, 2023-08-17:
Found that there is no smartd (service) present, but smartmontools.
Unfortunately I am unable to identify, where the relevant part in the installer procedure is, certainly not within install.sh as I can't find a matching phrase.

/Uwe