nextcloud/nextcloudpi #1436: Can not enable SSH

vcbranco, 2022-03-04:
SSH error (same as issue #356)

Running SSH
New password: Retype new password: passwd: password updated successfully
Wrong salt length: 3 bytes when 8 <= n <= 16 expected.
Synchronizing state of ssh.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable ssh
The user pi is using the default password. Refusing to activate SSH
SSH disabled

zedocrob, 2022-03-04:
Same here on Odroid HC2 image
NextCloudPi_OdroidHC2_03-01-22.img

nachoparker, 2022-03-04:
the default KDF for `/etc/passwd` in Debian Bullseye is now `yescrypt`, but there is an outstanding Debian bug preventing us from validating that we are not using the default password

https://www.mail-archive.com/debian-bugs-dist@lists.debian.org/msg1837456.html

nachoparker, 2022-03-05:
I see no other option other than to disable this check until the bug is fixed

rubyone, 2022-03-06:
same here, what's the fix?

Cyber-Goetz, 2022-03-06:
@rubyone Can you please tell us your solution?

Recvi, 2022-03-08:
Is there a workaround for this? I'm currently recovering my NCP instance after failure and after using the latest image `NextCloudPi_OdroidHC2_03-01-22.img` I get the same error:

```
New password: Retype new password:
passwd: password updated successfully
Wrong salt length: 5 bytes when 8 <= n <= 16 expected.
Synchronizing state of ssh.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable ssh
The user root is using the default password. Refusing to activate SSH
SSH disabled
```

Recvi, 2022-03-08:
@nachoparker thank you for your response, that did it!

closed 2022-07-27 by theCalcaholic