nextcloud/nextcloudpi #1437: The value 'bullseye-security' is invalid for APT::Default-Release as such a release is not available in the sources

Dennis-1991, 2022-03-06:
Hello,

I try to install NextcloudPi, but it failed. The installation stopps at "The value 'bullseye-security' is invalid for APT::Default-Release as such a release is not available in the sources". I hope you are able to help me.. Here is the hole cmd:

Linux raspberrypi 5.10.92-v7+ #1514 SMP Mon Jan 17 17:36:39 GMT 2022 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Mar  6 02:41:02 2022 from 192.168.22.36
pi@raspberrypi:~ $ wget https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh
--2022-03-06 02:44:21--  https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 2606:50c0:8000::154, 2606:50c0:8001::154, 2606:50c0:8002::154, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|2606:50c0:8000::154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 3483 (3.4K) [text/plain]
Saving to: ‘install.sh’

install.sh                    100%[=================================================>]   3.40K  --.-KB/s    in 0.001s

2022-03-06 02:44:21 (6.01 MB/s) - ‘install.sh’ saved [3483/3483]

pi@raspberrypi:~ $ sudo bash install.sh
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
ca-certificates is already the newest version (20210119).
lsb-release is already the newest version (11.1.0+rpi1).
lsb-release set to manually installed.
sudo is already the newest version (1.9.5p2-3).
The following additional packages will be installed:
  git-man liberror-perl
Suggested packages:
  git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk gitweb git-cvs git-mediawiki git-svn
The following NEW packages will be installed:
  git git-man liberror-perl
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 6,583 kB of archives.
After this operation, 33.1 MB of additional disk space will be used.
Get:1 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian bullseye/main armhf liberror-perl all 0.17029-1 [31.0 kB]
Get:2 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian bullseye/main armhf git-man all 1:2.30.2-1 [1,827 kB]
Get:3 http://mirror.pyratelan.org/raspbian/raspbian bullseye/main armhf git armhf 1:2.30.2-1 [4,724 kB]
Fetched 6,583 kB in 3s (2,287 kB/s)
Selecting previously unselected package liberror-perl.
(Reading database ... 41532 files and directories currently installed.)
Preparing to unpack .../liberror-perl_0.17029-1_all.deb ...
Unpacking liberror-perl (0.17029-1) ...
Selecting previously unselected package git-man.
Preparing to unpack .../git-man_1%3a2.30.2-1_all.deb ...
Unpacking git-man (1:2.30.2-1) ...
Selecting previously unselected package git.
Preparing to unpack .../git_1%3a2.30.2-1_armhf.deb ...
Unpacking git (1:2.30.2-1) ...
Setting up liberror-perl (0.17029-1) ...
Setting up git-man (1:2.30.2-1) ...
Setting up git (1:2.30.2-1) ...
Processing triggers for man-db (2.9.4-2) ...
Getting build code...
Cloning into '/tmp/nextcloudpi.DXqpLF/nextcloudpi'...
remote: Enumerating objects: 10357, done.
remote: Counting objects: 100% (1366/1366), done.
remote: Compressing objects: 100% (572/572), done.
remote: Total 10357 (delta 897), reused 1135 (delta 767), pack-reused 8991
Receiving objects: 100% (10357/10357), 3.19 MiB | 5.55 MiB/s, done.
[... 220 lines cut ...]
remote: Compressing objects: 100% (590/590), done.
remote: Total 10357 (delta 895), reused 1122 (delta 749), pack-reused 8991
Receiving objects: 100% (10357/10357), 3.19 MiB | 6.16 MiB/s, done.
Resolving deltas: 100% (7241/7241), done.

Installing NextCloudPi...
Installing lamp
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
apt-utils is already the newest version (2.2.4).
cron is already the newest version (3.0pl1-137).
curl is already the newest version (7.74.0-1.3+deb11u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
apache2 is already the newest version (2.4.52-1~deb11u2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
E: The value 'bullseye-security' is invalid for APT::Default-Release as such a release is not available in the sources
pi@raspberrypi:~ $

Marcus-En, 2022-03-06:
Hi Dennis,
I think, I stuck on the same spot yesterday. The install script doesnt run on (a fresh) bullseye, by now.
It seems that the pros are working on it.
I have a running instance on buster, so I decided to wait and see
BR

vcbranco, 2022-03-06:
Did you try the official NextCloudPi image?
https://ownyourbits.com/downloads/

allea78, 2022-03-06:
Hello,
I've the same problem: I've installed a fresh RasperriPi OS (2022-01-28-raspios-bullseye-armhf-lite.img) and I've run "curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash".
(https://ownyourbits.com/nextcloudpi/#download)

The official NextCloudPI is for 64bit systems and my old Raspberri is 32.

Thank you

closed 2022-03-06 by nachoparker

m-breitbach, 2022-03-07:
I'm running into the same problem when trying to 'sudo ncp-dist-upgrade' from Buster (kernel: 5.10.63-v7l+, NCP: 1.46.8).

(Is there a way to roll back the unfinished upgrade?)

allea78, 2022-03-08:
Adding the line:
`deb http://security.debian.org/ bullseye-security/updates main contrib non-free`
in /etc/apt/source.list
(and running `sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 112695A0E562B32A` for the keys)
the error disappear, but now I've downgrades errors whit some php packages and the service php7.4-fpm doesn't start.

I've try also whit a Buster fresh installations, but in this case the NextCloud Pi install script fail because it is not support this version of Raspberri Pi OS.

nitind, 2022-03-17:
I think the underlying problem is that the target release, `bullseye-security`, concocted in `etc/library.sh` doesn't exist for Raspbian like it does for Debian.

Iolaum, 2022-03-19:
I just hit this bug when updating raspbian on Rpi4 on the latest ncp version.

ovpc, 2022-03-22:
On Proxmox VM sudo ncp-dist-upgrade ends with;
> Running hooks in /etc/ca-certificates/update.d...
done.
Processing triggers for initramfs-tools (0.140) ...
update-initramfs: Generating /boot/initrd.img-5.10.0-10-amd64
Processing triggers for dbus (1.12.20-2) ...
Reading package lists... Done
E: The value 'bullseye-security' is invalid for APT::Default-Release as such a release is not available in the sources
Something went wrong. Fix it and try again

Siam1205, 2022-03-31:
I am no experienced Linux user but I ran into the issue today after I updated from buster to bullseye which broke my whole nextcloud installation. As a result, I reinstalled a new fresh buster image in order to set up a fresh ncp instance. However, I failed to install it on the fresh buster system because of the "unsupported dist" problem. I then found this issue but I'm not sure what the solution to this problem should be even though there are some suggestions.

I ran the command `curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash` after I had added `deb http://security.debian.org/ bullseye-security/updates main contrib non-free` and the key respectively. The last lines resulted in

```
Setting up samba-common-bin (2:4.13.13+dfsg-1~deb11u3) ...
Checking smb.conf with testparm
testparm: symbol lookup error: testparm: undefined symbol: samba_gnutls_weak_crypto_allowed, version SAMBA_4.13.13_DEBIAN
dpkg: error processing package samba-common-bin (--configure):
 installed samba-common-bin package post-installation script subprocess returned error exit status 127
dpkg: dependency problems prevent configuration of samba:
 samba depends on samba-common-bin (= 2:4.13.13+dfsg-1~deb11u3); however:
  Package samba-common-bin is not configured yet.

dpkg: error processing package samba (--configure):
 dependency problems - leaving unconfigured
Processing triggers for systemd (241-7~deb10u8+rpi1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for libc-bin (2.31-13+rpi1+deb11u2) ...
Errors were encountered while processing:
 samba-common-bin
 samba
E: Sub-process /usr/bin/dpkg returned an error code (1)
You have new mail in /var/mail/root
```

and now I don't know exactly what do do to set up the instance and load my backups.

Siam1205, 2022-03-31:
I tried what @Iolaum said but I didn't work. Firstly, there is no file `/usr/local/etc/ncp-recommended.cfg`. Because of that, I tried using the existing file `/usr/local/etc/ncp.cfg`. However, when I run line 76 and everything below manually, as mentioned, it fails with the same error I presented above.

Sujeom, 2022-04-01:
Try adding "deb http://security.debian.org/debian-security bullseye-security main contrib non-free" to your /etc/apt/sources.list file.

Siam1205, 2022-04-02:
Hi,
thank you for your answer. Meanwhile, I tried to use the official NextCloudPi image like @vcbranco suggested. With this approach, you don't have to install NextCloudPi anymore because it obviously already exists on the image. I was able to initialize it just to reload my backups which worked surprisingly well.

In fact, there were a few things and minor bugs I encountered, for example that my contacts were lost at first or that I was not able to enable SSH via the web interface (there's an issue well documented by other users in other issues or threads) but there was nothing I could not handle somehow.

The cloud now seems to work as expected and I'm grateful again, that the NCP project exists. Why? because during a frustration period, I looked for alternatives. Simply, there are none worth mentioning.

nachoparker, 2022-04-03:
@Siam1205 run ncp-update and then you'll be able to enable SSH again

Andy-Meissner, 2022-04-09:
Why is this issue closed ?

The workaround suggested by @nitind seemed to work for me.

theCalcaholic, 2022-04-09:
@Andy-Meissner This should be fixed in #1457

The fix can be applied as such:

```sh
sudo ncp-update fix/dist-upgrade-raspbian
sudo ncp-dist-upgrade
```

ovpc, 2022-04-14:
Processing triggers for initramfs-tools (0.140) ...
update-initramfs: Generating /boot/initrd.img-5.10.0-13-amd64
Processing triggers for dbus (1.12.20-2) ...
Reading package lists... Done
E: The value 'bullseye-security' is invalid for APT::Default-Release as such a release is not available in the sources
Something went wrong. Fix it and try again
oliver@ncpvm ~ $ sudo ncp-update fix/dist-upgrade-raspbian
INFO: updating to development branch 'fix/dist-upgrade-raspbian'
Downloading updates
warning: Could not find remote branch fix/dist-upgrade-raspbian to clone.
fatal: Remote branch fix/dist-upgrade-raspbian not found in upstream origin
No internet connectivity

So this is not fixed yet for VM on Proxmox, it seems.
Rerunning ncp-dist-upgrade gives no output at all even after manually adding 
 "deb http://security.debian.org/debian-security bullseye-security main contrib non-free" to sources-list

theCalcaholic, 2022-04-14:
@ovpc The fix is now in master. Just do a normal `ncp-update` and then `ncp-dist-upgrade`

ovpc, 2022-04-15:
@theCalcaholic Thanks and sorry I did not post ncp version, but it was already up to date at v1.47.1
Let me know if need more log or info, no problem running it again, It has been restored from a previously taken snapshot.

theCalcaholic, 2022-04-15:
Okay, did you attempt to execute `ncp-dist-upgrade`? You shouldn't need the workaround anymore.

theCalcaholic, 2022-04-15:
Ah okay... As you're not running ncp on a raspberry pi, this is very likely a separate issue (or the same issue affecting another base image). Could you please create a new issue with all the information asked by the GH issue template? I'll need that to find another workaround.

NicholasKirchner, 2022-05-22:
This remains an issue with a fresh installation.  I just took the latest fresh Raspberry Pi OS, and attempted to run install.sh.  It failed with this error message.  @nitind 's workaround worked to dodge the issue and continue the install.

useronegit, 2022-06-02:
the problem still exists, fresh raspberry pi debian lite install + tryd nextcloudpi install with command

eddietheeagle7, 2022-06-05:
With the suggestion from @NicholasKirchner the installation worked.

But when calling the web site I got a message that I am trying to access via an untrusted domain.
After adding the domain to the trusted domain I got only an empty "Index Of" page, nothing listed:
[image]

This didn't change when I removed the adress from the trusted domains.

Is it possible to uninstall everything the install script installed to try it again?

teledyn, 2023-11-16:
It is now 2023, I am running NextcloudPi version	v1.53.0 updated to Nextcloud version 27.1.3.2 on a rpi4B, and an upgrade to Bookworm has replicated this problem.