nextcloud/nextcloudpi #1438: Docker php version.

dragon2611, 2022-03-06:
Seems php was updated to 7.4 in the latest docker container, but it's looking for for php-7.3-fpm.sock which causes a 503 error.

I did a dirty hack to get nextcloud up quickly(won't persist on a restart) but this needs fixing in the webserver configuration.

**System information**

root@60ee61e01ccf:/# ncp-report
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.47.1
NextCloudPi image    NextCloudPi_docker_03-01-22
OS                   Debian GNU/Linux 11. 5.10.93+truenas (x86_64)
automount            no
USB devices          none
datadir              /data/nextcloud/data
data in SD           no
data filesystem      zfs
data disk usage      9.3G/352G
rootfs usage         714M/54G
swapfile             /dev/dm-2
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    23.0.2.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               10days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "22": "192.168.42.1",
            "11": "145.224.67.58",
            "1": "172.24.0.2",
            "3": "60ee61e01ccf",
            "21": "192.168.5.132",
            "20": "192.168.42.1",
            "14": "60ee61e01ccf",
            "6": "192.168.5.132"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "23.0.2.1",
[... 98 lines cut ...]
[Sun Mar 06 13:22:58.568772 2022] [proxy:error] [pid 108:tid 139746015708928] (2)No such file or directory: AH02454: FCGI: attempt to connect to Unix domain socket /run/php/php7.3-fpm.sock (*) failed
tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory
root@60ee61e01ccf:/# php
php         php-fpm7.4  php7.4      phpdismod   phpenmod    phpquery
root@60ee61e01ccf:/# php
php         php-fpm7.4  php7.4      phpdismod   phpenmod    phpquery
root@60ee61e01ccf:/# php-fpm7.4
[06-Mar-2022 13:26:57] ERROR: Another FPM instance seems to already listen on /run/php/php7.4-fpm.sock
[06-Mar-2022 13:26:57] ERROR: FPM initialization failed
root@60ee61e01ccf:/# cat /var/log/apache2/error.log
[Sun Mar 06 13:22:53.632648 2022] [ssl:error] [pid 102:tid 139746225233216] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=localhost / issuer: CN=localhost / serial: 77013E54EC5E5AF8723BFFC6B78B9031967A2436 / notbefore: Mar  1 23:30:13 2022 GMT / notafter: Feb 27 23:30:13 2032 GMT]
[Sun Mar 06 13:22:53.632682 2022] [ssl:error] [pid 102:tid 139746225233216] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Sun Mar 06 13:22:53.646888 2022] [ssl:error] [pid 103:tid 139746225233216] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=localhost / issuer: CN=localhost / serial: 77013E54EC5E5AF8723BFFC6B78B9031967A2436 / notbefore: Mar  1 23:30:13 2022 GMT / notafter: Feb 27 23:30:13 2032 GMT]
[Sun Mar 06 13:22:53.646919 2022] [ssl:error] [pid 103:tid 139746225233216] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Sun Mar 06 13:22:53.650318 2022] [mpm_event:notice] [pid 103:tid 139746225233216] AH00489: Apache/2.4.52 (Debian) OpenSSL/1.1.1k configured -- resuming normal operations
[Sun Mar 06 13:22:53.650403 2022] [core:notice] [pid 103:tid 139746225233216] AH00094: Command line: '/usr/sbin/apache2'
[Sun Mar 06 13:22:57.869887 2022] [proxy:error] [pid 108:tid 139746015708928] (2)No such file or directory: AH02454: FCGI: attempt to connect to Unix domain socket /run/php/php7.3-fpm.sock (*) failed
[Sun Mar 06 13:22:57.869965 2022] [proxy_fcgi:error] [pid 108:tid 139746015708928] [client 192.168.5.120:50284] AH01079: failed to make connection to backend: httpd-UDS
[Sun Mar 06 13:22:58.568772 2022] [proxy:error] [pid 108:tid 139746015708928] (2)No such file or directory: AH02454: FCGI: attempt to connect to Unix domain socket /run/php/php7.3-fpm.sock (*) failed
[Sun Mar 06 13:22:58.568837 2022] [proxy_fcgi:error] [pid 108:tid 139746015708928] [client 192.168.5.120:50284] AH01079: failed to make connection to backend: httpd-UDS

root@60ee61e01ccf:/# ln -s /run/php/php7.4-fpm.sock /run/php/php7.3-fpm.sock
root@60ee61e01ccf:/# ls /run/php
php7.3-fpm.sock  php7.4-fpm.pid  php7.4-fpm.sock
root@60ee61e01ccf:/#

nachoparker, 2022-03-06:
you need to

- stop and remove your container
- pull the latest image
- start the latest image with your old volume

please use the forums for further support

closed 2022-03-06 by nachoparker

dragon2611, 2022-03-06:
@nachoparker did that before posting here, did it again just now, same problem.

I don't specify a version tag and I pruned images after RM'ing the container and confirmed the image wasn't present.

dragon2611, 2022-03-06:
@nachoparker /etc/apache2/conf-available has a config for php7.3-fpm.conf where as php 7.4 is installed in the container, I'm yet to determine why this is the case.

nachoparker, 2022-03-07:
you are correct, you can update NC and restart the container. A fix has been pushed.

Or you can wait until I push the fixed docker image `v1.46.9` and stop + remove + pull + start container.