nextcloud/nextcloudpi #1440: nc-encrypt: fuse device not found / fusermount exited with code 256

kojid0, 2022-03-09:
I'm running nextcloudpi for a year in a docker container (portainer). Now I want to have a second container for nextcloudpi with nc-encrypt enabled.
When I do so, I get the following error with a red dot next to the apply-button:

```
[ nc-encrypt ] (Wed Mar 9 13:37:34 UTC 2022)
Maintenance mode enabled
/bin/fusermount: fuse device not found, try 'modprobe fuse' first
fuse.NewServer failed: fusermount exited with code 256
{"reqId":"vqqTVbUIVR1QL3dCjfNb","level":2,"time":"2022-03-09T13:37:35+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"23.0.2.1"}
{"reqId":"vqqTVbUIVR1QL3dCjfNb","level":2,"time":"2022-03-09T13:37:35+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"23.0.2.1"}
{"reqId":"vqqTVbUIVR1QL3dCjfNb","level":3,"time":"2022-03-09T13:37:36+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fopen(/data/nextcloud/data/data_dir_writability_test_6228ada00b844.tmp): failed to open stream: Permission denied at /data/nextcloud/lib/private/legacy/OC_Util.php#792","userAgent":"--","version":"23.0.2.1","exception":{"Exception":"Error","Message":"fopen(/data/nextcloud/data/data_dir_writability_test_6228ada00b844.tmp): failed to open stream: Permission denied at /data/nextcloud/lib/private/legacy/OC_Util.php#792","Code":0,"Trace":[{"function":"onError","class":"OC\\Log\\ErrorHandler","type":"::"},{"file":"/data/nextcloud/lib/private/legacy/OC_Util.php","line":792,"function":"fopen"},{"file":"/data/nextcloud/lib/private/Console/Application.php","line":157,"function":"checkServer","class":"OC_Util","type":"::"},{"file":"/data/nextcloud/console.php","line":98,"function":"loadCommands","class":"OC\\Console\\Application","type":"->"},{"file":"/data/nextcloud/occ","line":11,"args":["/data/nextcloud/console.php"],"function":"require_once"}],"File":"/data/nextcloud/lib/private/Log/ErrorHandler.php","Line":92,"CustomMessage":"--"}}
Your data directory is invalid
Ensure there is a file called ".ocdata" in the root of the data directory.

Your data directory is not writable
Permissions can usually be fixed by giving the webserver write access to the root directory. See https://docs.nextcloud.com/server/23/go.php?to=admin-dir_permissions.

An unhandled exception has been thrown:
Exception: Environment not properly prepared. in /data/nextcloud/lib/private/Console/Application.php:164
Stack trace:
#0 /data/nextcloud/console.php(98): OC\Console\Application->loadCommands()
#1 /data/nextcloud/occ(11): require_once('/data/nextcloud...')
#2 {main}{"reqId":"vqqTVbUIVR1QL3dCjfNb","level":3,"time":"2022-03-09T13:37:36+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fopen(/data/nextcloud/data/nextcloud.log): failed to open stream: Permission denied at /data/nextcloud/lib/private/Log/File.php#84","userAgent":"--","version":"23.0.2.1"}
```

Not quite sure what causes the issue. 
When I access ncp web interface (:4443), it redirects to :4443/decrypt with a blank page. 
When I access nc, it says that the data directory is invalid and not writable.
When I reboot the container and then access nc, it asks for the password to decrypt but it says it's wrong.

Here's a step-by-step video:

https://user-images.githubusercontent.com/24477557/157534705-0c74cbdb-8d05-4d09-8009-1ae05b116e31.mp4

The 2 NextCloudPi Containers are managed by Portainer. Portainer is installed on Proxmox.
ncp version: 1.46.9
nc version: 23.0.2.1

**System information (after nc-encrypt enabled+restart)**
<details>
<summary>NextCloudPi diagnostics</summary>

```

{"reqId":"LEZT2HlJkseU2IkIPk0Q","level":2,"time":"2022-03-09T20:54:05+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"23.0.2.1"}
{"reqId":"LEZT2HlJkseU2IkIPk0Q","level":2,"time":"2022-03-09T20:54:05+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"23.0.2.1"}
Nextcloud is in maintenance mode, hence the database isn't accessible.
Cannot perform any command except 'maintenance:mode --off'

{"reqId":"LEZT2HlJkseU2IkIPk0Q","level":3,"time":"2022-03-09T20:54:05+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fopen(/data/nextcloud/data/data_dir_writability_test_622913ed385bc.tmp): failed to open stream: Permission denied at /data/nextcloud/lib/private/legacy/OC_Util.php#792","userAgent":"--","version":"23.0.2.1","exception":{"Exception":"Error","Message":"fopen(/data/nextcloud/data/data_dir_writability_test_622913ed385bc.tmp): failed to open stream: Permission denied at /data/nextcloud/lib/private/legacy/OC_Util.php#792","Code":0,"Trace":[{"function":"onError","class":"OC\\Log\\ErrorHandler","type":"::"},{"file":"/data/nextcloud/lib/private/legacy/OC_Util.php","line":792,"function":"fopen"},{"file":"/data/nextcloud/lib/private/Console/Application.php","line":157,"function":"checkServer","class":"OC_Util","type":"::"},{"file":"/data/nextcloud/console.php","line":98,"function":"loadCommands","class":"OC\\Console\\Application","type":"->"},{"file":"/data/nextcloud/occ","line":11,"args":["/data/nextcloud/console.php"],"function":"require_once"}],"File":"/data/nextcloud/lib/private/Log/ErrorHandler.php","Line":92,"CustomMessage":"--"}}
{"reqId":"LEZT2HlJkseU2IkIPk0Q","level":3,"time":"2022-03-09T20:54:05+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fopen(/data/nextcloud/data/nextcloud.log): failed to open stream: Permission denied at /data/nextcloud/lib/private/Log/File.php#84","userAgent":"--","version":"23.0.2.1"}
NextCloudPi version  v1.46.9
NextCloudPi image    NextCloudPi_docker_03-07-22
OS                   Debian GNU/Linux 11. 5.11.22-1-pve (x86_64)
automount            no
USB devices          none
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      overlayfs
data disk usage      122G/147G
rootfs usage         122G/147G
swapfile             /dev/dm-0
dbdir                /data/database
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
[... 63 lines cut ...]
[Wed Mar 09 20:49:53.447740 2022] [mpm_event:notice] [pid 109:tid 140473502526784] AH00489: Apache/2.4.52 (Debian) OpenSSL/1.1.1k configured -- resuming normal operations
[Wed Mar 09 20:49:53.447748 2022] [core:notice] [pid 109:tid 140473502526784] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: cannot open '/data/nextcloud/data/nextcloud.log' for reading: No such file or directory

```
</details>

nachoparker, 2022-03-09:
You probably need to mount `/dev/fuse` and make sure the container has the correct capabilities to use FUSE

nachoparker, 2022-03-09:
maybe something like this

```
docker run -d --rm \
           --device /dev/fuse \
           --cap-add SYS_ADMIN \
           --security-opt apparmor:unconfined \
           ...
```

kojid0, 2022-03-09:
> You probably need to mount `/dev/fuse` and make sure the container has the correct capabilities to use FUSE (SYS_ADMIN)

I was thinking about the SYS_ADMIN capability as well and tried it before. No luck.
Mounting /dev/fuse with your example does nothing.
[image]

Same issue when I activate nc-encrypt after that

ZendaiOwl, 2022-08-20:
> > You probably need to mount `/dev/fuse` and make sure the container has the correct capabilities to use FUSE (SYS_ADMIN)
> 
> I was thinking about the SYS_ADMIN capability as well and tried it before. No luck.
> Mounting /dev/fuse with your example does nothing.
> [image]

Hmm, this command have not been executed

aghosh0605, 2023-10-15:
The same problem happened with me when I used `nc-encrypt`, and now the nextclould is not opening with a certificate issue. Also, the admin panel that runs on port 4443 stopped working. Rebooted my server, but no result.