nextcloud/nextcloudpi #1441: Extract of NextCloudPi_RPi_03-02-22.tar failed

MarvvanPal, 2022-03-11:
I'm not able to extract the NextCloudPi_RPi_03-02-22.tar to get the .img-File
[image]
Downloadlink: https://ownyourbits.com/downloads/NextCloudPi_RPi_03-02-22/NextCloudPi_RPi_03-02-22.tar.bz2

Getting the .img-file for Bananapi is possible.

Can someone check the archive NextCloudPi_RPi_03-02-22.tar.bz2 from the mentioned source?

Workaround: Use Torrent via Transmission.

OS: WIndows 10 21H2
7Zip: 21.07

adeekshith, 2022-04-25:
I was able to extract it fine but the image does not load on my Raspberry Pi 2B

heimwege, 2022-05-06:
I have exactly the same issue with the archive

Jeanoo, 2022-05-06:
Look here  https://help.nextcloud.com/t/impossible-to-write-nextcloudpi-image-3-02-22-to-sd-card/138388/12

heimwege, 2022-05-06:
Thanks for the quick reply @Jeanoo 
Actually I tried both 7zip (21.07) and winrar (6.11) and none of them worked for me 🙈 
Additionally tried Bandzip because it was mentioned in the link you provided but got the same error.

Jeanoo, 2022-05-06:
you have to download the file again with the torrent link. The normal download link, it always failed in my pc

closed 2022-07-23 by theCalcaholic