nextcloud/nextcloudpi #1442: When setting up no-ip: "warning: No domain found. Defaulting to 'nextcloudpi'", https://nextcloudpi doesn't work

teackot, 2022-03-14:
https://nextcloudpi/ URL doesn't work

Output:
`[ no-ip ] (Mon Mar 14 18:20:31 GMT 2022)

Auto configuration for Linux client of no-ip.com.

Only one host [**********.ddns.net] is registered to this account.
It will be used.

New configuration file '/usr/local/etc/no-ip2.conf' created.

Failed to enable unit: Unit /run/systemd/generator.late/noip2.service is transient or generated.
warning: No domain found. Defaulting to 'nextcloudpi'
System config value trusted_domains => 3 set to string nextcloudpi
System config value overwrite.cli.url set to string https://nextcloudpi/
System config value trusted_proxies => 11 set to string 127.0.0.1
System config value trusted_proxies => 12 set to string ::1
System config value trusted_proxies => 13 set to string nextcloudpi
System config value trusted_proxies => 14 set to empty string
✓ redis is configured
✓ push server is receiving redis messages
✓ push server can load mount info from database
✓ push server can connect to the Nextcloud server
✓ push server is a trusted proxy
✓ push server is running the same version as the app
configuration saved
noip DDNS enabled`

theCalcaholic, 2022-04-12:
That sounds like a configuration issue in your no-ip.com account. Do you have a domain registered there?

theCalcaholic, 2022-07-23:
closed due to inactivity

closed 2022-07-23 by theCalcaholic

HervePruvost, 2022-12-23:
same issue. set the domain in the right field on NCP WebUI, but keep getting this message.
So the Pi remains not reachable under this domain name.
Additionally the System Info panel shows ports 80 and 443 always closed altough they are opened in my Fritzbox.