nextcloud/nextcloudpi #1443: config.php modify dbpassword to 'default' after electricity outage(hard reset). After restart database is not accessible since not correct password

Brahypelma, 2022-03-16:
**System information**

Brahypelma, 2022-03-16:
So after electricity outage I faced that issue log:`PHP Fatal error:  Uncaught Doctrine\\DBAL\\Exception: Failed to connect to the database: An exception occurred in the driver: SQLSTATE[HY000] [1045] Access denied for user 'ncadmin'@'localhost' (using password: YES)PHP Fatal error:  Uncaught Doctrine\\DBAL\\Exception: Failed to connect to the database: An exception occurred in the driver: SQLSTATE[HY000] [1045] Access denied for user 'ncadmin'@'localhost' (using password: YES)`

Second step is check config.php -> and vualya , `dbpassword` somehow overwrited to `default`. After password restored back - everything is ok

my ncp-report: ```

NextCloudPi version  v1.46.9
NextCloudPi image    NextCloudPi_10-07-21
OS                   Debian GNU/Linux 10. 5.15.25-rockchip64 (aarch64)
automount            no
USB devices          sda 
datadir              /data/nextcloud/data
data in SD           no
data filesystem      ext2/ext3
data disk usage      281G/916G
rootfs usage         5.6G/15G
swapfile             none
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    23.0.2.1
HTTPD service        up
PHP service          down
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        open
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               11:36

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "1": "172.17.0.2",
            "6": "10.8.0.8",
            "3": "localhost",
            "11": "82.148.18.23",
            "22": "10.8.0.8",
            "14": "96b68b9d8cc6",
            "": "localhost"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "23.0.2.1",
        "overwrite.cli.url": "https:\/\/localhost\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/data\/nextcloud\/data\/tmp",
        "mail_smtpmode": "sendmail",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "preview_max_x": "2048",
        "preview_max_y": "2048",
        "jpeg_quality": "60",
        "overwriteprotocol": "https",
        "maintenance": false,
        "loglevel": 2,
        "theme": "",
        "trusted_proxies": "***REMOVED SENSITIVE VALUE***",
        "logfile": "\/data\/nextcloud\/data\/nextcloud.log",
        "data-fingerprint": "62d09ee618712c774891763868797317"
    }
}

```
</details>

theCalcaholic, 2022-04-12:
Thanks for reporting! Unfortunately, this will be hard to reproduce and fix. If we receive more reports like this we will look into it, but for now I'll consider this a fluke.

closed 2022-04-12 by theCalcaholic