nextcloud/nextcloudpi #1446: Installation of dnsmasq breaks DNS-resolution

theCalcaholic, 2022-03-25:
I'm currently trying to use the curl installer to install NCP on a fesh debian 11 server. However, the installer consistently fails after installing the dnsmasq app, because the system (namely apt) can't resolve any domains anymore.

I've dug into the issue and don't yet understand completely what's causing it, but I'm 90% sure dnsmasq only breaks in conjunction with something the installer did before and I also found a workaround. So here are my findings:

1. Installing dnsmasq on the fresh debian system works without issues. So the problem must be caused by some config change or package installed by the installer.
2. The dnsmasq service fails with a very generic error message which didn't help me to get to it's cause:
```sh
Job for dnsmasq.service failed because a timeout was exceeded.
See "systemctl status dnsmasq.service" and "journalctl -xe" for details.
invoke-rc.d: initscript dnsmasq, action "start" failed.
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
    Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled; vendor preset: enabled)
    Active: failed (Result: timeout) since Fri 2022-03-25 12:27:44 UTC; 27ms ago
    Process: 27746 ExecStartPre=/etc/init.d/dnsmasq checkconfig (code=exited, status=0/SUCCESS)
    Process: 27750 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status=0/SUCCESS)
    Process: 27755 ExecStartPost=/etc/init.d/dnsmasq systemd-start-resolvconf (code=killed, signal=TERM)
  Main PID: 27754 (code=exited, status=0/SUCCESS)
    CPU: 82ms

Mar 25 12:26:14 debian-11 dnsmasq[27754]: reading /run/dnsmasq/resolv.conf
Mar 25 12:26:14 debian-11 dnsmasq[27754]: using nameserver 2a01:4ff:ff00::add:2#53
Mar 25 12:26:14 debian-11 dnsmasq[27754]: using nameserver 2a01:4ff:ff00::add:1#53
Mar 25 12:26:14 debian-11 dnsmasq[27754]: using nameserver 185.12.64.1#53
Mar 25 12:26:14 debian-11 dnsmasq[27754]: using nameserver 185.12.64.2#53
Mar 25 12:27:44 debian-11 systemd[1]: dnsmasq.service: start-post operation timed out. Terminating.
Mar 25 12:27:44 debian-11 dnsmasq[27754]: exiting on receipt of SIGTERM
Mar 25 12:27:44 debian-11 systemd[1]: dnsmasq.service: Control process exited, code=killed, status=15/TERM
Mar 25 12:27:44 debian-11 systemd[1]: dnsmasq.service: Failed with result 'timeout'.
Mar 25 12:27:44 debian-11 systemd[1]: Failed to start dnsmasq - A lightweight DHCP and caching DNS server.
```
3. This error can be avoided, by stopping systemd-resolve after installing dnsmasq and then starting dnsmasq (it can be stopped afterwards and DNS resolution will still work). I've implemented this workaround and am going to create a PR where it can be discussed further. However, the workaround implies, that there is some sort of conflict between systemd-resolve and dnsmasq (maybe listening on the same port? Allthough I couldn't verify that).

theCalcaholic, 2022-03-25:
Another workaround is installing dnsmasq before running the curl installer, btw

I've found a mention of a bug that looks exactly what I'm observing which is nearly one year old: 

https://help.nextcloud.com/t/ncp-script-job-for-dnsmasq-service-failed-because-a-timeout-was-exceeded/113744

CaixaNegraPT, 2022-03-25:
Yeap, the script always failed in my installations, here is another example:

https://help.nextcloud.com/t/debian-10-clean-errors-running-ncp-curl-installer/91249/10

theCalcaholic, 2022-03-26:
@CaixaNegraPT If you're still affected by this (and didn't apply a workaround yourself) can you please try running the following in order to install directly from devel? The bug shouldn't occur anymore.

```sh
export BRANCH=devel
curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/devel/install.sh | bash
```

@nachoparker In your absence I have merged the workaround into devel. I made sure that it is only applied when the installation of dnsmasq fails, so it shouldn't cause issues when it's not needed. I'll keep this issue open for you to see it.

theCalcaholic, 2022-03-30:
> great, thank you. Let me know if I can tag/merge

I  already went ahead and merged it to devel, since it's just a minor change :)
Everything looks good, but I didn't want to merge it to master without hearing from you.

nvdw, 2022-04-09:
Hi,
I think I ran into an issue relating to this fix. I tried to create a docker nextcloud-pi armhf image were the 'systemd-resolved' service was not found and the image creation failed and exited:
```
# build/build-docker.sh armhf
...
#17 54.23 Applying workaround for dnsmasq bug (compare issue #1446)
#17 54.24 systemd-resolved: unrecognized service
#17 ERROR: executor failed running [/bin/bash -c set -e; rm -f ncp-web/{wizard.cfg,ncp-web.cfg}; touch /.ncp-image; touch /.docker-image; apt-get update; apt-get install --no-install-recommends -y wget ca-certificates; source /usr/local/etc/library.sh; cd /tmp/ncp-build/; install_app ncp.sh; mv /usr/local/etc/ncp-config.d/nc-init-copy.cfg /usr/local/etc/ncp-config.d/nc-init.cfg; run_app_unsafe bin/ncp/CONFIG/nc-init.sh; sed -i 's|data-ro|data|' /data-ro/nextcloud/config/config.php; sed -i 's|/media/USBdrive|/data/backups|' /usr/local/etc/ncp-config.d/nc-backup.cfg; sed -i 's|/media/USBdrive|/data/backups|' /usr/local/etc/ncp-config.d/nc-backup-auto.cfg; run_app_unsafe post-inst.sh; cd /; rm -r /tmp/ncp-build; rm /usr/local/etc/ncp-config.d/nc-init.cfg; rm -rf /usr/share/man/*; rm -rf /usr/share/doc/*; rm -f /var/log/alternatives.log /var/log/apt/*; rm /var/cache/debconf/*-old; echo "${ncp_ver}" > /usr/local/etc/ncp-version]: exit code: 1
```
To fix it temporary I made the following change:
```
diff --git a/bin/ncp/NETWORKING/dnsmasq.sh b/bin/ncp/NETWORKING/dnsmasq.sh
index 368403e..8903d91 100644
--- a/bin/ncp/NETWORKING/dnsmasq.sh
+++ b/bin/ncp/NETWORKING/dnsmasq.sh
@@ -17,7 +17,7 @@ install()
   service dnsmasq status > /dev/null 2>&1 || rc=$?
   [[ $rc -eq 3 ]] && {
     echo "Applying workaround for dnsmasq bug (compare issue #1446)"
-    service systemd-resolved stop
+    service systemd-resolved stop || :
     service dnsmasq start
   }
   update-rc.d dnsmasq disable
```
But maybe it can be better changed with e.g.:
```
service --status-all 2>&1 | grep -q ' systemd-resolved$' && service systemd-resolved stop
```

theCalcaholic, 2022-04-09:
@nvdw Thanks for reporting. This portion will only be executed if dnsmasq fails to start during installation. Since this workaround is only for a specific bug related to systemd-resolved, I didn't deem it necessary to do any additional checks. Do you still have the logs before the portion you posted (or do you know what caused dnsmasq to fail in the first place)?

nvdw, 2022-04-11:
Hi @theCalcaholic , thanks for you reply.

The following messages are show before checking if dnsmasq is running:
```
#17 50.32 Unpacking dnsmasq (2.85-1) ...
#17 50.46 Setting up runit-helper (2.10.3) ...
#17 50.49 Setting up libnfnetlink0:armhf (1.0.1-3+b1) ...
#17 50.52 Setting up libnetfilter-conntrack3:armhf (1.0.8-3) ...
#17 50.55 Setting up dnsmasq-base (2.85-1) ...
#17 50.94 Setting up dnsmasq (2.85-1) ...
#17 51.51 invoke-rc.d: could not determine current runlevel
#17 51.52 invoke-rc.d: policy-rc.d denied execution of start.
#17 51.55 Processing triggers for libc-bin (2.31-13+deb11u3) ...
#17 51.71 Applying workaround for dnsmasq bug (compare issue #1446)
#17 51.72 systemd-resolved: unrecognized service
```
Not sure what causes the `invoke-rc.d` command to fail.

Attached a more complete log file, with some minor changes (added some echo commands for extra logging):
[ncpi.log](https://github.com/nextcloud/nextcloudpi/files/8463428/ncpi.log)

closed 2022-04-11 by theCalcaholic