nextcloud/nextcloudpi #1447: Nextcloud 24 Support

sunjam, 2022-03-25:
For internal testing and notes. **Do not expect support!!!!!!** Please do report testing issues here.
[3/25/2022 nc24beta1](https://github.com/nextcloud/server/releases/tag/v24.0.0beta1) is gone to testing.
[Track which apps are incompatible with version 24 on the forum](https://help.nextcloud.com/t/any-apps-not-your-compatible-with-24/136275)
- #1365 
- https://github.com/nextcloud/server/pull/29286 in version 24.
  - https://github.com/nextcloud/server/pull/29285 introduced a warning.
- https://github.com/nextcloud/server/issues/29287 in development for future releases.

Krawei, 2022-06-13:
Any progress on this topic? Thanks for your great work! :)

sunjam, 2022-06-16:
> Any progress on this topic? Thanks for your great work! :)

Testers are needed to simply try out the upgrade process.  Make an export of your ncp config and backup your installation, then try upgrading and let us know whether things break.

Ideally, spin up a testing instance that is not your production ncp.

svenb1234, 2022-06-22:
I have updated to 23.0.4, .0.5 and 0.6 without problems. So at least those minor updates could be enabled for the NCP users which are still kept on .0.2 by the official NCP tools.

theCalcaholic, 2022-06-25:
@svenb1234 You are right. However, currently I'm still very occupied with providing the missing images for the latest release and getting nextcloudpi.com and the documentation back online. 
After that I'll probably look into upgrading NC.

theCalcaholic, 2022-07-12:
@svenb1234 Update to 23.0.6 is out now

theCalcaholic, 2022-07-12:
Thanks a lot @sunjam, for creating these. It's a great help!

theCalcaholic, 2022-07-21:
I just tested the upgrade to NC 24.0.3 and didn't have any problems.

If you tested upgrading to NC 24.0.3 as well and had no issues, please leave a reaction to this message - otherwise please leave a message describing the problems you encountered.

svenb1234, 2022-07-21:
No issues either.

FadeFx, 2022-08-05:
> I just tested the upgrade to NC 24.0.3 and didn't have any problems.
> 
> If you tested upgrading to NC 24.0.3 as well and had no issues, please leave a reaction to this message - otherwise please leave a message describing the problems you encountered.
> 
> Make sure you have backups. Also, be aware, that this disables the NextcloudPi and preview-generator apps. I'm looking into ensuring their compatibility

i did update to 23.0.6 without issues a while ago already...

theCalcaholic, 2022-08-13:
NC 24 support is now considered stable and can be activated using `ncp-update devel`. However, I want to fix the bugs evolving around nc-datadir and nc-encrypt-data before merging it into master (see #1517)

theCalcaholic, 2022-08-16:
Closing as NC 24 support has been rolled out in version v1.49.0

closed 2022-08-16 by theCalcaholic

mbousq, 2022-08-21:
Just upgraded to Nextcloud 24 and it went flawlessly. Many thanks !

svenb1234, 2022-09-06:
Update to 24.0.4. went well.