nextcloud/nextcloudpi #1452: nextcloudpi does not update nextcloud to v

Krawei, 2022-04-03:
Dear team,

the overview webpage of my nextcloupi instance tells me
"new version available: Nextcloud 23.0.3".

But the ncp-update-nextcloud tells me, that there's no newer version available than 23.0.2.

Could you please fix that? That would be great :).

Thanks for your hard work!

svenb1234, 2022-04-11:
Neytcloudpi is always late with minor updates. Supposedly because they are tested before applied. Not sure though what exactly that test involves and what the criteria for a successful test are.

theCalcaholic, 2022-04-12:
These tests mainly check if Nextcloud will work on the system or requires adjustments and if all mandatory Nextcloud apps are available and working on the new NC version.

You can always upgrade manually at the risk of breaking things via the Admin-Webinterface->nc-update (and enter the speciifc version you want to upgrade to).

closed 2022-04-12 by theCalcaholic