nextcloud/nextcloudpi #1453: Update to 1.46.10 from 1.46.9 broken NextCloud instance!

fdagostino97, 2022-04-04:
After a fresh installation i update the NextcloudPi Panel to 1.46.10 and it broken the Nextcloud Installation.

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.46.10
NextCloudPi image    NextCloudPi_docker_03-07-22
OS                   Debian GNU/Linux 11. 5.15.30-v8+ (aarch64)
automount            no
USB devices          sda 
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      22G/59G
rootfs usage         22G/59G
swapfile             /var/swap
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    23.0.2.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
internet check       no
port check 80        closed
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               17min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "0be1fe61f025",
            "11": "***.***.***.***",
            "1": "172.17.0.2",
            "14": "0be1fe61f025",
            "20": "***.xyz",
            "21": "***.***.***.***",
            "22": "***.***.***.***"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "23.0.2.1",
        "overwrite.cli.url": "https:\/\/0be1fe61f025\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/var\/www\/nextcloud\/data\/tmp",
        "mail_smtpmode": "sendmail",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "preview_max_x": "2048",
        "preview_max_y": "2048",
        "jpeg_quality": "60",
        "overwriteprotocol": "https",
        "trusted_proxies": "***REMOVED SENSITIVE VALUE***",
        "maintenance": false
    }
}

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Mon Apr 04 12:36:04.327669 2022] [ssl:error] [pid 108:tid 547828625424] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Mon Apr 04 12:36:04.351001 2022] [ssl:error] [pid 109:tid 547828625424] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=localhost / issuer: CN=localhost / serial: 4E265C163E1EF063FE306A5791B452C9E96ACC89 / notbefore: Mar  2 00:09:22 2022 GMT / notafter: Feb 28 00:09:22 2032 GMT]
[Mon Apr 04 12:36:04.351063 2022] [ssl:error] [pid 109:tid 547828625424] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Mon Apr 04 12:36:04.352665 2022] [core:warn] [pid 109:tid 547828625424] AH00098: pid file /var/run/apache2/apache2.pid overwritten -- Unclean shutdown of previous Apache run?
[Mon Apr 04 12:36:04.357127 2022] [mpm_event:notice] [pid 109:tid 547828625424] AH00489: Apache/2.4.53 (Debian) OpenSSL/1.1.1n configured -- resuming normal operations
[Mon Apr 04 12:36:04.357269 2022] [core:notice] [pid 109:tid 547828625424] AH00094: Command line: '/usr/sbin/apache2'
[Mon Apr 04 12:41:43.723422 2022] [ssl:error] [pid 108:tid 547916845072] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=localhost / issuer: CN=localhost / serial: 4E265C163E1EF063FE306A5791B452C9E96ACC89 / notbefore: Mar  2 00:09:22 2022 GMT / notafter: Feb 28 00:09:22 2032 GMT]
[Mon Apr 04 12:41:43.723962 2022] [ssl:error] [pid 108:tid 547916845072] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Mon Apr 04 12:41:43.749562 2022] [ssl:error] [pid 109:tid 547916845072] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=localhost / issuer: CN=localhost / serial: 4E265C163E1EF063FE306A5791B452C9E96ACC89 / notbefore: Mar  2 00:09:22 2022 GMT / notafter: Feb 28 00:09:22 2032 GMT]
[Mon Apr 04 12:41:43.749615 2022] [ssl:error] [pid 109:tid 547916845072] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Mon Apr 04 12:41:43.751715 2022] [core:warn] [pid 109:tid 547916845072] AH00098: pid file /var/run/apache2/apache2.pid overwritten -- Unclean shutdown of previous Apache run?
[Mon Apr 04 12:41:43.754993 2022] [mpm_event:notice] [pid 109:tid 547916845072] AH00489: Apache/2.4.53 (Debian) OpenSSL/1.1.1n configured -- resuming normal operations
[Mon Apr 04 12:41:43.755143 2022] [core:notice] [pid 109:tid 547916845072] AH00094: Command line: '/usr/sbin/apache2'
[Mon Apr 04 12:51:38.039873 2022] [ssl:error] [pid 109:tid 547800182800] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=localhost / issuer: CN=localhost / serial: 4E265C163E1EF063FE306A5791B452C9E96ACC89 / notbefore: Mar  2 00:09:22 2022 GMT / notafter: Feb 28 00:09:22 2032 GMT]
[Mon Apr 04 12:51:38.039937 2022] [ssl:error] [pid 109:tid 547800182800] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Mon Apr 04 12:51:38.061565 2022] [ssl:error] [pid 110:tid 547800182800] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=localhost / issuer: CN=localhost / serial: 4E265C163E1EF063FE306A5791B452C9E96ACC89 / notbefore: Mar  2 00:09:22 2022 GMT / notafter: Feb 28 00:09:22 2032 GMT]
[Mon Apr 04 12:51:38.061612 2022] [ssl:error] [pid 110:tid 547800182800] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Mon Apr 04 12:51:38.063885 2022] [core:warn] [pid 110:tid 547800182800] AH00098: pid file /var/run/apache2/apache2.pid overwritten -- Unclean shutdown of previous Apache run?
[Mon Apr 04 12:51:38.067020 2022] [mpm_event:notice] [pid 110:tid 547800182800] AH00489: Apache/2.4.53 (Debian) OpenSSL/1.1.1n configured -- resuming normal operations
[Mon Apr 04 12:51:38.067157 2022] [core:notice] [pid 110:tid 547800182800] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

tail: cannot open '/var/log/mysql/*.log' for reading: No such file or directory

```
</details>

svenb1234, 2022-04-11:
I think you will have to be a bit more verbose about what broke.

theCalcaholic, 2022-07-12:
closing due to inactivity

closed 2022-07-12 by theCalcaholic