nextcloud/nextcloudpi #1455: Warn Or Prevent non-root ncp-config operations

DoloresHA, 2022-04-05:
**Description**  

Warn or prevent users from launching `ncp-config` as non-root, or warn/prevent them from running certain scripts as non-root.

When trying to restore a backup using ncp-config, if a fool such as myself launches `ncp-config` without invoking `sudo ncp-config`, the operation will ultimately fail. This is extra-brutal in the event of a backup restore, as this failure can happen hours into the operation. The line that burned me is here: https://github.com/nextcloud/nextcloudpi/blob/be30663c7a4b86a4b91bf39b3b2d49c87f53d95c/bin/ncp/BACKUPS/nc-restore.sh#L76
**Example**

nachoparker, 2022-04-06:
```$ ncp-config
Must be run as root. Try 'sudo ncp-config'```

closed 2022-04-06 by nachoparker

DoloresHA, 2022-04-06:
Interesting. I see the same prompt when I invoke `ncp-config` as another user (e.g. pi), but if I am logged in as root and run `ncp-report` (no sudo) I do not get the prompt. But, I am quite confident that `service redis-server restart` fails when I run ncp-backup.

theCalcaholic, 2022-04-07:
If you are root, these commands should succeed