nextcloud/nextcloudpi #1456: upgrade to Bullseye always popping up, but already upgraded

dusscloud, 2022-04-08:
I am using NextcloudPI on a Raspbbery PI for a while now, and I recently upgraded to bullseye, as prompted when login :
`Update to bullseye available. Type 'sudo ncp-dist-upgrade' to upgrade`
The upgrade seemed to have partially worked, I had to install all PHP 7.4 modules manually anyway everything seems to work now.

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```

The issue is I still have the message "Update to bullseye available. Type 'sudo ncp-dist-upgrade' to upgrade" whenever I log in,
 and if I run the ncp-dist-upgrade command, I have a lot of errros while trying to locate php7.3 packages, and above all it ends with 
```
E: The value 'bullseye-security' is invalid for APT::Default-Release as such a release is not available in the sources
Something went wrong. Fix it and try again
```
I am using ncp 1.46.10

theCalcaholic, 2022-04-09:
I had the same issue on raspbian. I'll look into it soon

theCalcaholic, 2022-04-09:
@dusscloud Can you test executing the following commands?

```sh
sudo ncp-update fix/dist-upgrade-raspbian
sudo ncp-dist-upgrade
```

dusscloud, 2022-04-09:
> @dusscloud Can you test executing the following commands?
> 
> ```shell
> sudo ncp-update fix/dist-upgrade-raspbian
> sudo ncp-dist-upgrade
> ```

it worked ! :+1: 
thanks

PR #1457 (2022-04-11): ncp-dist-upgrade: Fix invalid target for apt-get on Raspbian
Should fix #1456