nextcloud/nextcloudpi #1459: document how to automate nextcloudpi docker container update / provide a script or service for doing so

jerabaul29, 2022-04-10:
It is not clear to me how to keep the docker container (when this is used as an installation method) up to date. I guess the process will involve i) downloading the new docker image, ii) stopping the running container providing nextcloudpi service iii) spawn a new container based on the updated image to restore nextcloudpi service.

I guess for automation purpose that there are a few challenges:

- how to actually automate the process, pull the container exactly when needed to stay up to date and safe,
- how to do so with as little service disruption as possible / if possible without any service disruption

I wonder if this would require both i) some additional documentation (I could not find some, but maybe I missed something), ii) some scripting. I am not a specialist of this at all, but maybe setting up a kubernetes instance on the RPi to run the container would allow to keep things running cleaner?

dvpfig, 2022-06-14:
The manual update of containers is not specific of NCP. One should look for documentation on docker containers. 
I think depending on how your docker run command looks like (or docker compose) the list of steps can be different. 

But an example would be as below. And to automate this process, check this project: https://github.com/containrrr/watchtower

* Find container id or container name. And stop and remove the container. 
`docker ps`
`docker stop <container id>`
`docker rm <container id>`

* Find image and remove it.  
`docker images`
`docker image rm <image id>`

* Run the same container command (it will download a new image from the tag)
`doker run .... <tag:latest>`

theCalcaholic, 2022-08-03:
The container can actually update itself just like any other NCP installation (via `ncp-update` or auto updates). Alternatively, you can replace the container image with the latest version.

closed 2022-08-03 by theCalcaholic

ZendaiOwl, 2022-09-26:
Just to add some info if someone comes here reading

Important
Note that the image must also have been updated on the docker hub for the internal upgrade to work persistently, else you will reverse part of the upgrade when you restart the container and it pulls the previous version from the docker registry.