nextcloud/nextcloudpi #1460: 2 Modifications in ncp-dist-upgrade to run successfully

CarlosCastaneda, 2022-04-12:
After many attempts (costing me countless hours of time) to successfully run ncp-dist-upgrade, I ran into 2 problems, which I solved as follows: Uninstalling gcc-8-base and commenting out the bullseye-security line. With both changes, I was able to successfully upgrade, although I'm not sure now that the upgrade is correct without bullseye-security.

The upgrade process also took hours in the end with various manual confirmations (needrestart). This is not something you just do in between and every single attempt is extremely time consuming.
```
# install latest distro
apt-get update
apt-get remove gcc-8-base -y                   <-- added this line
apt-get dist-upgrade -y
# install latest PHP version
release_new=$(jq -r '.release' < "${new_cfg}")
# the default repo in bullseye is bullseye-security
#release_new="${release_new}-security"        <-- commented this one
php_ver_new=$(jq -r '.php_version' < "${new_cfg}")

```

theCalcaholic, 2022-04-12:
Thanks for the report. Can you please add your system information? Are you running ncp on docker/VM/bare metal (if so, which system)? What's the output of `cat /etc/*-release`?

Also, what exactly happened before you applied these changes? I assume the second issue is already fixed (see https://github.com/nextcloud/nextcloudpi/pull/1457). Please let me know if you're not running Raspbian.

CarlosCastaneda, 2022-04-12:
Thanks for your quick reply. I'm running on bare metal. My first issue was running into the problem I described [here](https://help.nextcloud.com/t/ncp-dist-upgrade-fails-repeatedly/136846) in the hope to get an appropriate answer.

The other issue I faced might be this one #1437[ ](https://github.com/nextcloud/nextcloudpi/issues/1437) or that one [#1457](https://github.com/nextcloud/nextcloudpi/issues/1457) you mentioned above.
Now I'm not sure if my installation is valid without referring to bullseye-security!?

If you need more infos please let me know.

Here's the output of cat /etc/*-release

```
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```
and the output of ncp-report

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.47.0
NextCloudPi image    NextCloudPi_07-21-19
OS                   Raspbian GNU/Linux 11. 5.15.32-v7l+ (armv7l)
automount            yes
USB devices          sda sdb sdc
datadir              /media/myCloudDrive1/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      47G/1.9T
rootfs usage         6.0G/15G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    23.0.2.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               6:06

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
[... 111 lines cut ...]
```

{"reqId":"3NhcZEfvgDUbQEGF6s6z","level":3,"time":"2022-04-12T10:00:02+00:00","remoteAddr":"","user":"--","app":"core","method":"","url":"--","message":"Error while running background job (class: OCA\\DAV\\BackgroundJob\\EventReminderJob, arguments: )","userAgent":"--","version":"23.0.2.1","exception":{"Exception":"TypeError","Message":"Argument 1 passed to OCA\\DAV\\CalDAV\\Reminder\\ReminderService::parseCalendarData() must be of the type string, null given, called in /var/www/nextcloud/apps/dav/lib/CalDAV/Reminder/ReminderService.php on line 122","Code":0,"Trace":[{"file":"/var/www/nextcloud/apps/dav/lib/CalDAV/Reminder/ReminderService.php","line":122,"function":"parseCalendarData","class":"OCA\\DAV\\CalDAV\\Reminder\\ReminderService","type":"->"},{"file":"/var/www/nextcloud/apps/dav/lib/BackgroundJob/EventReminderJob.php","line":66,"function":"processReminders","class":"OCA\\DAV\\CalDAV\\Reminder\\ReminderService","type":"->"},{"file":"/var/www/nextcloud/lib/private/BackgroundJob/Job.php","line":51,"function":"run","class":"OCA\\DAV\\BackgroundJob\\EventReminderJob","type":"->"},{"file":"/var/www/nextcloud/lib/private/BackgroundJob/TimedJob.php","line":58,"function":"execute","class":"OC\\BackgroundJob\\Job","type":"->"},{"file":"/var/www/nextcloud/cron.php","line":128,"function":"execute","class":"OC\\BackgroundJob\\TimedJob","type":"->"}],"File":"/var/www/nextcloud/apps/dav/lib/CalDAV/Reminder/ReminderService.php","Line":666,"CustomMessage":"Error while running background job (class: OCA\\DAV\\BackgroundJob\\EventReminderJob, arguments: )"}}
{"reqId":"pynvOm9LmV5ELluLQ36E","level":3,"time":"2022-04-12T10:05:02+00:00","remoteAddr":"","user":"--","app":"core","method":"","url":"--","message":"Error while running background job (class: OCA\\DAV\\BackgroundJob\\EventReminderJob, arguments: )","userAgent":"--","version":"23.0.2.1","exception":{"Exception":"TypeError","Message":"Argument 1 passed to OCA\\DAV\\CalDAV\\Reminder\\ReminderService::parseCalendarData() must be of the type string, null given, called in /var/www/nextcloud/apps/dav/lib/CalDAV/Reminder/ReminderService.php on line 122","Code":0,"Trace":[{"file":"/var/www/nextcloud/apps/dav/lib/CalDAV/Reminder/ReminderService.php","line":122,"function":"parseCalendarData","class":"OCA\\DAV\\CalDAV\\Reminder\\ReminderService","type":"->"},{"file":"/var/www/nextcloud/apps/dav/lib/BackgroundJob/EventReminderJob.php","line":66,"function":"processReminders","class":"OCA\\DAV\\CalDAV\\Reminder\\ReminderService","type":"->"},{"file":"/var/www/nextcloud/lib/private/BackgroundJob/Job.php","line":51,"function":"run","class":"OCA\\DAV\\BackgroundJob\\EventReminderJob","type":"->"},{"file":"/var/www/nextcloud/lib/private/BackgroundJob/TimedJob.php","line":58,"function":"execute","class":"OC\\BackgroundJob\\Job","type":"->"},{"file":"/var/www/nextcloud/cron.php","line":128,"function":"execute","class":"OC\\BackgroundJob\\TimedJob","type":"->"}],"File":"/var/www/nextcloud/apps/dav/lib/CalDAV/Reminder/ReminderService.php","Line":666,"CustomMessage":"Error while running background job (class: OCA\\DAV\\BackgroundJob\\EventReminderJob, arguments: )"}}
{"reqId":"mnXjOKNVMIVEsthyMrKd","level":3,"time":"2022-04-12T10:10:02+00:00","remoteAddr":"","user":"--","app":"core","method":"","url":"--","message":"Error while running background job (class: OCA\\DAV\\BackgroundJob\\EventReminderJo [... cut ...] ype":"->"},{"file":"/var/www/nextcloud/lib/private/BackgroundJob/TimedJob.php","line":58,"function":"execute","class":"OC\\BackgroundJob\\Job","type":"->"},{"file":"/var/www/nextcloud/cron.php","line":128,"function":"execute","class":"OC\\BackgroundJob\\TimedJob","type":"->"}],"File":"/var/www/nextcloud/apps/dav/lib/CalDAV/Reminder/ReminderService.php","Line":666,"CustomMessage":"Error while running background job (class: OCA\\DAV\\BackgroundJob\\EventReminderJob, arguments: )"}}

```
</details>

theCalcaholic, 2022-04-12:
Okay, your second issue is indeed #1437 / #1457 (they are duplicates of each other). It should be fixed by now.

The second issue is actually an upstream issue (with raspbian's apt archives)... Are you using no-ip? Because apparently that's the only place we are using libc-dev (as dependency for the no-ip client).

CarlosCastaneda, 2022-04-12:
> Okay, your second issue is indeed #1437 / #1457 (they are duplicates of each other). It should be fixed by now.
> 
So, does this mean that my upgrade with reference to bullseye instead of bullseye-security is valid?

> The second issue is actually an upstream issue (with raspbian's apt archives)... Are you using no-ip? Because apparently that's the only place we are using libc-dev (as dependency for the no-ip client).

No, I'm not using no-ip, it's AVM's service myfritz.

theCalcaholic, 2022-04-12:
Yeas, your upgrade should be valid

closed 2022-04-12 by theCalcaholic

CarlosCastaneda, 2022-04-12:
> Yeas, your upgrade should be valid

Thanks, that's good to know :-)

pkra, 2022-10-20:
I'm not sure how helpful it is but since https://github.com/nextcloud/nextcloudpi/issues/1460#issuecomment-1096798794 suggested new info might be useful, I wanted to share my experience.

I'm an inexperienced ncp user running on a pi4b (with no changes beyond external storage for user files). This has been running perfectly for about a year -- thank you very much for such a beginner friendly NC experience!

I finally took the leap to run ncp-dist-upgrade and I think I ran into the same problem.

ncp-dist-upgrade ended in:

```
The following packages have unmet dependencies:
 libc6-dev : Breaks: libgcc-8-dev (< 8.4.0-2~) but 8.3.0-6+rpi1 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
Something went wrong. Fix it and try again
```

[I can share the full log if it's helpful]

After finding this thread I removed gcc-8-base manually. Afterwards, ncp-dist-upgrade immediately returned (after the y/N prompt).

So I ran `apt-get dist-upgrade` which seemed to complete fine. However, the NCP admin panel was partially and NC completely broken (giving me raw php). I could still run nc-update which seemed to complete fine but without effect.

I messed around for a while (e.g., installing php) to little avail. Ultimately, I manually stepped through the last parts of ncp-dist-upgrade (i.e., https://github.com/nextcloud/nextcloudpi/blob/21d9894e63d0e6d732d925adbdb09256ae982c31/bin/ncp-dist-upgrade#L78 pp) which seems to have worked.

I'm still not sure why nc-dist-upgrade couldn't get far enough after the initial problem and dist-upgrade. That would have helped save time but ultimately it was also enough for me to recover -- thank you again for all the work that goes into ncp!