nextcloud/nextcloudpi #1462: NC-Update --> useradd: cannot open /etc/shadow

Tommy-01, 2022-04-14:
**System information**
Nextcloudpi Docker on Qnap Nas

Problem description:
The NC-Update Process throws an error an exits.
Hope someone can help. Thanks in advantage!

`Update NextCloudPi

[ nc-update ] (Thu Apr 14 18:46:47 UTC 2022)
Downloading updates
Performing updates
Installing metrics
--2022-04-14 18:47:09-- https://github.com/theCalcaholic/ncp-metrics-exporter/releases/download/v1.1.0/ncp-metrics-exporter-x86_64
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response...
302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/467263699/9cb2ac49-2d78-4bb5-8630-a4dd062edfa0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220414%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220414T184709Z&X-Amz-Expires=300&X-Amz-Signature=ddb132bb16d90b8f6a98a7a84265fb6ebd6cf0650cce8d8f788925ef0e383147&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=467263699&response-content-disposition=attachment%3B%20filename%3Dncp-metrics-exporter-x86_64&response-content-type=application%2Foctet-stream [following]
--2022-04-14 18:47:09-- https://objects.githubusercontent.com/github-production-release-asset-2e65be/467263699/9cb2ac49-2d78-4bb5-8630-a4dd062edfa0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220414%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220414T184709Z&X-Amz-Expires=300&X-Amz-Signature=ddb132bb16d90b8f6a98a7a84265fb6ebd6cf0650cce8d8f788925ef0e383147&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=467263699&response-content-disposition=attachment%3B%20filename%3Dncp-metrics-exporter-x86_64&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.110.133, 185.199.111.133, 185.199.108.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.110.133|:443... connected.
HTTP request sent, awaiting response...
200 OK
Length: 8038504 (7.7M) [application/octet-stream]
Saving to: '/usr/local/lib/ncp-metrics/ncp-metrics-exporter'

0K .......... .......... .......... .......... .......... 0% 2.40M 3s
50K .......... .......... .......... .......... .......... 1% 7.40M 2s
100K .......... .......... .......... .......... .......... 1% 4.15M 2s
150K .......... .......... .......... .......... .......... 2% 7.30M 2s
200K .......... .......... .......... .......... .......... 3% 6.13M 2s
250K .......... .......... .......... .......... .......... 3% 7.43M 2s
300K .......... .......... .......... .......... .......... 4% 7.56M 1s
350K .......... .......... .......... .......... .......... 5% 7.18M 1s
400K .......... .......... .......... .......... .......... 5% 7.26M 1s
450K .......... .......... .......... .......... .......... 6% 7.06M 1s
500K .......... .......... .......... .......... .......... 7% 7.22M 1s
550K .......... .......... .......... .......... .......... 7% 7.57M 1s
600K .......... .......... .......... .......... .......... 8% 5.47M 1s
650K .......... .......... .......... .......... ......
.... 8% 7.50M 1s
700K .......... .......... .......... .......... .......... 9% 7.34M 1s
750K .......... .......... .......... .......... .......... 10% 7.28M 1s
800K .......... .......... .......... .......... .......... 10% 7.13M 1s
850K .......... .......... .......... .......... .......... 11% 6.43M 1s
900K .......... .......... .......... .......... .......... 12% 7.41M 1s
950K .......... .......... .......... .......... .......... 12% 7.13M 1s
1000K .......... .......... .......... .......... .......... 13% 7.40M 1s
1050K .......... .......... .......... .......... .......... 14% 7.40M 1s
1100K .......... .......... .......... .......... .......... 14% 7.39M 1s
1150K .......... .......... .......... .......... .......... 15% 3.21M 1s
1200K .......... .......... .......... .......... .......... 15% 20.0M 1s
1250K .......... .......... .......... .......... .......... 16% 10.6M 1s
1300K .......... .......... .......... .......... .......... 17% 7.10M 1s
1350K .......... .......... .......... .......... .......... 17% 6.69M 1s
1400K .......... .......... .......... .......... .......... 18% 7.46M 1s
1450K .......... .......... .......... .......... .......... 19% 7.50M 1s
1500K .......... .......... .......... .......... .......... 19% 7.17M 1s
1550K .......... .......... .......... .......... .......... 20% 5.47M 1s
1600K .......... .......... .......... .......... .......... 21% 7.29M 1s
1650K .......... .......... .......... .......... .......... 21% 7.56M 1s
1700K .......... .......... .......... .......... .......... 22% 6.41M 1s
1750K .......... .......... .......... .......... .......... 22% 7.76M 1s
1800K .......... .......... .......... .......... .......... 23% 2.32M 1s
1850K .......... .......... .......... .......... .......... 24% 20.2M 1s
1900K .......... .......... .......... .......... .......... 24% 20.2M 1s
1950K .......... .......... .......... .......... .......... 25% 15.8M 1s
2000K .......... .......... .......... .......... .......... 26% 7.26M 1s
2050K .......... .......... .......... .......... .......... 26% 7.19M 1s
[... 201 lines cut ...]
useradd: cannot open /etc/shadow
adduser: `/usr/sbin/useradd -d /var/lib/prometheus -g prometheus -s /usr/sbin/nologin -u 105 prometheus' returned error code 1. Exiting.
dpkg: error processing package prometheus-node-exporter (--configure):
installed prometheus-node-exporter package post-installation script subprocess returned error exit status 1
Setting up dbus (1.12.20-2) ...
useradd: cannot open /etc/shadow
adduser: `/usr/sbin/useradd -d /nonexistent -g messagebus -s /usr/sbin/nologin -u 105 messagebus' returned error code 1. Exiting.
dpkg: error processing package dbus (--configure):
installed dbus package post-installation script subprocess returned error exit status 1
Setting up daemon (0.7-1) ...
Setting up libipc-run-perl (20200505.0-1) ...
Setting up libtime-duration-perl (1.21-1) ...
Setting up libtimedate-perl (2.3300-2) ...
dpkg: dependency problems prevent configuration of prometheus-node-exporter-collectors:
prometheus-node-exporter-collectors depends on prometheus-node-exporter; however:
Package prometheus-node-exporter is not configured yet.

dpkg: error processing package prometheus-node-exporter-collectors (--configure):
dependency problems - leaving unconfigured
Setting up moreutils (0.65-1) ...
Errors were encountered while processing:
prometheus-node-exporter
dbus
prometheus-node-exporter-collectors
E: Sub-process /usr/bin/dpkg returned an error code (1)`

theCalcaholic, 2022-04-16:
Hey, thanks for the bug report. I'm not sure how this happened - I tried to take precautions to prevent that particular bug.

To find out, what's the issue and fix it, could you please do the following:

1.  Open the file /usr/local/bin/ncp/SYSTEM/metrics.sh (e.g. with `sudo nano /usr/local/bin/ncp/SYSTEM/metrics.sh`)
2. Add the following line to the top in line 2 (right after **but not before** `#!/bin/bash`):
    ```sh
    set -x
    ```
3. Execute the following commands and post the output here.

```sh
sudo su -
source /usr/local/etc/library.sh
install_app metrics
```

rondadon, 2022-04-17:
Hi there! I have the same issue when trying to update from v1.46.10 to 1.747.1 ..

I have run your suggested commands @theCalcaholic and I still get the error that /etc/shadow could not be edited.
See the following(a part of the ourput) after running your suggested commands in the ncp Docker container:

```
...
+ chmod +x /usr/local/lib/ncp-metrics/ncp-metrics-exporter
+ sed -i 's|status_of_proc "$DAEMON" "$NAME" ${PIDFILE:="-p ${PIDFILE}"}|status_of_proc ${PIDFILE:+-p "$PIDFILE"} "$DAEMON" "$NAME"|' /lib/init/init-d-script
+ [[ '' == 1 ]]
+ apt_install_with_recommends prometheus-node-exporter
+ apt-get update --allow-releaseinfo-change
Hit:1 http://security.debian.org/debian-security buster/updates InRelease
Hit:2 http://deb.debian.org/debian buster InRelease
Hit:3 http://deb.debian.org/debian buster-updates InRelease
Reading package lists... Done
+ DEBIAN_FRONTEND=noninteractive
+ apt-get install -y -o Dpkg::Options::=--force-confdef -o Dpkg::Options::=--force-confold prometheus-node-exporter
Reading package lists... Done
Building dependency tree       
Reading state information... Done
prometheus-node-exporter is already the newest version (0.17.0+ds-3+b11).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up dbus (1.12.20-0+deb10u1) ...
useradd: cannot open /etc/shadow
adduser: `/usr/sbin/useradd -d /nonexistent -g messagebus -s /usr/sbin/nologin -u 105 messagebus' returned error code 1. Exiting.
dpkg: error processing package dbus (--configure):
 installed dbus package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 dbus
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

theCalcaholic, 2022-04-17:
@rondadon Can you give me the full output? I need the lines before that.

My commands are not a fix for this issue, but a way to gather more information about it in order to understand it.

rondadon, 2022-04-17:
Sorry for not providing the whole output. Had done it over Portainer in Webshell and couldn't copy the whole output. Now I've done it with docker exec. Btw. I have the same issue on a newly installed NCP Docker Container.

**NCP Server No1 (Docker Container) Info:**

NextCloudPi version | v1.46.10
-- | --
NextCloudPi image | NextCloudPi_docker_03-16-20
OS | Debian GNU/Linux 10. 4.19.0-18-amd64 (x86_64)
Nextcloud version | 22.2.2.0

Following the whole Output when trying to update from v1.46.10 to 1.47.1:

<details>
    <summary>Click to expand!</summary>

```
[ nc-update ] (Sun Apr 17 23:22:03 CEST 2022)
Downloading updates
Performing updates
Installing metrics
--2022-04-17 23:22:15-- https://github.com/theCalcaholic/ncp-metrics-exporter/releases/download/v1.1.0/ncp-metrics-exporter-x86_64
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/467263699/9cb2ac49-2d78-4bb5-8630-a4dd062edfa0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220417%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220417T212215Z&X-Amz-Expires=300&X-Amz-Signature=9290a28381e22f7230c74ee8a662a2f0b61b08b8e3c86ebb079b68bffb4bd8a5&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=467263699&response-content-disposition=attachment%3B%20filename%3Dncp-metrics-exporter-x86_64&response-content-type=application%2Foctet-stream [following]
--2022-04-17 23:22:15-- https://objects.githubusercontent.com/github-production-release-asset-2e65be/467263699/9cb2ac49-2d78-4bb5-8630-a4dd062edfa0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220417%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220417T212215Z&X-Amz-Expires=300&X-Amz-Signature=9290a28381e22f7230c74ee8a662a2f0b61b08b8e3c86ebb079b68bffb4bd8a5&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=467263699&response-content-disposition=attachment%3B%20filename%3Dncp-metrics-exporter-x86_64&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.108.133, 185.199.109.133, 185.199.110.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.108.133|:443...
connected.
HTTP request sent, awaiting response...
200 OK
Length: 8038504 (7.7M) [application/octet-stream]
Saving to: '/usr/local/lib/ncp-metrics/ncp-metrics-exporter'

0K .......... .......... .......... .......... .......... 0% 2.15M 4s
50K .......... .......... .......... .......... .......... 1% 2.96M 3s
100K .......... .......... .......... .......... .......... 1% 2.20M 3s
150K .......... .......... .......... .......... .......... 2% 2.51M 3s
200K .......... .......... .......... .......... .......... 3% 2.47M 3s
250K .......... .......... .......... .......... .......... 3% 2.50M 3s
300K .......... .......... .......... .......... .......... 4% 2.52M 3s
350K .......... .......... .......... .......... .......... 5% 712K 4s
400K .......... .......... .......... .......... .......... 5% 1016K 4s
450K .......... .......... .......... .......... .......... 6% 990K 5s
500K .......... .......... .......... .......... .......... 7% 769K 5s
550K .......... .......... .......... .......... .......... 7% 1.38M 5s
600K .......... .......... .......... .......... .......... 8% 5.10M 5s
650K .......... .......... .......... .......... .......... 8% 4.77M 4s
700K .......... .......... .......... .......... .......... 9% 4.53M 4s
750K .......... .......... .......... .......... .......... 10% 3.00M 4s
800K .......... .......... .......... .......... .......... 10% 3.99M 4s
850K .......... .......... .......... .......... .......... 11% 8.07M 4s
900K .......... .......... .......... .......... .......... 12% 8.29M 3s
950K .......... .......... .......... .......... .......... 12% 10.4M 3s
1000K .......... .......... .......... .......... .......... 13% 9.44M 3s
1050K .......... .......... .......... .......... .......... 14% 9.27M 3s
1100K .......... .......... .......... .......... .......... 14% 10.6M 3s
1150K .......... .......... .......... .......... .......... 15% 6.93M 3s
1200K .......... .......... .......... .......... .......... 15% 8.71M 3s
1250K .......... .......... .......... .......... .......... 16% 7.18M 3s
1300K .......... .......... .......... .......... .......... 17% 7.03M 3s
1350K .......... .......... .......... .......... .......... 17% 8.42M 2s
1400K .......... .......... .......... .......... .......... 18% 10.8M 2s
1450K .......... .......... .......... .......... .......... 19% 16.6M 2s
1500K .......... .......... .......... .......... .......... 19% 9.48M 2s
1550K .......... .......... .......... .......... .......... 20% 6.78M 2s
1600K .......... .......... .......... .......... .......... 21% 8.13M 2s
1650K .......... .......... .......... .......... .......... 21% 8.85M 2s
1700K .......... .......... .......... .......... .......... 22% 8.51M 2s
1750K .......... .......... .......... .......... .......... 22% 6.76M 2s
[... 491 lines cut ...]
--2022-04-17 23:59:05--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/467263699/9cb2ac49-2d78-4bb5-8630-a4dd062edfa0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220417%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220417T215905Z&X-Amz-Expires=300&X-Amz [... cut ...]
+ sed -i 's|status_of_proc "$DAEMON" "$NAME" ${PIDFILE:="-p ${PIDFILE}"}|status_of_proc ${PIDFILE:+-p "$PIDFILE"} "$DAEMON" "$NAME"|' /lib/init/init-d-script
+ [[ '' == 1 ]]
+ apt_install_with_recommends prometheus-node-exporter
+ apt-get update --allow-releaseinfo-change
+ DEBIAN_FRONTEND=noninteractive
+ apt-get install -y -o Dpkg::Options::=--force-confdef -o Dpkg::Options::=--force-confold prometheus-node-exporter
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

</details>

Thank you very much for your help and I wish you a nice day! :)

theCalcaholic, 2022-04-18:
Thanks for the detailed logs! I can see what's causing the issue now. Apparently my script didn't reliably detect whether it was running in a docker container.

I have pushed a fix to devel. Can you try executing the following command and report if it works?

```sh
ncp-update devel
```

Tommy-01, 2022-04-18:
Thank you for take care!

After the execution of:
`ncp-update devel`

I get this:

<details>
  <summary>Click to expand!</summary>
`INFO: updating to development branch 'devel'
Downloading updates
Performing updates
Installing metrics
--2022-04-18 16:42:43--  https://github.com/theCalcaholic/ncp-metrics-exporter/releases/download/v1.1.0/ncp-metrics-exporter-x86_64
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/467263699/9cb2ac49-2d78-4bb5-8630-a4dd062edfa0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220418%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220418T164243Z&X-Amz-Expires=300&X-Amz-Signature=04b08a95b7af8a54d6acf5acee043cdf15d2de1e065d5a43981c30f53944c100&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=467263699&response-content-disposition=attachment%3B%20filename%3Dncp-metrics-exporter-x86_64&response-content-type=application%2Foctet-stream [following]
--2022-04-18 16:42:43--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/467263699/9cb2ac49-2d78-4bb5-8630-a4dd062edfa0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220418%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220418T164243Z&X-Amz-Expires=300&X-Amz-Signature=04b08a95b7af8a54d6acf5acee043cdf15d2de1e065d5a43981c30f53944c100&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=467263699&response-content-disposition=attachment%3B%20filename%3Dncp-metrics-exporter-x86_64&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.109.133, 185.199.111.133, 185.199.110.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 8038504 (7.7M) [application/octet-stream]
Saving to: '/usr/local/lib/ncp-metrics/ncp-metrics-exporter'

/usr/local/lib/ncp- 100%[===================>]   7.67M   255KB/s    in 24s     

2022-04-18 16:43:08 (325 KB/s) - '/usr/local/lib/ncp-metrics/ncp-metrics-exporter' saved [8038504/8038504]

Hit:1 http://deb.debian.org/debian bullseye InRelease                          
Get:2 http://deb.debian.org/debian bullseye-updates InRelease [39.4 kB]
Get:3 http://security.debian.org/debian-security bullseye-security InRelease [44.1 kB]
Get:4 http://security.debian.org/debian-security bullseye-security/main amd64 Packages [125 kB]
Fetched 208 kB in 1min 53s (1845 B/s)                                          
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
prometheus-node-exporter is already the newest version (1.1.2+ds-2.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
3 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up prometheus-node-exporter (1.1.2+ds-2.1) ...
invoke-rc.d: could not determine current runlevel
invoke-rc.d: policy-rc.d denied execution of start.
Setting up dbus (1.12.20-2) ...
invoke-rc.d: could not determine current runlevel
invoke-rc.d: policy-rc.d denied execution of start.
Setting up prometheus-node-exporter-collectors (0+git20210115.7d89f19-1) ...
Stopping Prometheus exporter for machine metrics: prometheus-node-exporter.
Stopping ncp-exporter: ncp-exporter.
disabled ncp-metrics
prometheus-node-exporter is not running ... failed!
Stopping Prometheus exporter for machine metrics: prometheus-node-exporter.
ncp-exporter is not running ... failed!
Stopping ncp-exporter: ncp-exporter.
Installing metrics
--2022-04-18 16:56:51--  https://github.com/theCalcaholic/ncp-metrics-exporter/releases/download/v1.1.0/ncp-metrics-exporter-x86_64
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/467263699/9cb2ac49-2d78-4bb5-8630-a4dd062edfa0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220418%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220418T165652Z&X-Amz-Expires=300&X-Amz-Signature=917dc03bf623bf2cd50b098a9023551e1b3e815bd0d6fff5ec0741a245036c2b&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=467263699&response-content-disposition=attachment%3B%20filename%3Dncp-metrics-exporter-x86_64&response-content-type=application%2Foctet-stream [following]
--2022-04-18 16:56:52--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/467263699/9cb2ac49-2d78-4bb5-8630-a4dd062edfa0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220418%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220418T165652Z&X-Amz-Expires=300&X-Amz-Signature=917dc03bf623bf2cd50b098a9023551e1b3e815bd0d6fff5ec0741a245036c2b&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=467263699&response-content-disposition=attachment%3B%20filename%3Dncp-metrics-exporter-x86_64&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.108.133, 185.199.110.133, 185.199.111.133, ...
[... cut ...]
Hit:3 http://deb.debian.org/debian bullseye-updates InRelease
Reading package lists... Done                                
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
prometheus-node-exporter is already the newest version (1.1.2+ds-2.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Stopping Prometheus exporter for machine metrics: prometheus-node-exporter.
Stopping ncp-exporter: ncp-exporter.
disabled ncp-metrics
NextCloudPi updated to version v1.47.1`
</details>

theCalcaholic, 2022-04-18:
Looks good, thank you :)
Don't worry about the development branch. It only affects this single execution of ncp-update. NCP will still receive further updates from master

rondadon, 2022-04-18:
Thank you @theCalcaholic!
I ran "ncp-update devel" and it updated to 1.47.1 .
Following the output of the Update Process:

<details>
    <summary>Click to expand!</summary>

```

root@ncp:~# docker exec -it ncpdocker ncp-update devel      
INFO: updating to development branch 'devel'
Downloading updates
Performing updates
Installing metrics
--2022-04-18 21:17:15--  https://github.com/theCalcaholic/ncp-metrics-exporter/releases/download/v1.1.0/ncp-metrics-exporter-x86_64
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/467263699/9cb2ac49-2d78-4bb5-8630-a4dd062edfa0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220418%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220418T191715Z&X-Amz-Expires=300&X-Amz-Signature=d45982f5072a50c1801f0130e3ccfaae0e24bf8e6fc08470cad0898040714d7d&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=467263699&response-content-disposition=attachment%3B%20filename%3Dncp-metrics-exporter-x86_64&response-content-type=application%2Foctet-stream [following]
--2022-04-18 21:17:15--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/467263699/9cb2ac49-2d78-4bb5-8630-a4dd062edfa0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220418%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220418T191715Z&X-Amz-Expires=300&X-Amz-Signature=d45982f5072a50c1801f0130e3ccfaae0e24bf8e6fc08470cad0898040714d7d&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=467263699&response-content-disposition=attachment%3B%20filename%3Dncp-metrics-exporter-x86_64&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.108.133, 185.199.109.133, 185.199.110.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 8038504 (7.7M) [application/octet-stream]
Saving to: '/usr/local/lib/ncp-metrics/ncp-metrics-exporter'

/usr/local/lib/ncp-metrics/ncp-metrics-exp 100%[=======================================================================================>]   7.67M  8.35MB/s    in 0.9s    

2022-04-18 21:17:16 (8.35 MB/s) - '/usr/local/lib/ncp-metrics/ncp-metrics-exporter' saved [8038504/8038504]

Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://security.debian.org/debian-security bullseye-security InRelease
Get:3 http://deb.debian.org/debian bullseye-updates InRelease [39.4 kB]
Fetched 39.4 kB in 1s (65.4 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
prometheus-node-exporter is already the newest version (1.1.2+ds-2.1).
0 upgraded, 0 newly installed, 0 to remove and 31 not upgraded.
3 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up prometheus-node-exporter (1.1.2+ds-2.1) ...
invoke-rc.d: could not determine current runlevel
invoke-rc.d: policy-rc.d denied execution of start.
Setting up dbus (1.12.20-2) ...
invoke-rc.d: could not determine current runlevel
invoke-rc.d: policy-rc.d denied execution of start.
Setting up prometheus-node-exporter-collectors (0+git20210115.7d89f19-1) ...
Stopping Prometheus exporter for machine metrics: prometheus-node-exporter.
Stopping ncp-exporter: ncp-exporter.
disabled ncp-metrics
prometheus-node-exporter is not running ... failed!
Stopping Prometheus exporter for machine metrics: prometheus-node-exporter.
ncp-exporter is not running ... failed!
Stopping ncp-exporter: ncp-exporter.
Installing metrics
--2022-04-18 21:17:48--  https://github.com/theCalcaholic/ncp-metrics-exporter/releases/download/v1.1.0/ncp-metrics-exporter-x86_64
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/467263699/9cb2ac49-2d78-4bb5-8630-a4dd062edfa0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220418%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220418T191715Z&X-Amz-Expires=300&X-Amz-Signature=d45982f5072a50c1801f0130e3ccfaae0e24bf8e6fc08470cad0898040714d7d&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=467263699&response-content-disposition=attachment%3B%20filename%3Dncp-metrics-exporter-x86_64&response-content-type=application%2Foctet-stream [following]
--2022-04-18 21:17:48--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/467263699/9cb2ac49-2d78-4bb5-8630-a4dd062edfa0?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220418%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220418T191715Z&X-Amz-Expires=300&X-Amz-Signature=d45982f5072a50c1801f0130e3ccfaae0e24bf8e6fc08470cad0898040714d7d&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=467263699&response-content-disposition=attachment%3B%20filename%3Dncp-metrics-exporter-x86_64&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.111.133, 185.199.110.133, 185.199.109.133, ...
[... cut ...]
Stopping Prometheus exporter for machine metrics: prometheus-node-exporter.
Stopping ncp-exporter: ncp-exporter.
disabled ncp-metrics
NextCloudPi updated to version v1.47.1

```

</details>

There are some `failed` mentions of `prometheus-node-exporter`  and `ncp-exporter` not running but I think the failed messages are referencing the stopping process not being able to stop `prometheus-node-exporter`  and `ncp-exporter` because they are allready not running.

Thank you for your help, your time and solution to the Problem. I wish you a nice day! :)

theCalcaholic, 2022-04-18:
Yes, I think so too :)

I'm glad it worked

gcarlino97, 2022-05-22:
Hi,
I hava the same problem but when run "ncp-update devel" 

the result is: 
- prometheus-node-exporter: unrecognized service
- Error while applying update 1.47.0. Exiting...

Any idea?

TvT82, 2022-06-03:
> Hi, I hava the same problem but when run "ncp-update devel"
> 
> the result is:
> 
>     * prometheus-node-exporter: unrecognized service
> 
>     * Error while applying update 1.47.0. Exiting...
> 
> 
> Any idea?

Same here !

svenb1234, 2022-07-03:
Read the comments above https://github.com/nextcloud/nextcloudpi/issues/1462#issuecomment-1100631895 describes how to solve the issue manually.