nextcloud/nextcloudpi #1463: Installing dnsmasq with creating docker image does not automatically start dnsmasq while this is expected

nvdw, 2022-04-16:
After installing `dnsmasq` (`apt-get install --no-install-recommends -y dnsmasq`) the status of the `dnsmasq` service is checked, if the outcome of this check shows that the `dnsmasq` service is not running it is assumed that `dnsmasq` service is not started because the `systemd-resolved `is running. 
With issue #1446 it is the case that `systemd-resolved` is the reason why the `dnsmasq` service is not running, but with creating a docker image `systemd-resolved` does not prevent the `dnsmasq` service from starting. The `dnsmasq` service is not started because it is tried to be started in a docker container.

Please see also issue #1446

```
# build/build-docker.sh armhf
...
#17 50.32 Unpacking dnsmasq (2.85-1) ...
#17 50.46 Setting up runit-helper (2.10.3) ...
#17 50.49 Setting up libnfnetlink0:armhf (1.0.1-3+b1) ...
#17 50.52 Setting up libnetfilter-conntrack3:armhf (1.0.8-3) ...
#17 50.55 Setting up dnsmasq-base (2.85-1) ...
#17 50.94 Setting up dnsmasq (2.85-1) ...
#17 51.51 invoke-rc.d: could not determine current runlevel
#17 51.52 invoke-rc.d: policy-rc.d denied execution of start.
#17 51.55 Processing triggers for libc-bin (2.31-13+deb11u3) ...
#17 51.71 Applying workaround for dnsmasq bug (compare issue #1446)
#17 51.72 systemd-resolved: unrecognized service
```
```
# cat /usr/sbin/policy-rc.d
#!/bin/sh

# For most Docker users, "apt-get install" only happens during "docker build",
# where starting services doesn't work and often fails in humorous ways. This
# prevents those failures by stopping the services from attempting to start.

exit 101
```

Maybe with the fix for issue #1446 a additional test can be added to check (also) the status of the `systemd-resolved` service (if it is running and/or if it exists), before trying to stop the `systemd-resolved` service and start the `` service?

Thanks, Niels
[ncpi.log](https://github.com/nextcloud/nextcloudpi/files/8500319/ncpi.log)

**System information**

theCalcaholic, 2022-04-18:
Oh, that's interesting. Thanks for the synopsis!
If I understand it correctly, the easiest fix would probably be to only apply the workaround on non-docker instances.

theCalcaholic, 2022-04-18:
@nvdw I believe, I have fixed the error (and a few others that came up). Fix is on devel. I can now build succesfully, feel free to verify that :)

nvdw, 2022-04-19:
Hi @theCalcaholic ,
It also works for me. Thanks!

closed 2022-04-19 by theCalcaholic