nextcloud/nextcloudpi #1466: NCP Proxmox VM The value 'bullseye-security' is invalid for APT::Default-Release as such a release is not available in the sources

ovpc, 2022-04-19:
Created on request from https://github.com/nextcloud/nextcloudpi/issues/1437#

Please view ncp-report output at https://pastebin.com/8bJi0Nf5

Running ncp-dist-upgrade fails on NCP running on a Proxmox VM (installed using debian 10 base and install.sh)

NextCloudPi version v1.47.1
NextCloudPi image NextCloudPi_11-30-20
OS Debian GNU/Linux 10. 4.19.0-18-amd64 (x86_64)

On Proxmox VM sudo ncp-dist-upgrade ends with;

    Running hooks in /etc/ca-certificates/update.d...
    done.
    Processing triggers for initramfs-tools (0.140) ...
    update-initramfs: Generating /boot/initrd.img-5.10.0-10-amd64
    Processing triggers for dbus (1.12.20-2) ...
    Reading package lists... Done
    E: The value 'bullseye-security' is invalid for APT::Default-Release as such a release is not available in the sources
    Something went wrong. Fix it and try again

theCalcaholic, 2022-04-19:
Thank you for the report. I'll try to figure out the best solution for the inconsistent apt sources.

Don't mind the following link, I'm just pinning it for myself to use later :)

[bullseye official sources](https://gist.github.com/gustavorv86/d60a25ad5f70b0dfc382670d3dc6da8d)

theCalcaholic, 2022-04-19:
@ovpc Could you please share the output of `cat /etc/apt/sources.list`?

ovpc, 2022-04-19:
```
~ $ cat /etc/apt/sources.list
deb http://httpredir.debian.org/debian buster main
```

theCalcaholic, 2022-04-28:
@ovpc Sorry for the delay. The update should now respect your configured sources more reliably. Coud you please execute `ncp-update devel` and then `ncp-dist-upgrade`?

ovpc, 2022-04-30:
Thanks @theCalcaholic 

```
sudo ncp-update devel
INFO: updating to development branch 'devel'
Downloading updates
Performing updates
Update to bullseye available. Type 'sudo ncp-dist-upgrade' to upgrade
                                                                               
Broadcast message from oliver@ncpvm (pts/1) (Sat Apr 30 08:15:28 2022):        
                                                                               
Update to bullseye available. Type 'sudo ncp-dist-upgrade' to upgrade
                                                                               
ls: cannot access '/etc/php/7.4/fpm/conf.d/*-opcache.ini': No such file or directory
ls: cannot access '/etc/php/7.4/cli/conf.d/*-opcache.ini': No such file or directory
NextCloudPi updated to version v1.47.1
```
NCP was already on v1.47.1
Running sudo ncp-dist-upgrade  anyway...reverting when done ;-)

closed 2022-04-30 by ovpc