nextcloud/nextcloudpi #1469: Freshly built docker image for arm64 is broken

theCalcaholic, 2022-05-04:
When building for arm64, the container fails to initialize due to an error when starting apache.

## Steps to reproduce

1. Build the image using `./build/build-docker.sh arm64`
2. Start the container using `docker run -p 80:80 -p 443:443 -p 4443:4443 --rm --name nextcloudpi ownyourbits/nextcloudpi-arm64` (ensure that the packages `qemu-user` and `qemu-user-static` are installed on your system)

## Expected outcome

The container should initialize and present us with the activation page on https://localhost:443

## Actual outcome

The container breaks during startup caused by an error when starting apache. These are the container logs:

```
WARNING: The requested image's platform (linux/arm64/v8) does not match the detected host platform (linux/amd64) and no specific platform was requested
Initializing empty volume..
Making /usr/local/etc/ncp-config.d persistent ...
Making /etc/services-enabled.d persistent ...
Making /etc/letsencrypt persistent ...
Making /etc/shadow persistent ...
Making /etc/cron.d persistent ...
Making /etc/cron.daily persistent ...
Making /etc/cron.hourly persistent ...
Making /etc/cron.weekly persistent ...
Making /usr/local/bin persistent ...
Making /etc/apache2 persistent ...
Starting PHP-fpm
Starting Apache
Action 'start' failed.
The Apache error log may have more information.
Starting Redis
Starting Cron
Starting Postfix
ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/run/mysqld/mysqld.sock' (2)
Init done
^Cstopping Cron...
stopping Redis...
stopping Postfix...
stopping HPB...
notify_push: no process found
Stopping apache
httpd (no pid file) not running
Stopping PHP-fpm
Stopping mariaDB
/usr/bin/mysqladmin: connect to server at 'localhost' failed
error: 'Can't connect to local MySQL server through socket '/run/mysqld/mysqld.sock' (2)'
Check that mysqld is running and that the socket: '/run/mysqld/mysqld.sock' exists!
```

As suggested, I had a look at the apache error log and, alas, it did indeed have more information:

```
[Wed May 04 18:37:14.413857 2022] [core:emerg] [pid 125:tid 274907627536] (95)Operation not supported: AH00023: Couldn't create the ssl-cache mutex 
AH00016: Configuration Failed
```

I'm not sure what's causing this issue exactly - however I found this issue which looks very similar and might be worth having a look at: https://github.com/cytopia/devilbox/issues/862

theCalcaholic, 2022-05-04:
@nachoparker  Could you have a look at this? I'm a bit clueless what I'm doing wrong, but I think you might be able to see the issue :)

theCalcaholic, 2022-05-06:
I have applied [a workaround](https://github.com/nextcloud/nextcloudpi/blob/master/build/docker/Dockerfile#L192) for now. Still would like to fix the actual issue at some point

henriquemattos, 2022-05-23:
Hi, @theCalcaholic . I stumbled upon your repo and given your fix, I think the platform property in the FROM command is more important than the `Mutex posixsem`. At least in my case (in another project), I can't build for platform/arm64 and adding `Mutex posixsem` stops the error in the log but I'm still loosing requests. I'm assuming most are related to SSL configurations but I can't find the sweet spot. I will watch your thread to see how it develops. 

If your fix is really the fix, than my problem is different and I have no clue where to go. Sorry for the noise. 😁

Kind regards,
Henrique

closed 2022-06-04 by theCalcaholic