nextcloud/nextcloudpi #1473: Can't upgrade ncp in latest docker from v1.46.9 to v1.47.0: prometheus-node-exporter: unrecognized service

StableNarwhal, 2022-05-08:
**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.46.9
NextCloudPi image    NextCloudPi_docker_03-07-22
OS                   Debian GNU/Linux 11. 5.16.0-0.bpo.4-amd64 (x86_64)
automount            no
USB devices          sdb sdc sdd sde 
datadir              /data/nextcloud/data
data in SD           no
data filesystem      ext2/ext3
data disk usage      12T/13T
rootfs usage         25G/109G
swapfile             none
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    23.0.2.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               2:07

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            ***MANUALLY REMOVED SENSITIVE VALUE***
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "23.0.2.1",
        "overwrite.cli.url": "https:\/\/7e4e011d4624\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
[... 83 lines cut ...]
{"reqId":"T1IiMQkLnlrVb1sYbZR7","level":0,"time":"2022-05-08T10:10:09+02:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"new parser added : FeedIo\\Standard\\Rss","userAgent":"--","version":"23.0.2.1"}
{"reqId":"T1IiMQkLnlrVb1sYbZR7","level":0,"time":"2022-05-08T10:10:09+02:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"new parser added : FeedIo\\Standard\\Rdf","userAgent":"--","version":"23.0.2.1"}

```
</details>

***

Today I upgraded my server (OMV5 -> OMV6) and in the process I had to change some mount names (e.g. /srv/45sdf435435sdfgf435345 to /srv/example1)

When I redeployed my NCP docker and tried ncp-update, this was the output:

```
Downloading updates
Performing updates
prometheus-node-exporter: unrecognized service
prometheus-node-exporter: unrecognized service
Error while applying update 1.47.0. Exiting...
```

I also noticed, that in the system info, my port check 80 and 433 show closed, despite being open.

Nextcloud itself seems to function just fine, I suspect that something broke since I might already have been on 1.47.0 already, but since I had to redeploy the last docker image, those updates are not present anymore. 

Any help would be much appreciated.

rspierenburg, 2022-05-09:
I've noticed the port issue as well on my 2 deployments.

StableNarwhal, 2022-05-09:
After performing the steps laid out in https://github.com/nextcloud/nextcloudpi/issues/1462#issuecomment-1100631895, I was able to update to 1.47. Not sure why prometheus was missing, how that situation came to be in the first place and if this "fix" was sufficient or if some underlying issues still persist like the port issue.

StableNarwhal, 2022-05-14:
The ports are still shown as closed, additionally the HPB service is down.

`can't connect to push server: cURL error 7: Failed to connect toexample.com port 443: No route to host (see https://curl.haxx.se/libcurl/c/libcurl-errors.html) for https://example.com/push/test/cookie`

Despite the ports being forwarded and DDNS being setup correctly, I now can't reach my Nextcloud instance from outside its network.

theCalcaholic, 2022-05-16:
Sorry, i may have forgotten to push the fix to master. I'll do that in a bit.

Edov, 2022-05-26:
Any update on this topic? Problem still persists

Fooughhy, 2022-05-26:
Using "ncp-update devel" has not worked for me, though I am unsure if I am running the command correctly.

I'm using:
> docker exec -it nextcloudpi-network bash -c "source /usr/local/etc/library.sh && ncp-update devel"

As that is the only way I know of running a command in the ncp container. But perhaps that is not the best way?

TvT82, 2022-06-03:
> I had the same problem and I was able to fix it by manually choosing the v1.47.1 docker image and then upgrading to 1.47.2 with ncp-update. For some reason the v1.47.1 image was replaced by 1.46.9, does anyone know why?

Thx, this "fix" worked ! 
I am wondering why the "latest tag"  doesnt't seem to refer to the latest image version ?

raspberryenvoie, 2022-06-11:
The fix was pushed but I was unable to update too. A part of the solution was in [#1467 (comment)](https://github.com/nextcloud/nextcloudpi/issues/1462#issuecomment-1100631895).
This worked for me and lets me stick with the `latest` docker image:
1. Open the file /usr/local/bin/ncp/SYSTEM/metrics.sh (e.g. with `sudo nano /usr/local/bin/ncp/SYSTEM/metrics.sh`)
2. Add the following to the top in line 2 (the line right after but not before #!/bin/bash):
```
set -x
```
3. Update ncp:
```
ncp-update devel
```

svenb1234, 2022-07-03:
Still an issue with the latest docker image:

```
[ nc-update ] (Sun Jul 3 07:09:29 UTC 2022)
Downloading updates
Performing updates
prometheus-node-exporter: unrecognized service
prometheus-node-exporter: unrecognized service
Error while applying update 1.47.0. Exiting...
```

raspberryenvoie, 2022-07-05:
I don't understand why v1.47.2 isn't the latest tag on Docker hub.

svenb1234, 2022-07-06:
Most probably simply because nobody had the time to update the image yet.

theCalcaholic, 2022-07-07:
Thanks for reporting. There's actually an issue with the release automation that didn't set the `latest` tag. I updated the tag manually now, but still have to fix the release automation :)

theCalcaholic, 2022-07-28:
Should be fixed in latest master. Please attempt the update again if you had the issue previously

closed 2022-07-28 by theCalcaholic