nextcloud/nextcloudpi #1475: NC-Info: ports closed

vcbranco, 2022-05-11:
NCP v1.47.2
Debian 11 x86_64 curl instalation
The ports are shown closed but are open and everything is working has expected.
The problem happens after a new installation (clean Debian install)

NextCloudPi version  v1.47.2
NextCloudPi image    NextCloudPi_05-11-22
OS                   Debian GNU/Linux 11. 5.10.0-14-amd64 (x86_64)
automount            yes
USB devices          sdb 
datadir              /data/ncdata/data
data in SD           no
data filesystem      btrfs
data disk usage      105G/207G
rootfs usage         5,7G/16G
swapfile             /var/swap
dbdir                /data/ncdatabase
Nextcloud check      ok
Nextcloud version    23.0.4.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enp1s0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         yes
uptime               9:02

suxlex, 2022-05-11:
My installation also shows closed ports although both are open:
NCP v.1.47.1
OS | Debian GNU/Linux 10. 4.19.0-20-amd64 (x86_64)
port check 80 | closed
port check 443 | closed

Jeanoo, 2022-05-11:
they are also closed, and I can't access them from outside

CarlosCastaneda, 2022-05-15:
Same to me

mart-ch, 2022-05-15:
Same to me

portchecker.co shows 80/443 as open
ncp dashboard shows them as closed

NextCloudPi version | v1.47.2
-- | --
NextCloudPi image | NextCloudPi_10-08-21
OS | Debian GNU/Linux 10. 5.10.103-v8+ (aarch64)
automount | no
USB devices | sda sdb
datadir | /mnt/hd1/nc-datadir/data
data in SD | no
data filesystem | btrfs
data disk usage | 1.1T/1.9T
rootfs usage | 2.5G/29G
swapfile | /var/swap
dbdir | /var/lib/mysql
Nextcloud check | ok
Nextcloud version | 23.0.2.1
HTTPD service | up
PHP service | up
MariaDB service | up
Redis service | up
HPB service | up
Postfix service | up
internet check | ok
port check 80 | closed
port check 443 | closed
IP | xxxxxxx
gateway | xxxxx
interface | eth0
certificates | xxxxxxx
NAT loopback | no
uptime | 8days

Jeanoo, 2022-05-16:
I had no internet because my internet provider cut off the ipv4 without warning me. 
Now the ports are still closed but I can connect to my server from outside in ipv6

xaavro, 2022-05-27:
Has anybody tried to renew Let's Encrypt while having this issue?
Does it work?

mart-ch, 2022-06-15:
Hello, my lets encrypt renewal is automated with nextcloudpi (everytime successfull). ports still closed, tried restart. not sure what happened there.

theCalcaholic, 2022-08-15:
Fixed in ac064dfaba4536e1ea3f066e2a7f32e856e4cd74

closed 2022-08-15 by theCalcaholic

FadeFx, 2022-10-15:
seems the isue is back in 1.50.3:

'NextCloudPi version | v1.50.3
-- | --
NextCloudPi image | NextCloudPi_06-09-22
OS | Debian GNU/Linux 11
  |  
[9;0]. 5.15.61-v8+ (aarch64) |  
automount | yes
USB devices | sda
datadir | /media/myCloudDrive/ncdata
data in SD | no
data filesystem | btrfs
data disk usage | 1.4T/1.9T
rootfs usage | 7.1G/15G
swapfile | /var/swap
dbdir | /media/USBdrive/ncdatabase
Nextcloud check | ok
Nextcloud version | 24.0.5.1
HTTPD service | up
PHP service | up
MariaDB service | up
Redis service | up
HPB service | up
Postfix service | up
Internet check | ok
public IPv4 | xxx.xxx.xxx.xxx
public IPv6 | not found
Port check 80 | closed
Port check 443 | closed
IP | 10.0.0.4
gateway | 10.0.0.1
Interface | eth0
certificates | nextcloud.classified.url
NAT loopback | yes
Uptime | 6days
`

CarlosCastaneda, 2022-10-17:
I can confirm this too!