nextcloud/nextcloudpi #1479: Lets Encrypt renewal fails after moving to current Docker image

sunjam, 2022-05-15:
Pulled nextcloudpi:latest
Import ncp-config
Re-enable DNS
Re-enable webui
Updating ncp fails #1478 
Static IP and Domain are correctly listed under Trusted Domains
Port forwarding is properly setup, but incorrectly listed as closed with `nc-info` #1475 

```
Doctrine\DBAL\Exception: Failed to connect to the database: An exception occurred in the driver: SQLSTATE[HY000] [2002] No such file or directory in /data-ro/nextcloud/lib/private/DB/Connection.php:87
Stack trace:
#0 /data-ro/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1519): OC\DB\Connection->connect()
#1 /data-ro/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1041): Doctrine\DBAL\Connection->getWrappedConnection()
#2 /data-ro/nextcloud/lib/private/DB/Connection.php(237): Doctrine\DBAL\Connection->executeQuery()
#3 /data-ro/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php(345): OC\DB\Connection->executeQuery()
#4 /data-ro/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php(287): Doctrine\DBAL\Query\QueryBuilder->execute()
#5 /data-ro/nextcloud/lib/private/AppConfig.php(344): OC\DB\QueryBuilder\QueryBuilder->execute()
#6 /data-ro/nextcloud/lib/private/AppConfig.php(109): OC\AppConfig->loadConfigValues()
#7 /data-ro/nextcloud/lib/private/AppConfig.php(300): OC\AppConfig->getApps()
#8 /data-ro/nextcloud/lib/private/legacy/OC_App.php(972): OC\AppConfig->getValues()
#9 /data-ro/nextcloud/lib/private/Server.php(692): OC_App::getAppVersions()
#10 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(160): OC\Server->OC\{closure}()
#11 /data-ro/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#12 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(127): Pimple\Container->offsetGet()
#13 /data-ro/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#14 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query()
#15 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(182): OC\AppFramework\Utility\SimpleContainer->get()
#16 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(160): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#17 /data-ro/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(114): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#18 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(127): Pimple\Container->offsetGet()
#19 /data-ro/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#20 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query()
#21 /data-ro/nextcloud/lib/private/Server.php(1079): OC\AppFramework\Utility\SimpleContainer->get()
#22 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(160): OC\Server->OC\{closure}()
#23 /data-ro/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#24 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(127): Pimple\Container->offsetGet()
#25 /data-ro/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#26 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query()
#27 /data-ro/nextcloud/lib/private/Server.php(2040): OC\AppFramework\Utility\SimpleContainer->get()
#28 /data-ro/nextcloud/lib/private/Files/View.php(118): OC\Server->getLockingProvider()
#29 /data-ro/nextcloud/lib/private/Server.php(439): OC\Files\View->__construct()
#30 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(160): OC\Server->OC\{closure}()
#31 /data-ro/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#32 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(127): Pimple\Container->offsetGet()
#33 /data-ro/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#34 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query()
#35 /data-ro/nextcloud/lib/private/Server.php(1403): OC\AppFramework\Utility\SimpleContainer->get()
#36 /data-ro/nextcloud/lib/base.php(594): OC\Server->boot()
#37 /data-ro/nextcloud/lib/base.php(1087): OC::init()
#38 /data-ro/nextcloud/console.php(48): require_once('/data-ro/nextcl...')
#39 /data-ro/nextcloud/occ(11): require_once('/data-ro/nextcl...')
#40 {main}An unhandled exception has been thrown:
Doctrine\DBAL\Exception: Failed to connect to the database: An exception occurred in the driver: SQLSTATE[HY000] [2002] No such file or directory in /data-ro/nextcloud/lib/private/DB/Connection.php:87
Stack trace:
#0 /data-ro/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1519): OC\DB\Connection->connect()
#1 /data-ro/nextcloud/3rdparty/doctrine/dbal/src/Connection.php(1041): Doctrine\DBAL\Connection->getWrappedConnection()
#2 /data-ro/nextcloud/lib/private/DB/Connection.php(237): Doctrine\DBAL\Connection->executeQuery()
#3 /data-ro/nextcloud/3rdparty/doctrine/dbal/src/Query/QueryBuilder.php(345): OC\DB\Connection->executeQuery()
#4 /data-ro/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php(287): Doctrine\DBAL\Query\QueryBuilder->execute()
#5 /data-ro/nextcloud/lib/private/AppConfig.php(344): OC\DB\QueryBuilder\QueryBuilder->execute()
#6 /data-ro/nextcloud/lib/private/AppConfig.php(109): OC\AppConfig->loadConfigValues()
#7 /data-ro/nextcloud/lib/private/AppConfig.php(300): OC\AppConfig->getApps()
#8 /data-ro/nextcloud/lib/private/legacy/OC_App.php(972): OC\AppConfig->getValues()
#9 /data-ro/nextcloud/lib/private/Server.php(692): OC_App::getAppVersions()
#10 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(160): OC\Server->OC\{closure}()
#11 /data-ro/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#12 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(127): Pimple\Container->offsetGet()
#13 /data-ro/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#14 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query()
#15 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(182): OC\AppFramework\Utility\SimpleContainer->get()
[... 45 lines cut ...]
#18 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(127): Pimple\Container->offsetGet()
#19 /data-ro/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#20 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query()
#21 /data-ro/nextcloud/lib/private/Server.php(1079): OC\AppFramework\Utility\SimpleContainer->get()
#22 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(160): OC\Server->OC\{closure}()
#23 /data-ro/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#24 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(127): Pimple\Container->offsetGet()
#25 /data-ro/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#26 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query()
#27 /data-ro/nextcloud/lib/private/Server.php(2040): OC\AppFramework\Utility\SimpleContainer->get()
#28 /data-ro/nextcloud/lib/private/Files/View.php(118): OC\Server->getLockingProvider()
#29 /data-ro/nextcloud/lib/private/Server.php(439): OC\Files\View->__construct()
#30 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(160): OC\Server->OC\{closure}()
#31 /data-ro/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}()
#32 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(127): Pimple\Container->offsetGet()
#33 /data-ro/nextcloud/lib/private/ServerContainer.php(136): OC\AppFramework\Utility\SimpleContainer->query()
#34 /data-ro/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(56): OC\ServerContainer->query()
#35 /data-ro/nextcloud/lib/private/Server.php(1403): OC\AppFramework\Utility\SimpleContainer->get()
#36 /data-ro/nextcloud/lib/base.php(594): OC\Server->boot()
#37 /data-ro/nextcloud/lib/base.php(1087): OC::init()
#38 /data-ro/nextcloud/console.php(48): require_once('/data-ro/nextcl...')
#39 /data-ro/nextcloud/occ(11): require_once('/data-ro/nextcl...')
#40 {main}enabled letsencrypt
Done. Press any key...
```

sunjam, 2022-05-15:
Related to #673 
Used apt-get in docker to add systemctl and full-upgrade

`systemctl status mysqld`
```
# systemctl status mariadb
mariadb.service - MariaDB 10.5.15 database server
    Loaded: loaded (/lib/systemd/system/mariadb.service, enabled)
    Active: failed (dead)
root@321d749b679d:/# systemctl status mysqld
mysqld.service - MariaDB 10.5.15 database server
    Loaded: loaded (/lib/systemd/system/mysqld.service, disabled)
    Active: inactive (dead)
```

theCalcaholic, 2022-07-28:
@sunjam Since this issue is 2 months old, I suppose you don't have that container anymore (or found a solution)?

theCalcaholic, 2022-08-14:
I'm closing the issue for now. Feel free to respond and I'll reopen it :)

sunjam, 2022-08-17:
I did not find a solution.  Got no response on the forum or anywhere else.  I still have the container.  Any suggestions?

theCalcaholic, 2022-08-17:
What exactly is the issue at this point? That mysql is not running? Or that the letsencrypt certificates are not renewed automatically?

Does a simple `service mysql start` do anything?

sunjam, 2022-08-18:
Tried `systemctl start mysql` but it does nothing within the docker container.  Also, systemctl is not included within the container.

ZendaiOwl, 2022-08-19:
Mm you're right, `systemd` with `systemctl` isn't in the container, when I've been experimenting with building the docker containers I had to add "systemd" to be installed inside the base container for that to work 🙏

theCalcaholic, 2022-08-20:
@sunjam Yes, that's why I was suggesting to run `service mysql start` :)

theCalcaholic, 2022-08-20:
> I had to add "systemd" to be installed inside the base container
@ZendaiOwl Please don't do that outside of experimentation, though. Systemd doesn't work reliably inside docker

ZendaiOwl, 2022-08-20:
> > I had to add "systemd" to be installed inside the base container
> @ZendaiOwl Please don't do that outside of experimentation, though. Systemd doesn't work reliably inside docker

Alright 🙏 I didn't know that, do you know why that is?

theCalcaholic, 2022-08-20:
It's just not designed for that. Docker is meant to be a containerization for single processes (we're abusing it a bit in that regard) and a process management engine like systemd is meant for the opposite of the spectrum :)

Apparently, [it's not impossible to do anymore](https://developers.redhat.com/blog/2019/04/24/how-to-run-systemd-in-a-container) (some time ago you needed host system access in order to run systemd, but it's not a very good idea for production still. That's why we use a very simple bash script for process management.

ZendaiOwl, 2022-08-20:
@theCalcaholic 
Alright, yeah I found an article about this from someone working with container runtimes at red hat 🥳 

Seems podman was developed to resolve such issues, he also outlines some of the problems with using systemd on a docker container without properly setting it up, if I understood it correctly 🙏

https://developers.redhat.com/blog/2019/04/24/how-to-run-systemd-in-a-container#other_cool_features_about_podman_and_systemd

theCalcaholic, 2022-08-26:
@sunjam Are you still interested in fixing this? I'm still curious about the output of `service mysql start` :)

> @sunjam Yes, that's why I was suggesting to run `service mysql start` :)

ZendaiOwl, 2022-09-26:
This issue can be closed, it is caused by the internal upgrade within docker, which should not be done unless the image you're pulling from has gone through precisely & exactly the same upgrade process as you did within your container, including configurations, package installation and all. 

And if anyone comes here reading, you can use systemd within a container, it is not recommended to use with docker directly however and you should use podman for that, which was developed by red hat to resolve the issues of using systemd within a container more easily. As can be read in the article linked in one of the comments above

closed 2022-09-26 by ZendaiOwl