nextcloud/nextcloudpi #1480: Opcache Settings via NCP

feutl, 2022-05-20:
**Description**  
The error message `The OPcache interned strings buffer is nearly full.` in Nextcloud, starting with NC23, was the reason to dig a little deeper into the problem and I found this thread
https://help.nextcloud.com/t/nextcloud-23-02-opcache-interned-strings-buffer/134007

As far as I have understood, with NC23 it is capable of giving you recommendations for settings to be changed based on the individual usage of the server. In my case the `opcache` was full therefore the recommendation was to increase `opcache.interned_strings_buffer`

This was as not a problem for me to do manually, but this setting should be added to the `nc-limits` in ncp-web / ncp-cli.

saschabrockel, 2022-10-17:
@feutl sorry for asking but how to set the opcache value? I have the exact same "problem". I'm using docker and I see that the path `/etc/php/8.1/fpm/conf.d/10-opcache.ini` leads me to the target but I really do not want to map the different values of the config with a volume. Seems very dirty to me.

feutl, 2022-10-18:
Honestly, I have just changed my opcache.ini as recommended by NC
Even did that for the latest upgrade with PHP 8.1, NCP 15 and NC 24.

It would be nice to have that setting inside NCP, but as for now, it needs to be done manually.

feutl, 2022-11-17:
@saschabrockel I have no docker setup, so I just changed the values in the file according to the recommendations from NC itself. no volumes needed.

zedocrob, 2022-11-17:
Hi,
I've also modified `/etc/php/8.1/fpm/conf.d/10-opcache.ini` with only these:
`opcache.interned_strings_buffer=16`
`opcache.revalidate_freq=60`
like so: [https://github.com/nextcloud/nextcloudpi/pull/1661/files](url)

And enabled these ones directly in `/etc/php/8.1/fpm/php.ini`
`opcache.enable=1`
`opcache.max_accelerated_files=10000`
`opcache.interned_strings_buffer=16`
`opcache.memory_consumption=128`
`opcache.save_comments=1`
`opcache.revalidate_freq=60`

Am i right?

feutl, 2022-12-29:
OK, I have migrated to NCP Docker today, and now I am on the same boat.
How do I permanently set the opcache for php ?! The php config files are not stored in the mounted volume. I can only find apache and cron there.

@theCalcaholic this would be very important to be integrated into the docker setup, because otherwise the values are not persistent after a docker reboot.